opentrack/opentrack #275: Regression vs rc20: Stuttering seen in Elite Dangerous

wstephenson, 2015-12-03:
Just tried rc21p2 on PC with EDTrackerPro and Accela filter. 

I get extremely jerky movement and stutter in Elite Dangerous.  The game moves my ingame view a small amount then appears to jump back to before the point where I started moving.

In the mappings view and the preview in the opentrack UI, the stutter is not noticeable. If I revert back to rc20, the stutter goes away again. Also tried Kaiman filter, no improvement in rc21p2.

opentrack .ini file 
[HORIZONS.txt](https://github.com/opentrack/opentrack/files/51341/HORIZONS.txt)

sthalik, 2015-12-03:
Hey, here's a new build. if #275 and #274 confirm it's working it's gonna be patchlevel 3.

Build: https://db.tt/UthSld0u

wstephenson, 2015-12-05:
Yes, that build fixes this issue.

closed 2015-12-05 by wstephenson