opentrack/opentrack #276: How translate old profiles to new.

kzfr, 2015-12-04:


kzfr, 2015-12-04:
Hello, i have a some issue with old .ini's I try run this in new unstable version. I got all time a crash. This is old ini.

[quote][opentrack-ui]
compensate-translation=false
compensate-translation-disable-z-axis=false
filter-dll=Accela Filter Mk4
key-alt-center=false
key-alt-toggle=false
key-ctrl-center=false
key-ctrl-toggle=false
key-index-center=0
key-index-toggle=0
key-shift-center=false
key-shift-toggle=false
pitch-alt-axis-sign=false
pitch-invert-axis=false
pitch-zero-pos=0
protocol-dll=
roll-alt-axis-sign=false
roll-invert-axis=false
roll-zero-pos=0
tracker-dll=
tracker2-dll=
x-alt-axis-sign=false
x-invert-axis=false
x-zero-pos=0
y-alt-axis-sign=false
y-invert-axis=false
y-zero-pos=0
yaw-alt-axis-sign=false
yaw-invert-axis=false
yaw-zero-pos=0
z-alt-axis-sign=false
z-invert-axis=false
z-zero-pos=0

[tracker-pt]
active-model-panel=0
camera-focal-length=1
camera-fps=75
camera-index=0
camera-pitch=0
camera-res-height=480
camera-res-width=640
camera-roll=1
camera-yaw=0
cap-x=0
cap-y=0
cap-z=0
clip-by=70
clip-bz=80
clip-ty=40
clip-tz=30
dynamic-pose-resolution=false
enable-pitch=true
enable-roll=false
enable-x=false
enable-y=false
enable-yaw=true
enable-z=false
m_01-x=0
m_01-y=40
m_01-z=-30
m_02-x=0
m_02-y=-70
m_02-z=-80
max-point-size=50
min-point-size=10
model-centroid-x=150
model-centroid-y=80
model-centroid-z=5
reset-time=2000
threshold-primary=31
threshold-secondary=26
video-widget=true

[proto-freetrack]
ezca-mode=false
use-memory-hacks=false
used-interfaces=1

[Curves-rx]
point-count=4
point-0-x=5.19999980926514
point-0-y=23
point-1-x=10.3999996185303
point-1-y=56
point-2-x=15.8000001907349
point-2-y=114
point-3-x=19.3999996185303
point-3-y=180

[Curves-ry]
point-count=1
point-0-x=10
point-0-y=90

[Curves-rz]
point-count=1
point-0-x=60
point-0-y=60

[Curves-tx]
point-count=1
point-0-x=59.8888893127441
point-0-y=100

[Curves-ty]
point-count=1
point-0-x=60
point-0-y=100

[Curves-tz]
point-count=2
point-0-x=19.7142848968506
point-0-y=51
point-1-x=60
point-1-y=100

[Curves-tx_alt]
point-count=0

[Curves-ty_alt]
point-count=0

[Curves-tz_alt]
point-count=1
point-0-x=19.7142848968506
point-0-y=50

[Curves-rx_alt]
point-count=0

[Curves-ry_alt]
point-count=2
point-0-x=0
point-0-y=0
point-1-x=9.80000019073486
point-1-y=89

[Curves-rz_alt]
point-count=0

[Accela]
exponent=2.15
rotation-alpha=60
rotation-deadband=0
second-order-alpha=100
third-order-alpha=180
translation-alpha=0
translation-deadband=0

[Tracking]
tcomp-rz=false
compensate=true
rx_alt=false
ry_alt=false
rz_alt=false
tx_alt=false
ty_alt=false
tz_alt=false
zero_tx=0
zero_ty=0
zero_tz=0
zero_rx=0
zero_ry=0
zero_rz=0
invertYaw=false
invertPitch=false
invertRoll=false
invertX=false
invertY=false
invertZ=false

[GameProtocol]
DLL=libopentrack-proto-freetrack.dll

[TrackerSource]
DLL=libopentrack-tracker-pt.dll
2ndDLL=

[Filter]
DLL=libopentrack-filter-accela.dll
[/quote]

sthalik, 2015-12-04:
See if this build fixes crashes: https://db.tt/UthSld0u

I'm releasing it as patchlevel 3 without further testing cause of all the problems.

sthalik, 2015-12-04:
I don't understand the part about leaning.

There's no code to do migration, not enough developer resources.

kzfr, 2015-12-05:
https://www.youtube.com/watch?v=As3yV_wREFU - this is leaning setting but when you set that you cant use "x - curve" normal.

sthalik, 2015-12-05:
I don't wanna watch 10 minutes of unrelated video, please explain.

sthalik, 2015-12-17:
I'm closing this due to lack of further response.

closed 2015-12-17 by sthalik