opentrack/opentrack #277: write a quick start manual

sthalik, 2015-12-14:
Can someone write a quick start manual on the wiki, please? Some trackers and filters are documented, you can link to those.

Len62, 2015-12-16:
I might be willing to give this a try for you. I take it that I can just add-a-page to the wiki and then link to the docs? Also, I really only work with the windows version, using accela and PT. I'll have to install a version using the installer just to see how that sets things up, I normally just use the zipped version. If you want me to give it a try let me know and then give me a couple of days to work something up.

sthalik, 2015-12-16:
What's needed first is basic explanation of mappings, shortcuts and most important settings. Then we can think of quick start for individual trackers. I keep reading of people seeing the mapping screen and not knowing what it's about, and so on.

sthalik, 2015-12-17:
The basic program pipeline is

tracker -> filter -> mapping -> protocol

These need to be described generically. Then we can get to describing individual protocols.

Len62, 2015-12-17:
Ok, I'll start drafting something out tomorrow along those lines.

Len62, 2015-12-19:
Sorry for the delay. Got wrapped up finishing a project for my wife. I should have something ready to go by late tomorrow. Any problems with including a few screen pics?

sthalik, 2015-12-19:
You can include photos and videos with no issues. These can be later hosted on the wiki repo itself.

Len62, 2015-12-19:
Okay, I have to store some screen shots on-line with drop-box so I can include them. I'll do some practice w/o saving a new page to get familiar with how it works.

Len62, 2015-12-19:
Okay, cannot get a png linked from dropbox to display in the wiki. Any suggestions as to how/where I can upload the pics?

sthalik, 2015-12-19:
It'll work if you link directly to the file.

Len62, 2015-12-19:
Okay, got that sorted out, needed to edit the url.

Len62, 2015-12-20:
Well, I posted a quick start guide. Hopefully it will be of some help for people just trying out OT. Feel free to edit it as you see fit. I tend to be detail oriented and it can be hard for me to just get to the basics. I'll continue to work on it and at some point I would like to do a video, now that may take some time. What would you propose to add next ? I was thinking more detail on entering camera data.

Len62, 2015-12-20:
How about an installation page on the wiki? Might not be needed for some users but people new to opentrack might want to know how to use the zip version. How does the OSX version get installed? Same as the zip for Win?

sthalik, 2016-01-07:
@Len62 I don't think users unable to decompress the portable installation would want the portable installation in the first place. Actually you can copy the installation directory onto a USB stick and it'll work. The contents of the installer and the portable archive are identical. The former is merely wrapped in an executable installer.

For Mac it's simple - double-clicking the .zip file extracts it automatically, and double-clicking on the .app folder starts the application just like that. We don't need instructions for that, that's how installing/running software on OSX works in general.

I still have no time to contribute on the wiki. I'll eventually make separate pages for the core concepts (mappings, filtering, profiles, basic options like centering), and a page for each of the popular trackers.

Also I'd love to make Accela a single slider with obviously-named presets controlling the way the slider affects the internal options, that are now exposed to the user in all their complex glory. The only way to understand the options is to understand how Accela exactly works, and I can't ask anyone for that.

Len62, 2016-01-07:
I'm working on a short video to show a basic set-up for OT. Now that the holidays are over I have some time to work on it. I won't worry about doing an installation guide. I'm new at this video making and I'm experimenting with screen capture software. Open Broadcast seems to fit the bill.

suhaibmalik, 2016-01-14:
I'm also working on getting some documentation out soon. I was already working on creating a guide for internal use but I can easily extend its scope to be more inclusive.

Len62, 2016-01-15:
Ok, finished some testing creating a video using Open Broadcast, and shrinking the file size using Handbrake. Probably take a few more tries to get one I like. My plan is to get this done later today, or by tomorrow. It takes some getting used to speaking to the computer.

Len62, 2016-01-19:
@sthalik Just added a short video re quick start. It's located in my dropbox account. Not sure if you want to upload it to youtube. I preparing to go out of town for about 10 days and will not be able to add anything else to after my return.

sthalik, 2016-01-20:
@Len62 you can share it now if you want. We're not on a schedule though.

Len62, 2016-01-20:
@sthalik Okay, I uploaded it to youtube and updated the link in the wiki.

Len62, 2016-01-25:
@sthalik I'm getting ready to do a short video on the mapping window. On the rotation side I know the scales are in degrees. What do the numbers on the translation side indicate?

iDevPro, 2016-04-23:
// OPENTRACK OS X BUILD

# git clone https://github.com/opentrack/opentrack.git
# cd opentrack
# brew update
# brew install qt5
# export Qt5_DIR="/usr/local/opt/qt5»
# cmake -G Xcode .
# xcodebuild
# cp /tmp/opentrack***/opentrack.app /Applications

sthalik, 2016-06-16:
@Len62 thanks for the work and youtube videos.

sthalik, 2016-06-16:
@iDevPro that's not the right way. Use makefiles. I use the xcode cmdline toolchain only for freshest llvm. Skip xcode build system entirely. If you must edit, use Qt Creator.

closed 2018-01-04 by sthalik