opentrack/opentrack #279: Very high CPU usage

aka2k, 2015-12-15:
Hi,

 First and foremost, thanks for the great job with opentrack. It's an amazing piece of software. Now onto the issue: since rc21 the CPU usage is very high, averaging around 34%-39% on my i5 2500k. The older versions seldom go above 7-8% and even less if I switch to 320x240 resolution or switch to 30fps.

 My hardware: core i5 2500k, Windows 10 Pro TH2, Delanclip and PS3 Eye w/o IR filter.

[image]
[image]

sthalik, 2015-12-16:
Few things:

- are you using automatic thresholding (the checkbox next to the slider)
- is it also high if you select a UDP tracker or a joystick tracker?

sthalik, 2015-12-16:
From the screenshots it looks like the CPU usage is as high when the tracking's not active. This must be the joystick binding bug.

sthalik, 2015-12-16:
To make it clear - is the CPU usage at 25% or more when your joystick-like devices are unplugged?

sthalik, 2015-12-16:
Get the "process explorer" app from Microsoft SysInternals suite.

Click on the opentrack process, go to the "threads" tab.

Double-click on the errant thread to get a crude backtrace. Save a screenshot and paste it here.

aka2k, 2015-12-16:
Give me a few hours and I'll do that for you, I'm a bit busy right now but as soon as possible I'll have that. Thanks for replying.

aka2k, 2015-12-16:
Unplugged all my joysticks, didn't changed anything. CPU usage hovers on 20-25% on Task Manager and opentrack process hovers around 17% on Process Explorer. Pic attached.

[image]

sthalik, 2015-12-16:
Click the high-CPU thread and press 'stack', then paste.

aka2k, 2015-12-17:
Not sure if I'm following your instructions right, but is this it?

ntoskrnl.exe!KeSynchronizeExecution+0x3de6
ntoskrnl.exe!KeWaitForSingleObject+0xc7a
ntoskrnl.exe!KeWaitForSingleObject+0x709
ntoskrnl.exe!KeWaitForSingleObject+0x375
ntoskrnl.exe!IoQueueWorkItem+0x370
ntoskrnl.exe!KeRemoveQueueEx+0x16ba
ntoskrnl.exe!KeSynchronizeExecution+0x2543
KERNELBASE.dll!RegOpenKeyExInternalW+0x137

sthalik, 2015-12-17:
Given this is the stack for the high-CPU process and given one from the screenshot hangs in DirectInput8Create - it's definitely a joystick bug.

The reason is unknown. We need to find out what rubs dinput the wrong way. Please follow these instructions, as I can't reproduce it on my end.

- install mingw-w64 using their official installer
- open Windows console
- change directory to where opentrack is installed
- run gdb.exe from mingw-w64 installation by dragging gdb.exe into the console window
- since you're in opentrack install directory, just type **file opentrack.exe**
- type **run**
- wait for the bug to occur
- press **control + c**
- type **thread apply all bt full**

For that you need a debug build. Here's one. The release build won't have the necessary information.

Build - https://db.tt/M7wDbrMn

exulant, 2015-12-17:
Hello there. First off all I would like to thank you very much for your dedicated work on this project sthalik.

I actually have the noticed the same problem as aka2k. Just launching Opentrack 2.3 rc21p4 and letting it idle consumes about 20% CPU usage and starting the Aruco-Tracker it raises to about 30%, whereas the version 2.3 rc20 consumes about 0% in idle and 10% CPU usage with Aruco started.

I am running on an Intel i5 3750K and Win7 64-bit with a PS3 Eye-Cam with CL-Driver 5.3.0.0341 and a Saitek 45 with standard Microsoft drivers in case this information is to any use of you. I actually tried unplugging the joystick as well as the webcam but to no avail.

I also tried your instructions regarding mingw-w64 with the debug build but regardless whether exception I choose during the installation (seh or sjlj) and let CMD shell run twice for about half an hour I never could cancel the gdb.exe and it stuck so I couldn't type "thread apply all bt full".

Anyway I will send you whatever mingw-w64 put out via email and hope that it still helps.

sthalik, 2015-12-17:
@exulant Unfortunately I can't do anything without the backtrace.

Do the following instead:

- start Task Manager
- go to the details tab
- click on the column name, choose 'select columns'
- select PID
- start gdb
- choose opentrack PID as **attach 1234** where 1234 is the PID
- send me output of "thread apply all bt" as before

exulant, 2015-12-17:
That worked, thank you. I already sent the backlog.

As a disclaimer, for me the PID for opentrack.exe didn't show in the Taskmanager but in the Resource Manager, just as a reference in case anybody else is looking for it.

sthalik, 2015-12-17:
@exulant @aka2k check this build instead, it should solve the problem fully with no lag spikes.

Build - https://db.tt/XEW3iTZR

sthalik, 2015-12-17:
Please confirm so we can do p5. Thanks!

aka2k, 2015-12-17:
@sthalik it's working great now, man. Thanks a lot for fixing it up. :)

sthalik, 2015-12-17:
Thanks for confirmation!

closed 2015-12-17 by sthalik