opentrack/opentrack #280: UDP sender reading Sensor Data app differently

Akira-28, 2015-12-18:
I read an earlier solved issue indicating that Opentrack should be able to read UDP data from iPhone apps like Sensor Data. I was able to get the free app Sensor Stream to register with Opentrack(although badly), but Sensor Data appears to no longer be recognized by the OT UDP tracker.

Sensor Data app is still able to connect to FreePIE, in fact it seems to be the goto app for iOS connecting to FreePIE. 

Now for the strange part. I was using the previous version of Opentrack, and would try to connect with Sensor Data app, and while it wouldn't change any tracker data readouts, the white square outline behind the octopus would "shimmer" slightly as I moved the phone. And it would only do this if I moved the phone, put it down, nothing. Never registered with the tracker. I just downloaded the new version and the square is gone, but that's an odd piece to this.

sthalik, 2015-12-18:
There were no recent changes to the UDP tracker.

Whether the transmission works, depends only on whether the app speaks the same protocol as the UDP tracker.

Akira-28, 2015-12-19:
I'm referring to https://github.com/opentrack/opentrack/issues/138, for reference. I purchased 3 different iphone apps that claim to be able to stream Accel and gyro data over UDP to specific IPs and ports. I was able to configure 2 of them to do this (the third only streamed over tcp/ip), although only one of them actually appeared to actually connect over UDP to move the cursor. 

I thought the person referred to in the above ticket might have been successful in connecting that app via UDP. Unfortunately, after downloading a number of different versions to test, Sensor Data app's UDP function doesn't appear to connect with Opentrack. I was able to get it to work with FreePie, using a script found on their forum. However, I did connect Sensor Stream to Opentrack, but the output was so erratic, I'm guessing there's some issue there too.

Is there a special function of your UDP receiver that I might be able to relay to the support folks over at Sensor Data? Maybe they can make a change their next update.

sthalik, 2015-12-19:
It expects X, Y, Z, yaw, pitch, roll as double-floats. There's nothing else to it.

Akira-28, 2015-12-19:
double floats? you'll have to forgive me. I asked the support folks for the app if they would consider putting double floating point in their UDP so it could communicate. Is there a better way I could put it in a statement for what they need to do to have their accel info and UDP sender link to your UDP receiver. I can tell them to make sure it has double-floats...is that double floating point?

sthalik, 2015-12-19:
Yes, these are double-floating-point values, 64 bit size.

Keep in mind that the opentrack UDP receiver only operates on final yaw/pitch/roll values, it doesn't integrate raw sensor data.

You might want to try this software: https://github.com/gagagu/VR-Streamer-Windows-Server @gagagu was active on this wiki, maybe he can make a simplified version for head tracking only, without VR functionality

Akira-28, 2015-12-19:
Ah yes, I've emailed him in the past over his app. I'll forward this to him to see what he thinks. Unfortunately my slow broadband is too much for the VR+tracking streaming, which is why I've had to split everything up over various programs. Yours has worked wonderfully with LED tracking btw. But I've been trying my hand at Accelerometer based.  

 Thanks for your suggestions

closed 2015-12-20 by Akira-28