opentrack/opentrack #281: Minor Rotational Spiking

Exiffa, 2015-12-22:
Hello, I've recently setup a point tracking clip and have been configuring opentrack to work with it. It mostly works fine and is very smooth compared to my previous Aruco tracker. I am having a small, slightly infuriating problem though.

Every couple of seconds at seemingly random intervals either the yaw or pitch axis will spike a small amount and then reset to where is was previously looking. It's as if someone has nudged my head.

It's probably a setup problem on my end but i've been messing with all the settings with several days to no effect. The "twitch" even appears when rotational sensitivity and rotational dead zone are all the way to the right in the accela filter.

I'm running 2.3 rc21p6 and have been trying to solve this problem since rc21p1 (I didn't own a point tracker before that point).

sthalik, 2015-12-23:
Check if your model dimensions are correct, they need to be very accurate.

Is your camera PS3 Eye, and what's your gain and exposure?

Exiffa, 2015-12-23:
I'm using the "delan clip" which should be the default dimensions of 30 40 70 80.

I'm using a PS3 Eye with the ir filter removed. The gain is 0 and the exposure is set to the LAST "E" in exposure. I'm sorry I don't know how to get the decimal value.

I am tilting the clip up slightly so that it points more squarely at the camera and remove points merging.

It appears as if the problem is more prevalent when looking straight ahead but does still exist slightly (but not as pronounced)  when looking left or right

sniperwolfpk5, 2015-12-23:
@sthalik did you change something  in PT. I no more use automatic threshold because it don't detect my points. The slider has no effect. I uncheck automatic threshold and set slider manually. The issue is in version p5. It was working fine in rc21.

sthalik, 2015-12-28:
@sniperwolfpk5 your points aren't much brighter than the background.

There was other cause for auto threshold slowdown. I'm increasing bucket size again. It should detect fine now.

sthalik, 2015-12-28:
opentrack-rc21p7 released.

sthalik, 2015-12-28:
@Exiffa couple things.

- what are the raw values of your neutral clip pose?
- do you have dynamic pose resolution enabled?
- what games have you tried?

Exiffa, 2015-12-28:
@sthalik 

• My values are 
yaw 13
pitch -11
roll -4
tx 18
ty 6
tz 63

• I did have the dynamic pose resolution disabled because it was reseting my centre pose when two of my points merged. I have since solved the merging problem by attaching the clip with velcro, allowing me to tilt it towards the camera. I just tried it again with dynamic pose enabled and the problem still persists.

• I have tried it on both DCS World and Elite Dangerous and the spiking is present in both. The spiking is large in elite dangerous due to it needed more sensitive curves.

I just noticed when I was testing that the spiking only seems to occur when looking straight ahead. When I look up or to the sides and keep my view there i don't experience spiking. I will have to test this later by centring my pose while i'm looking upwards and see what happens.

sthalik, 2016-01-07:
@Exiffa send me a video from the raw camera feed. Play that video yourself through a virtual webcam first to see if it reproduces the issue. Please compress the video with low quality in whatever format, don't send uncompressed footage.

sthalik, 2016-01-07:
You can use ManyCam software for virtual webcam functionality or any other alternative.

sthalik, 2016-05-10:
I'm closing, presumed fixed. You can reopen.

closed 2016-05-10 by sthalik