opentrack/opentrack #282: opentrack crashing with pixel size set to 1.0 and low treshold

MathijsG, 2015-12-23:
When I have my settings like this all is good:

[image]

But when I reduce the min pixel size to 1.0 with the arrow application is crashing immediately.

petsild, 2015-12-23:
You've got enough absurd values of all parameters!
The PS3 Eye can handle at a resolution of 640x480 max 75Hz. The point size of the IR leds corresponding to the used lens, I have both variants of the PS3, so I carried out the comparison of the cameras.

- old the lens without the Ir filter 8 – 12 px
- new lens with the IR filter 3,50 – 4,20 px

Values are measured at the 70% treshold using the leds Harvatek HE3-290AC, at a distance of 60cm from the clip. If you've got a point size somewhere around 1px, the problem will be with the power of the dark filter in front of the lens, or inappropriate to the ir led on the clip.

MathijsG, 2015-12-23:
The FPS set to 120 is to easily test difference between 75 and 120 when I set resolution to 320 * 240.
I has no downside keeping this to 120, because it will serve the highest FPS it can get at selected resolution anyway.

I just used a single piece of floppy diskette and put it within the PS3 Webcam behind the lense. I don't necessarily have too few size of the pixels, but the problem is that sometimes Opentrack doesn't see the dots, while they are white enough to mee. It seems that increasing the threshold instead of lowering this, fixes this issue.

sthalik, 2015-12-24:
@petsild even if the parameters are absurd the software must not ever crash.

@MathijsG it's likely that there are too many points extracted. Compare this build https://db.tt/iwxRRNhV

sniperwolfpk5, 2015-12-25:
What are the prefered resolutions for ps3 eye 640x480 or  320 X 240 ?

sthalik, 2015-12-25:
You mean FPS? 75 for 640x480. Don't use 320x240 at all due to decreased precision.

MathijsG, 2015-12-25:
It depends on preference. Some prefer the higher speeds of 120 fps, but I
get indeed decreased precision and sometimes dots falling over each other.
Just play with it and try out what you like best.

Groeten,

Mathijs Groothuis

sthalik, 2015-12-25:
There's no preference to it. Speeds higher than 30 Hz are placebo. Shaking my head to make motion blur noticeable enough just makes my head and eyes hurt.

@MathijsG can you reproduce the crash in the test build?

MathijsG, 2015-12-25:
Would even lower it to 30 fps lower the amount of cpu being used? Will try
test build Sunday, celebrating Christmas now.

sthalik, 2015-12-25:
It would lower CPU usage, but in recent builds we're already as low as we can get while the software is minimized. The more significant bit is altered exposure with different framerates.

MathijsG, 2015-12-26:
Specific error didn't occur in test build here. Didn't try for hours, but no crashes with the treshold and pixel size combination. Well done again! :-)

sthalik, 2015-12-28:
Thanks! Marking as fixed.

closed 2015-12-28 by sthalik