opentrack/opentrack #283: Switch assymetric curves

MathijsG, 2015-12-23:
In Euro truck simulator 2 I made a nice specific curve to look out of the window on the left. After I made the curve I realised after clicking the 'assymetric mapping below' that I made the curve on the wrong graph, and that I want to switch it.

I want two report two things:

1. It's unclear to me which is left and which is right, until you actually try.
2. A feature to swap the asymmetric curves.

sthalik, 2015-12-24:
I don't see a reason for left and right asymmetry to be useful at all. It's included only for symmetry (no pun intended) with the rest of the splines.

Swapping splines is barely ever useful. It's more of clutter in the UI so I won't add it.

MathijsG, 2015-12-24:
But how can one see now which axis is which? Only with pitch you have the
logical top and bottom layout, but how does someone know which one is for
left and wich for right with yaw and roll?

Groeten,

Mathijs Groothuis

sthalik, 2015-12-24:
Like I said needing asymmetric yaw means something is wrong with the setup. Probably the camera rotation values. It's included only for completeness, as you see all tabs have asymmetric spline option despite it being useless for more than half of the tabs total.

sthalik, 2015-12-24:
As for 'swap splines' option, it's barely ever usable. Such functionality is just clutter. I've taken strongly against the "and the kitchen sink" approach.

closed 2015-12-24 by sthalik

MathijsG, 2015-12-24:
I think there are many use cases where the in-game viewpoint isn't
symmetric either. For example in euro truck simulator where you sit on the
left chair where you can look out of the window almost 180 degrees, but to
the right you can only move about 90 degrees. Or some people want the pitch
up to be a bigger range than looking down, because quite often you can only
see a few instruments down and nothing more. I find those asymmetric
functions very useful and couldn't use the software to the full extent
without them. My point is that currently it's quite unclear which
assymetric graph stands for which side.

But its okay if we disagree :-) but don't narrow the use cases down, many
games have asymmetric things.

Groeten,

Mathijs Groothuis

sthalik, 2015-12-24:
Now I'm aware of the use-case, yes.

Putting labels on each spline editor is a pain. They must fit in the graph's border and can overlap with other labels, i.e. number marks.

My point is that some things might not be totally clear at all times, but additing more complexity to make it all easier makes for a cluttered interface, and an intimidating amount of options.

I can't dispute the use-case existing, but the direction I'm taking with the software is to remove stuff, rather than add them at slightest pretense.

Oh, and did the test build fix crashes for you with lowest threshold and lowest point size?

MathijsG, 2015-12-26:
Will try that tonight :). And I understand the small point between clutter and userfriendliness. I only hope you won't remove functionality which often is used but maybe somtimes out of the scope of the main user :).