opentrack/opentrack #284: Open Track - Win32:Malware-gen

TheWarHog, 2015-12-25:
I just thought I ought to make you all ware, but as I attempted to download the opentrack-2.3 rc21p6 today my Malwarebytes has flagged up the download from both this site and from the Track Hat website as containing Win32:Malware-gen which is a high threat Trojan virus.

HornetMaX, 2015-12-25:
Checking both the .exe installer and the .zip on virustotal.com gives 1 and 2 positives (respectively) but  not from the same AV programs (and not by Malwarebytes). Sounds like a false positive.

For sthalik: I recently had a false positive issue with Qt 5.5 libs (ClamAV).

sthalik, 2015-12-28:
There's nothing I can do about Qt libraries being flagged. There are few `contrib/` files flagged on virustotal. They're useful regardless, and I'd like not to remove them just because of a faulty antivirus.

closed 2015-12-28 by sthalik