opentrack/opentrack #285: Crash when starting Point Tracker 1.1 - rc21p6

TehSamsquantch, 2015-12-26:
Hi,

I'm getting a crash immediately after launching rc21p6. 

It crashes shortly after the camera preview is drawn. The image is static.

I was not getting this crash on rc20p1. Let me know if I can provide any more info.

 Problem Event Name:	APPCRASH
  Application Name:	opentrack.exe
  Application Version:	0.0.0.0
  Application Timestamp:	5675b884
  Fault Module Name:	StackHash_0a9e
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	00000000
  Exception Code:	c0000005
  Exception Offset:	00000006
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	0a9e
  Additional Information 2:	0a9e372d3b4ad19135b953a78882e789
  Additional Information 3:	0a9e
  Additional Information 4:	0a9e372d3b4ad19135b953a78882e789

sthalik, 2015-12-28:
What's your CPU?

I think LTO broke this, but let's try the other possibilities first.

sthalik, 2015-12-28:
We still need a backtrace however. Follow both of these instructions. The latter updates the former.

https://github.com/opentrack/opentrack/issues/279#issuecomment-165391434
https://github.com/opentrack/opentrack/issues/279#issuecomment-165454561

TehSamsquantch, 2015-12-28:
>What's your CPU?
>I think LTO broke this, but let's try the other possibilities first.

AMD FX-8350

>We still need a backtrace however. Follow both of these instructions. The latter updates the former.

Okay, I'll check these out!

sthalik, 2016-02-09:
@TehSamsquantch the binaries are built differently not (in particular LTO got disabled). See if it changes anything.

sthalik, 2016-06-16:
No response.

closed 2016-06-16 by sthalik