opentrack/opentrack #286: Linux build problem options.hpp

runningman84, 2015-12-26:
I have fetched the latest git sources which do not build anymore:
```
user@aladin:/usr/local/src/opentrack$ cmake .
-- project tracker-aruco
-- project tracker-freepie-udp
-- project tracker-hatire
-- project tracker-ht
-- project tracker-hydra
-- project tracker-joystick
-- project tracker-pt
-- project tracker-rift-025
-- project tracker-rift-042
-- project tracker-rift-080
-- project tracker-rs
-- project tracker-udp
-- project proto-fg
-- project proto-fsuipc
-- project proto-ft
-- project proto-ftn
-- project proto-libevdev
-- project proto-mouse
-- project proto-sc
-- project proto-vjoy
-- project proto-wine
-- project filter-accela
-- project filter-ewma2
-- project filter-kalman
-- project opentrack-compat
-- project opentrack
-- project gui
-- project x-plane-plugin
-- project csv
-- project pose-widget
-- project spline-widget
-- project qxt-mini
-- project macosx
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/opentrack
user@aladin:/usr/local/src/opentrack$ make
[  0%] Built target opentrack-version
[  0%] Generating moc_moc_moc_moc_moc_moc_moc_qplatformnativeinterface.cpp
[  0%] Generating moc_moc_moc_moc_moc_moc_moc_qxtglobal.cpp
[  0%] Generating moc_moc_moc_moc_moc_moc_moc_qxtglobalshortcut.cpp
[  0%] Generating moc_moc_moc_moc_moc_moc_moc_qxtglobalshortcut_p.cpp
[  0%] Generating moc_moc_moc_moc_moc_moc_qxtglobalshortcut_mac.cpp
[  1%] Generating moc_moc_moc_moc_moc_moc_qxtglobalshortcut_x11.cpp
Scanning dependencies of target opentrack-qxt-mini
[  1%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_moc_moc_moc_moc_moc_moc_qxtglobal.cpp.o
[  1%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_moc_moc_moc_moc_moc_moc_qxtglobalshortcut.cpp.o                            
[  1%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_moc_moc_moc_moc_moc_moc_qxtglobalshortcut_p.cpp.o                          
[  1%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_moc_moc_moc_moc_moc_qxtglobalshortcut_mac.cpp.o                            
[  1%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_moc_moc_moc_moc_moc_qxtglobalshortcut_x11.cpp.o                            
[  2%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_moc_moc_moc_moc_moc_moc_qplatformnativeinterface.cpp.o                     
Linking CXX shared library libopentrack-qxt-mini.so
[ 12%] Built target opentrack-qxt-mini
[ 12%] Generating moc_moc_moc_moc_moc_moc_compat-import.cpp
[ 12%] Generating moc_moc_moc_moc_moc_moc_export.cpp
[ 12%] Generating moc_moc_moc_moc_moc_moc_import.cpp
[ 12%] Generating moc_moc_moc_moc_moc_moc_process-list.cpp
[ 12%] Generating moc_moc_moc_moc_moc_moc_sleep.cpp
[ 12%] Generating moc_moc_moc_moc_moc_moc_timer.cpp
[ 13%] Generating moc_moc_moc_moc_moc_moc_camera-names.cpp
[ 13%] Generating moc_moc_moc_moc_moc_moc_options.cpp
[ 13%] Generating moc_moc_moc_moc_moc_moc_shm.cpp
[ 13%] Generating moc_moc_moc_moc_moc_moc_qcopyable-mutex.cpp
[ 13%] Generating moc_moc_moc_moc_moc_moc_compat-export.cpp
Scanning dependencies of target opentrack-compat
[ 13%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_export.cpp.o
[ 13%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_import.cpp.o                                     
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_process-list.cpp.o                               
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_sleep.cpp.o                                      
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_timer.cpp.o                                      
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_camera-names.cpp.o                               
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_options.cpp.o                                    
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_shm.cpp.o
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_qcopyable-mutex.cpp.o
[ 14%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_compat-export.cpp.o
[ 15%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_moc_moc_moc_moc_moc_compat-import.cpp.o
Linking CXX shared library libopentrack-compat.so
[ 31%] Built target opentrack-compat
[ 31%] Generating moc_moc_moc_moc_moc_moc_functionconfig.cpp
[ 31%] Generating moc_moc_moc_moc_moc_moc_qfunctionconfigurator.cpp
Scanning dependencies of target opentrack-spline-widget
[ 31%] Building CXX object spline-widget/CMakeFiles/opentrack-spline-widget.dir/moc_moc_moc_moc_moc_moc_qfunctionconfigurator.cpp.o
[ 31%] Building CXX object spline-widget/CMakeFiles/opentrack-spline-widget.dir/moc_moc_moc_moc_moc_moc_functionconfig.cpp.o
Linking CXX shared library libopentrack-spline-widget.so
[ 34%] Built target opentrack-spline-widget
[ 34%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_camera-names.cpp
[ 34%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_main-settings.cpp
[ 34%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_mappings.cpp
[ 34%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_opencv-camera-dialog.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_keybinding-worker.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_win32-joystick.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_options.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_plugin-api.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_plugin-support.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_simple-mat.cpp
[ 35%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_state.cpp
[ 36%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_work.cpp
[ 36%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_selected-libraries.cpp
[ 36%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_shortcuts.cpp
[ 36%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_tracker.cpp
[ 36%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_win32-shortcuts.cpp
Scanning dependencies of target opentrack-api
[ 36%] Building CXX object opentrack/CMakeFiles/opentrack-api.dir/moc_options.cpp.o
/usr/local/src/opentrack/opentrack/moc_options.cpp:9:23: fatal error: options.hpp: No such file or directory
 #include "options.hpp"
                       ^
compilation terminated.
make[2]: *** [opentrack/CMakeFiles/opentrack-api.dir/moc_options.cpp.o] Error 1
make[1]: *** [opentrack/CMakeFiles/opentrack-api.dir/all] Error 2
make: *** [all] Error 2
user@aladin:/usr/local/src/opentrack$ find . -name options.hpp
./opentrack-compat/options.hpp
```

Is this a cmake problem? The file options.hpp is in the opentrack-compat folder...

sthalik, 2015-12-26:
> [ 36%] Generating moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_moc_win32-shortcuts.cpp

Something's up with your build environment.

runningman84, 2015-12-27:
Ok I started make clean in order to cleanup my build directory, now I get the same error but the build looks fine
```
[  1%] Building CXX object CMakeFiles/opentrack-version.dir/version.cc.o
Linking CXX static library libopentrack-version.a
[  1%] Built target opentrack-version
[  1%] Generating moc_qxtglobal.cpp
[  2%] Generating moc_qxtglobalshortcut.cpp
[  2%] Generating moc_qxtglobalshortcut_mac.cpp
[  3%] Generating moc_qxtglobalshortcut_x11.cpp
[  3%] Generating moc_qplatformnativeinterface.cpp
[  4%] Generating moc_qxtglobalshortcut_p.cpp
Scanning dependencies of target opentrack-qxt-mini
[  5%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut.cpp.o
[  5%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_mac.cpp.o                                                    
[  6%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/qxtglobalshortcut_x11.cpp.o                                                    
[  6%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut.cpp.o                                                    
[  7%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut_mac.cpp.o                                                
[  7%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut_x11.cpp.o                                                
[  8%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qplatformnativeinterface.cpp.o                                             
[  8%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobalshortcut_p.cpp.o                                                  
[  9%] Building CXX object qxt-mini/CMakeFiles/opentrack-qxt-mini.dir/moc_qxtglobal.cpp.o                                                            
Linking CXX shared library libopentrack-qxt-mini.so
[  9%] Built target opentrack-qxt-mini
[  9%] Generating moc_timer.cpp
[ 10%] Generating moc_camera-names.cpp
[ 10%] Generating moc_options.cpp
[ 10%] Generating moc_shm.cpp
[ 11%] Generating moc_qcopyable-mutex.cpp
[ 11%] Generating moc_compat-export.cpp
[ 12%] Generating moc_compat-import.cpp
[ 12%] Generating moc_export.cpp
[ 13%] Generating moc_import.cpp
[ 13%] Generating moc_process-list.cpp
[ 14%] Generating moc_sleep.cpp
Scanning dependencies of target opentrack-compat
[ 15%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/camera-names.cpp.o
[ 15%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/options.cpp.o                                                            
[ 16%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/shm.cpp.o                                                                
[ 16%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_camera-names.cpp.o                                                   
[ 17%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_options.cpp.o                                                        
[ 17%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_shm.cpp.o                                                            
[ 18%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_qcopyable-mutex.cpp.o                                                
[ 18%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_compat-export.cpp.o                                                  
[ 19%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_compat-import.cpp.o                                                  
[ 19%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_export.cpp.o                                                         
[ 20%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_import.cpp.o                                                         
[ 20%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_process-list.cpp.o                                                   
[ 21%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_sleep.cpp.o
[ 21%] Building CXX object opentrack-compat/CMakeFiles/opentrack-compat.dir/moc_timer.cpp.o
Linking CXX shared library libopentrack-compat.so
[ 21%] Built target opentrack-compat
[ 22%] Generating moc_qfunctionconfigurator.cpp
[ 22%] Generating moc_functionconfig.cpp
Scanning dependencies of target opentrack-spline-widget
[ 22%] Building CXX object spline-widget/CMakeFiles/opentrack-spline-widget.dir/functionconfig.cpp.o
[ 23%] Building CXX object spline-widget/CMakeFiles/opentrack-spline-widget.dir/qfunctionconfigurator.cpp.o
[ 23%] Building CXX object spline-widget/CMakeFiles/opentrack-spline-widget.dir/moc_functionconfig.cpp.o
[ 23%] Building CXX object spline-widget/CMakeFiles/opentrack-spline-widget.dir/moc_qfunctionconfigurator.cpp.o
Linking CXX shared library libopentrack-spline-widget.so
[ 23%] Built target opentrack-spline-widget
[ 23%] Generating moc_work.cpp
[ 23%] Generating moc_selected-libraries.cpp
[ 24%] Generating moc_moc_camera-names.cpp
[ 24%] Generating moc_moc_options.cpp
[ 25%] Generating moc_keybinding-worker.cpp
[ 25%] Generating moc_shortcuts.cpp
[ 26%] Generating moc_tracker.cpp
[ 26%] Generating moc_win32-joystick.cpp
[ 26%] Generating moc_win32-shortcuts.cpp
[... cut ...]
[ 31%] Building CXX object opentrack/CMakeFiles/opentrack-api.dir/moc_camera-names.cpp.o
[ 32%] Building CXX object opentrack/CMakeFiles/opentrack-api.dir/moc_options.cpp.o
/usr/local/src/opentrack/opentrack/moc_options.cpp:9:23: fatal error: options.hpp: No such file or directory
 #include "options.hpp"
                       ^
compilation terminated.
make[2]: *** [opentrack/CMakeFiles/opentrack-api.dir/moc_options.cpp.o] Error 1
make[1]: *** [opentrack/CMakeFiles/opentrack-api.dir/all] Error 2
make: *** [all] Error 2
```

sthalik, 2015-12-27:
There's no `options.cpp` in `opentrack/`, you haven't cleaned your build directory completely. Try deleting `CMakeFiles/`

sthalik, 2016-01-07:
You also might have used the source directory as a build directory. Use a subdirectory.

Also, sync your tree, then run

`git reset --hard`
`git clean -fd`

I'm closing this due to lack of response. Issues can be still commented upon while closed, and reopened.

closed 2016-01-07 by sthalik