opentrack/opentrack #287: Pointtracking issues 3 point cap configuration

Akira-28, 2015-12-28:
Last time I was trying wireless accelerometer tracking. This time I am trying old fashioned camera based head tracking. 

I finally got my 3 LEDs which is the preferred setup for the PT plugin for OpenTrack  I did not get an official IRclip, I have a VR HMD that I have arrayed 3 leds on the front. The plan is to wire them in with battery switch etc.
http://i.imgur.com/3haqxVj.png

Now I have been able to get this to track relatively well with FreeTrack, but for some reason it just will not calibrate properly. The red cross, which I assume means midpoint for the model, is always flying 200 mm above or below my position. And I have tried manually configuring the model position settings and that doesn't make much difference. My Yaw and Pitch are all off, and I have tried moving the curve guides using the information in the Quick Start page, still no improvement. 
http://i.imgur.com/Pwrj0Sf.jpg

I am doing something wrong, but have no idea what.

sthalik, 2015-12-28:
You need to input accurate model dimensions first.

sthalik, 2015-12-28:
As for model calibration follow the procedure from the wiki. It has its separate page.

The quick start page isn't done yet.

Akira-28, 2015-12-28:
Ok I did try to input proper dimensions. I have one light on the top of the HMD, and two lights on the sides. I first tried a custom model, although I later tried with the 3pt hat model. I broke out the ruler and starting from the middle light measured the millimeter distances between the points, x y z axis. Like 10 cm over and 5 cm down, 2 foward, etc. I also tried swapping the dimensions in case I had the thing backwards or upside down. 

I also tried different dimensions in the hat array, nothing seemed to catch on. I can get the tracker to see all the points, but the dimensions and calibrating I'm stuck at.

sthalik, 2015-12-28:
The hat/clip pages in PT configuration have symbology as to which distances to measure.

The custom page works fine, measure distances relative to one of the points.

Akira-28, 2015-12-28:
That's where I started, using millimeters, and distance of the dimensions. I input them into the custom, then tried the calibration instructions listed in the wiki. Lifted my head, turned left right, put it down further to bottom range, calibrated. Then the red X is off in the -3XX territory in the model listing, always some wild number. And I can change it manually but not really knowing what those numbers mean, I just change them so the red X is in the middle of the model on my face.

But then the tracking never really works correctly. It may move to the left or right, semi- accurately, but then the pitch is also tied to the yaw for some reason so it only goes diagonally. Then I try to recalibrate and I get different results all over again. Trying to restrict ranges with curves has only given me limited progress, still can't configure the ranges. 

The workaround with using OpenTrack's pointtrack plugin as a bootstrap for Freetrack still works, since FT tracking I can get to work with the LED model inputs I have input, but I am blocked at Opentrack and point tracking. It sees the lights, if I could somehow manually set the model position maybe?

sthalik, 2015-12-28:
Leave calibration as-is. Calibration depends on pose estimation working, and as long as you can't get that to work, the red 'X' won't point where it should. Leave calibration at zero for now so that the red cross is at one of the points.

Show me your point 1 and point 2 coordinates please, not the cap values but grab absolute point positions from the .ini file. The whole "tracker-pt" section will suffice.

Also drop me one image of the points being tracked.

gyrovague, 2016-02-21:
FWIW, I made a cap tracker a while ago, and it works great with Windows 7 and PS3 Eye. I remember I had some trouble initially figuring out the custom model params, but I got it right after a while.
http://imgur.com/gallery/4fpfP

Here is the section from my default.ini file, perhaps this will help @Akira-28 :

> [tracker-pt]
active-model-panel=2
automatic-threshold=false
camera-fov=56
camera-fps=187
camera-name=PS3Eye Camera
camera-res-height=240
camera-res-width=320
cap-x=40
cap-y=60
cap-z=100
clip-by=70
clip-bz=80
clip-ty=40
clip-tz=30
dynamic-pose-resolution=true
init-phase-timeout=500
m_01-x=42
m_01-y=-80
m_01-z=-55
m_02-x=-42
m_02-y=-80
m_02-z=-55
max-point-size=7
min-point-size=0
model-centroid-x=0
model-centroid-y=-130
model-centroid-z=0
threshold-primary=11

Some useful features for opentrack might be:
-  Have some more help and pictures in the custom model tab
-  Display a preview of the custom model based on the user inputs, the user can then compare this to his actual model expectations and more easily fix input errors. This preview can perhaps just show side and front views, similar to the views in the clip and hat tabs
- Support multiple custom models instead of only one
I wouldn't mind implementing these features if somebody can point me to the simplest steps to get an opentrack build going locally on my machine.

closed 2016-06-21 by sthalik