opentrack/opentrack #289: Cap model sometimes loses calibration point and inverts

MathijsG, 2016-01-04:
I recently soldered a Cap model, so I could test both types of optical tracking, clip and cap. I like most of the parts more of the cap: the axis are less interconnected, so when you zoom you don't also pitch up, and you can make nice rolls without moving to the left or right. 

However I also noticed some downsides, which I don't know are maybe something in the software or an inherent problem of the design:

When I sometimes look away from the screen, the red cursor is moving to a different position and the orientation is totally messed up. Recentering doesn't work then, only actually stopping the tracking again. I do like the recent keymap options for toggling the tracking itself, because they could come in handy when this happens.

Maybe it's a bit hard to understand, but here are some pictures:

Before:

[image]

After:

[image]

It acts somewhat inverted somehow.

I just discovered that I also can reset its behavior again by moving too far up so the two bottom leds will be above the top led, and when I then move them down again they are normal.

Will make a video of it when I have the time.

sthalik, 2016-01-07:
Do you have dynamic pose resolution set, and its timer set to at least a second or so?

MathijsG, 2016-01-07:
[image]

I should set it to 1000MS? Will try. What exactly does it do anyway? Couldn't really figure out when I googled it earlier.

MathijsG, 2016-01-11:
I still have with 1000MS that when I shake wildly, or after a while the octopus isn't really rotating around the middle anymore, but it looks like it is going in a circle around. Should make a video of it :-P.

sthalik, 2016-01-15:
@MathijsG "dynamic pose" helps remember which point is which. With it off, no information is preserved.

Look for any anomalies like a point not visible after rotating a lot, or points merging.

MathijsG, 2016-01-15:
Will check further. Why not keep this to for example 10.000MS? Does that cost more CPU or RAM or something?

sniperwolfpk5, 2016-01-15:
Is it better to turn on dynamic pose? any benifits using this option.

sthalik, 2016-01-15:
@MathijsG it's meant not to reset on brief failure

@sniperwolfpk5 yes

sthalik, 2016-02-09:
> Does that cost more CPU or RAM or something?

It doesn't cost any more cpu time or memory no matter the setting.

If you set it to a really high value it may remember an incorrect position for too long. Set it to a typical amount your LEDs may be not recognizable, with a margin of error.

MathijsG, 2016-03-13:
Could it be that this is fixed in the latest build (P10)? How long I look behind my doesn't matter anymore, even when I look behind me 10+ seconds, when I look back the orientation is normal, instead of weird.

THANKS! :-)

sthalik, 2016-03-15:
@MathijsG this isn't a bug. Dynamic pose timeout happens and the software can no longer distinguish points from memory. Return to center pose and cover one of the LEDs, or toggle tracking twice.

MathijsG, 2016-03-15:
But when I set the Dynamic Pose timeout for example to 9999 MS (9,9 seconds). The program already loses all orientation while looking away for example for just 1,5 second. This is pretty annoying when you are looking away for a brief moment.

sthalik, 2016-03-16:
It remembers exact on-screen positions and they don't unambiguously correspond to LED lights when your model return to the view.

MathijsG, 2016-03-18:
But I think it's very common to be able to look away from the screen for a brief moment and resume looking. With a clip this is no problem here.

When I disable Dynamic Pose Resolution I have the problem that my yaw range is very limited, when the top led is vertically aligned with one of the outer leds it stops moving, while with Dynamic Pose enabled it can go further. Will make a movie if this explanation is too hard to follow :-P.

sthalik, 2016-03-18:
I understand your explanation. But there seems to be no good way to take the problem you're describing into account.

MathijsG, 2016-03-18:
Okay, I understand. But I don't know why it sometimes flips back normally on certain settings. I will still test further. Don't know if other people with a Cap like the official trackhat have this problem?

sthalik, 2016-03-18:
What flips back again?

MathijsG, 2016-03-18:
Sorry I mean when I get back in the webcam detection area, most of the times the octopus is being backwards and everything is inverted, but it sometimes DOES work normally. That's why I earlier thought this issue was fixed.

sthalik, 2016-03-18:
Dynamic pose remembers the last point positions where they were visible. If two points (after they become visible again) is closest to a single one of the remembered positions, dynamic pose gets reset.

It may also give a bad result if wrong points are closest to the remembered points.

MathijsG, 2016-03-18:
Yeah I think that is the problem somewhere. A clip is mostly 3 points vertically aligned, which can't really be mistaken for another pose, while a cap is some triangle form, and when you turn too far while moving out of the screen, the top led can be mistaken for a led from one of the sides, rendering the octopus sideways.

sthalik, 2016-05-06:
I don't see any technical solution here. You can reopen with a technical solution.

closed 2016-05-06 by sthalik

MathijsG, 2016-05-12:
Update: When I set the dynamic pose time out to extreme low values, like 1 ms it's flipping back normally again after going back in screen.

sthalik, 2016-05-12:
@MathijsG the build I linked in the other issue has a "reset tracker" shortcut. There's no better solution.

Effectively disabling dynamic pose is a bad idea.

sthalik, 2016-05-12:
You can cover a single LED for the timeout's duration and it has the same effect as the reset function.