opentrack/opentrack #291: [feature] "while-keydown" option for global shortcuts

McDiod, 2016-01-05:
Adding the option to the *Toggle* and *Zero* shortcuts to activate them on keydown and deactivate them on key-release would help immensely in shooters. 

Specifically in Arma, for long range, zoomed in shots, you hold a key to hold your breath. If you could enable the *Zero* function while the key is pressed, instead of toggling it, it would make taking accurate shots way more seamless.

sniperwolfpk5, 2016-01-05:
I think there is already this option start and stop toggle. If that is not working then  this should be another option. Stop Tracking while key pressed some thing like that.

MathijsG, 2016-01-05:
His point is that he only wants to stop tracking while pressing the button. So the toggle is not suitable for him. Maybe some option for hold instead of toggle?

sthalik, 2016-01-07:
Several points here.

The zero option was introduced specifically for Elite Dangerous, where menus are unusable while head tracking is active. Navigating the menus takes time and having to hold a button while at it doesn't make sense.

In case of "toggle", it's true that in Arma FOV is way smaller when zoomed in with a scope. Small movements then correspond to way larger shifts in view. This can be a problem in two ways:

Either you've misconfigured the filter to remove too little noise or way too much causing sudden jumps, or you naturally move your head during scoping.

In case the software is configured right, I'll consider it. From my experience small yaw/pitch changes can be corrected fast not to lose a target. But I also understand that humans make spontaneous head movements and suppressing that is bad.

All this hair-splitting is necessary because we're cutting down user-selectable features. The software used to have tons of poorly thought-out features and it was intimidating. The software configuration is still too hard even despite the drastic changes.

PS I really hate how 3D scopes are implemented in Arma. They zoom in the view outside the scope, rather than only the scope itself. It can be trivially implemented[1] without performance degradation. I saw two games with ACOG scopes like that and it was way more immersive.

[1] Save the scope center and on-screen radius for the vertex shader. Pass two world-to-view transforms, zoomed-out and zoomed-in. Choose one of them depending on whether given vertex is within the scope radius after passing through zoomed-out world-to-view transformation.

MathijsG, 2016-01-07:
But I do see a general use case, one I mentioned earlier, and a feature which Freetrack had: 

[image]

Something that reduces the output of all other axes. In many games you see the problem that zooming in on the Z effectively makes the other axes affect too much, you need to be very still with your head, or otherwise you can't aim precisely. I see this in Rise of Flight and many other games. Isn't it possible to make something like this again? When you want to keep the range and precision of your setup but you want to decrease the movement while zooming in, you currently have no way to go. This user even stops his tracking to prevent movement at all.

I understand you don't want featuritis and don't want a broken program which is too hard to use, but some optional solution for zooming would be very welcome here and I know a few other users who would also really like that feature.

McDiod, 2016-01-08:
The software is configured correctly, I believe. I can keep my head still enough to not cause the view to move even when zoomed in. 

Unfortunately, movement is not the problem, but centering is. I right click to use iron sights, then hold right to zoom in. At this point, if my head rotation is only slightly off center, the sights will be on the edge of my monitor. I then have to rotate my head ever so slightly to center the sights. Even if this only takes a second, it's still time that I don't want to have to spend if it could mean my in-game death.

I understand the need to unclutter, but I mean look at all this real estate :)
[image]

sthalik, 2016-01-08:
Okay, I'll implement before 2.3.

commit 3a7011ff (2016-04-29): api, main: implement new keybindings

The following keybindings are now implemented:

- restart tracking. if not started, starts.
- toggle (keep position) while key held
- zero (keep zero) while key held

The latter two options only work for Windows as libqxt doesn't support
them.

Closes #291
closed 2016-04-29 by sthalik