opentrack/opentrack #292: [Feature] Specific zooming options for disabling or decreasing the output of other axes

MathijsG, 2016-01-08:
A feature which Freetrack had:

image

Something that reduces the output of all other axes. In many games you see the problem that zooming in on the Z effectively makes the other axes affect too much, you need to be very still with your head, or otherwise you can't aim precisely. I see this in Rise of Flight and many other games. Isn't it possible to make something like this again? When you want to keep the range and precision of your setup but you want to decrease the movement while zooming in, you currently have no way to go. This user even stops his tracking to prevent movement at all.

I understand you don't want featuritis and don't want a broken program which is too hard to use, but some optional solution for zooming would be very welcome here and I know a few other users who would also really like that feature.

I'm wondering about other people having the same? For me it was a bigger issue with the tracking clip rather than with the tracking hat, but it still is a thing to me.

sthalik, 2016-01-08:
But if you

- yaw left
- zoom in

You'll look at a different place. You zoom in to look at something, not to have the view move more toward center.

We can't scale *deltas* as that makes for miscentered view.

MathijsG, 2016-01-08:
But you want to be that yaw left reduced heavily, because you are zooming in, so you somehow want a different curve or something, or something that kicks in while the Z axis is at a certain level. Otherwise the object where you are aiming at moves right out of the view, because the yaw is too responsive at that current Z level.

sthalik, 2016-01-08:
Still, we need a way to implement it so it works for both cases. In flight sims it's common to look at stuff scanning for targets and zooming. In Arma you need to look at a distant piece of ground to your side as well. In Arma it's necessary to keep scanning the surroundings at all time.

The software can't read the user's mind to figure out what he wants to do at a given time.

Is #291 enough though?

You'll also get better results by increasing exposure of the PS3 Eye so that LEDs are bigger. With automatic exposure light level isn't as significant. Then you can use very sensitive filtering where small movements are less delayed.

MathijsG, 2016-01-08:
I will mess with the increased exposure levels later this weekend. Thanks. In my opinion the user would control the zoom behaviour and thus the behavior of the other axes while zooming, so it would be some optional setting. But I think i'm not the only one here, many people as a workaround, even map zoom to a button instead of a tracking axis because of this. With Rise of Flight I often lose my target with this.

sthalik, 2016-01-08:
Users can control the behavior only with regard to the workings of the code that implements a given feature. I don't know if there's a good solution for both zooming to attack and zooming to scan the surroundings, there might be done. If you have one, please provide it.

MathijsG, 2016-01-08:
If there would be a technical option to limit the other axes when zooming with for example the Z axis the solution to both would be to not enable this solution when you want to check surroundings. Maybe we can implement this trigger with a button. But from my perspective (as biased as it is) I see more in the use case of people zooming in on a target and keeping that zoom steady while still being able to move the axes, rather than the use case of having the same amount of yaw and pitch as being zoomed out, and scanning the horizon on full zoom.

sthalik, 2016-01-08:
See #291

MathijsG, 2016-01-08:
He isn't describing the scanning of the horizon and thus needing axes with regular curves. It looks like he is describing my scenario, where he want to lock and even disables the tracking because the current curves are way too responsive to have a clear shot. This would exactly fit in my use case to lower the response (optionally) of all axes while zoomed in at a certain level with Z.

sthalik, 2016-01-08:
I'd implement a feature without such serious drawbacks as the zoom-affecting-response proposal. This might fit an idiosyncratic use-case but that's not enough.

What are the scenarios where you're having problems with?

MathijsG, 2016-01-08:
I don't really understand why this is such a weird usecase in your opinion, TrackIR has this, Freetrack has it. Because the ability to zoom is quite an essential one in these games that support headtracking.

For me when I zoom in with Rise of Flight, I have way too much yaw and pitch movement when I move a little. Every measure you would take to prevent this, in its current state, would be also in effect when not zooming in. And to me that is a problem.

What if you want fast response with a little deadzone in regular view, but when you zoom in, you want less yaw and pitch and more deadzone so you can just aim slightly?

sthalik, 2016-05-10:
@MathijsG with a correct fov value we have a truly accurate 6DOF pose. We shouldn't need that. If the model is in millimeters then translation is in millimeters also. We shouldn't skew the results as part of the PT tracker.

You should try decreasing the rotation sensitivity and smoothing sliders. If you must, increase the deadzone to no more than .5 degrees. Then you should be able to correct the minor movement fast enough not to impact your play.

Additionally you have "toggle" and "zero" bindable to a held key now. No mouse support yet. Is this more adequate.

MathijsG, 2016-05-10:
For me it's still a use case that I want fast responsiveness, but when I'm zooming in with my head I want to have decreased responsiveness or something with the other axes, so I don't have to hold my head extremely still. But I will play around with the filters again.

sthalik, 2016-05-14:
You can get far lower than 1 deg. for rotation sensitivity. You won't notice a deadzone below 0.5 but it'll greatly improve responsiveness compared to higher sensitivity.

sthalik, 2016-05-14:
As usual I recommend having exposure set to more than the lowest value. Having the LED bigger greatly reduces noise.

sthalik, 2016-05-22:
I don't like the idea since it interferes with geometry so much. Also note you're first asking for this 3 years after the project began.

closed 2016-05-22 by sthalik