opentrack/opentrack #293: Support die translations

MathijsG, 2016-01-09:
In an effort to lower the difficulty localized translations could also be a thing. I'm willing to translate this software to Dutch, my native language. Is it easy to have this supported? What's your opinion about it.

MathijsG, 2016-01-09:
Somehow the title got messed up, using this on my phone. Was talking about supporting localized translations

sthalik, 2016-01-10:
It requires some code changes. I'll look into it only after the software is adequately documented.

MathijsG, 2016-01-10:
I understand, I'll help with the Wiki first.

denysenkok, 2016-09-13:
Russian users will appreciate the ability to translate languages. Maybe We can do previous work? We will prepare what you say.
Thank you.

sthalik, 2016-09-13:
@denysenkok I recommend for now creating Wiki pages with a `[RU] ` prefix. I'll add a note to the main wiki page for these pages once they're done.

For a program to get translation to begin with, some work needs to be done in the code. Right now none of the strings are translatable, and there's no build support for generating translations. I'll get on to that but it's a larger undertaking.

Also please keep in mind that translation will initially be incomplete, leaving the program with both Russian and English text. Also, new functionality will initially only have an English translation.

I'd like for you guys to have a repository where you can push translation data on your own, and be able to test new translation strings without rebuilding opentrack from the source code. BUT the code work needs to be done and it's actually a ton of tiny changes! opentrack is now over 20k lines of code.

The large Sturmovik community definitely deserves support of the opentrack project.

To edit the wiki you only need a Github account, there are no restrictions on our wiki.

sthalik, 2016-09-13:
Another simple idea is to add a link to Russian documentation in the program if the user has Windows localized in Russian. What do you think?

sthalik, 2016-12-01:
We're good. Done!

closed 2016-12-01 by sthalik