opentrack/opentrack #294: [Feature] OSVR as Tracker

trevorirwin, 2016-01-10:
With the introduction of so many new HMDs to the market and the addition of Oculus support to OpenTrack, it would be nice to see OpenTrack integrate support for OSVR, an open source standard that would allow for tracking from a multitude of HMDs. Would OSVR support be a possibility in the near future?

sthalik, 2016-01-10:
It might. Can you provide customer data for OSVR devices? For Rift we already have a plugin so I don't need that.

trevorirwin, 2016-01-10:
What exactly are you looking for?

To be clear, I'm not formally affiliated with the OSVR project. However, I do own an OSVR-compatible HMD and am eager to see the platform be adopted more widely. I can definitely help test any plugins developed and could help with tweaking things, but I don't have enough experience to code anything directly.

sthalik, 2016-01-11:
Hey,

Drop me any info (press release, news, ...) on popularity of the other devices.

I'm and old-timer and I don't follow trends like VR.

trevorirwin, 2016-01-11:
Here is a recent press release from Sensics describing OSVR's industry partners. It's apparently the largest open source VR ecosystem right now:
http://sensics.com/osvr-closes-2015-with-a-total-of-300-industry-partners-now-the-largest-open-source-vr-ecosystem-in-the-industry/

Here are two reviews, one recent and one from about a year ago, describing the Razer HDK, what you might call the "flagship" OSVR HMD, and why it's such a big deal:
http://www.digitaltrends.com/virtual-reality/osvr-hacker-dev-kit-hands-on/
http://time.com/3655575/razer-osvr/

sthalik, 2016-01-11:
I'll do it. I still need someone to check whether the code works.

trevorirwin, 2016-01-11:
I will happily test it for you and provide as much feedback as possible.

Note that right now, on the OSVR side of things, positional tracking is a little buggy so I may not be able to test that aspect effectively until it's updated. However, rotational tracking works very well, so I can certainly test that.

sthalik, 2016-01-11:
Okay. It's gonna be the next feature implemented.

I'll implement position tracking also at first, you can disable it for the time being.

birdyisme, 2016-02-22:
Hi,

Maybe that can help, it exist now a freepie plugins : https://github.com/thomasgauthier/FreePIE-OSVR/releases

Freepie + opentrack work well, but use most CPU.
https://www.reddit.com/r/OSVR/comments/44ox8q/tridef_3d_and_freepie_setup_guide/

I can test and help on windows and linux if needed.

sthalik, 2016-02-25:
@birdyisme thanks. For now it's an issue of my lack of development time due to events out of my control. I'll keep you posted after there's any working build.

sthalik, 2016-11-03:
Are there any more supported controllers than Rift and HTC Vive?

nanospork, 2016-11-03:
Do you mean other headsets that OSVR supports, or hand controllers like the Vive Wands or Oculus Touch?

For headsets, OSVR currently supports: HDK2, Oculus Rift, HTC Vive, the upcoming FOVE VR, and a number of existing and upcoming niche headsets. It will probably support PSVR soon as well, assuming the open source community figures out how to connect the PSVR to PC.

For hand controllers, there are some efforts to get Kinect, WiiMote, and Playstation Move integration, but I don't know much about them. There's also the Nod Backspin.

sthalik, 2016-11-03:
Okay. The SDK source code is separated into multiple repositories. I'm gonna support the official modules (other than Rift and Vive since they have modules), and some third-party modules. It seems Wiimote etc. are missing. Are these third-party modules?

nanospork, 2016-11-03:
You should be able to just support the OSVR tracker - you don't have to support any hardware specifically. That's the nice thing about OSVR, it unifies all the hardware interfaces. I believe you want to create an OSVR "client". This article may be useful: http://resource.osvr.com/docs/OSVR-Core/TopicWritingClientApplication.html

You just need a client that accesses the tracker path /me/head and it will work no matter what hardware is being used.

nanospork, 2016-11-03:
The folks over at http://gitter.im/osvr are extremely helpful if you need some real-time advice.

sthalik, 2016-11-03:
That's great, thank you.

robsonswiss, 2017-06-27:
Hi there

Have you had any more time to look into this feature?

sthalik, 2018-05-22:
No, sorry. Consider implementing it yourself then I'd maintain the finished code for the future.

closed 2018-08-03 by sthalik