opentrack/opentrack #295: OpenTrack 2.3 rc19p2 crash.

Sovereign-Man, 2016-01-10:
After a few seconds of activity it crashes, could this be a result of my attempts to "tidy" my installed programs as I decided to uninstall all versions of Microsoft Visual C++ 2005. I figured I don't need C++ from every year since 2005.
I stil have both x86 and x64 vesrions of 2008,2010,2012,2013 and 2015.

sthalik, 2016-01-10:
Follow these instructions to submit debug information

https://github.com/opentrack/opentrack/issues/279#issuecomment-165391434
https://github.com/opentrack/opentrack/issues/279#issuecomment-165454561

For that you need a debug build. https://db.tt/2ez5XYaQ

sthalik, 2016-01-10:
opentrack doesn't require any MSVC runtimes at all.

Sovereign-Man, 2016-01-10:
Good to know, even so, I did a system restore back to earlier today and it works now :)

closed 2016-01-10 by Sovereign-Man