opentrack/opentrack #296: Stable 2.2 segfault on Linux

helmchenlord, 2016-01-11:
Hi I am running Gentoo AMD64, opentracker compiles fine, but starting it ends with a Segmentation Fault.

gdb) r
Starting program: /usr/local/opentrack 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe8aa4f8c in ?? () from /usr/lib64/qt4/libQtTest.so.4

I would like to use Opentrack with the edtracker to play War Thunder.

T1ckL35, 2016-01-12:
Do you have any instructions on how you compile this project for linux? (I'm using Ubuntu)
I'm a developer but not of C or compiling programs like this from source.

Any help/pointers would be greatly appreciated

sthalik, 2016-01-12:
@helmchenlord show the backtrace, please.

@T1ckL35 Linux is supported only for developer convenience. See FAQ

T1ckL35, 2016-01-13:
@sthalik Thanks for pointing it out. I've managed to build it now anyway, just need to get it recognising the webcam.

helmchenlord, 2016-01-13:
@sthalik Backtrace as follows:

(gdb) bt
#0  0x00007fffe8aa4f8c in ?? () from /usr/lib64/qt4/libQtTest.so.4
#1  0x00007ffff7de9dca in ?? () from /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff7de9ec5 in ?? () from /lib64/ld-linux-x86-64.so.2
#3  0x00007ffff7ddc11a in ?? () from /lib64/ld-linux-x86-64.so.2
#4  0x0000000000000001 in ?? ()
#5  0x00007fffffffde3b in ?? ()
#6  0x0000000000000000 in ?? ()

sthalik, 2016-01-13:
@helmchenlord it doesn't make sense, we don't even use QtTest. Run under valgrind to get a proper backtrace.

helmchenlord, 2016-01-13:
This is how I have built opentrack:
git clone -b stable/2.2 https://github.com/opentrack/opentrack.git
cd opentrack
cmake CMakeLists.txt
make
make install

ls -l /usr/local/opentrack
-rwxr-xr-x 1 root root 612024 Jan 13 13:34 /usr/local/opentrack

Here is the output from valgrind:
valgrind --track-fds=yes /usr/local/opentrack
==11727== Memcheck, a memory error detector
==11727== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==11727== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==11727== Command: /usr/local/opentrack
==11727== 
==11727== 
==11727== Process terminating with default action of signal 11 (SIGSEGV)
==11727==  Bad permissions for mapped region at address 0x6C5F950
==11727==    at 0x13F56F8C: ??? (in /usr/lib64/qt4/libQtTest.so.4.8.6)
==11727==    by 0x400EDA9: call_init.part.0 (dl-init.c:72)
==11727==    by 0x400EEA4: call_init (dl-init.c:126)
==11727==    by 0x400EEA4: _dl_init (dl-init.c:120)
==11727==    by 0x4001119: ??? (in /lib64/ld-2.21.so)
==11727== 
==11727== FILE DESCRIPTORS: 4 open at exit.
==11727== Open file descriptor 3: /dev/urandom
==11727==    at 0x7392300: __open_nocancel (syscall-template.S:81)
==11727==    by 0x1A1E8478: ??? (in /usr/lib64/libgnutls.so.28.41.9)
==11727==    by 0x1A1E8A74: ??? (in /usr/lib64/libgnutls.so.28.41.9)
==11727==    by 0x1A15C13B: ??? (in /usr/lib64/libgnutls.so.28.41.9)
==11727==    by 0x1A1500AD: gnutls_global_init (in /usr/lib64/libgnutls.so.28.41.9)
==11727==    by 0x1A130A08: ??? (in /usr/lib64/libgnutls.so.28.41.9)
==11727==    by 0x400EDA9: call_init.part.0 (dl-init.c:72)
==11727==    by 0x400EEA4: call_init (dl-init.c:126)
==11727==    by 0x400EEA4: _dl_init (dl-init.c:120)
==11727==    by 0x4001119: ??? (in /lib64/ld-2.21.so)
==11727== 
==11727== Open file descriptor 2: /dev/pts/0
==11727==    <inherited from parent>
==11727== 
==11727== Open file descriptor 1: /dev/pts/0
==11727==    <inherited from parent>
==11727== 
==11727== Open file descriptor 0: /dev/pts/0
==11727==    <inherited from parent>
==11727== 
==11727== 
==11727== HEAP SUMMARY:
==11727==     in use at exit: 130,560 bytes in 879 blocks
==11727==   total heap usage: 1,507 allocs, 628 frees, 153,287 bytes allocated
==11727== 
==11727== LEAK SUMMARY:
==11727==    definitely lost: 0 bytes in 0 blocks
==11727==    indirectly lost: 0 bytes in 0 blocks
==11727==      possibly lost: 4,676 bytes in 83 blocks
==11727==    still reachable: 125,884 bytes in 796 blocks
==11727==         suppressed: 0 bytes in 0 blocks
==11727== Rerun with --leak-check=full to see details of leaked memory
==11727== 
==11727== For counts of detected and suppressed errors, rerun with: -v
==11727== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Segmentation fault

sthalik, 2016-01-13:
Your Qt installation is messed up. There are many ways to break things on Gentoo.

sthalik, 2016-01-15:
I don't think in this state you can even run qtconfig-qt5. Recompile qttest:5 for a start.

ff2000, 2016-01-22:
He is using the stable branch. In the CMakeLists.txt you only ask for OpenCV and not OpenCV-3.0. He is using OpenCV-2.4 (which only has Qt4-support), therefor opencv_highgui links against Qt4. opentrack pulls in Qt5, which causes trouble.
I hope the upgrade to OpenCV-3.0 + Qt5-support will fix this issue. Would be great if you could fix the find_package(opencv). At least I think it's not a broken system.

Furthermore please use proper install destinations. install([...] DESTINATION .) will throw everything into CMAKE_INSTALL_PREFIX, which is /usr/local by default (on linux) - not that nice ;) Left-over of the windows origin?

For reference: https://forums.gentoo.org/viewtopic-p-7870194.html#7870194

sthalik, 2016-01-23:
> Furthermore please use proper install destinations

We don't support hier(7) in this build system. You can package everything into `$prefix/libexec` if you must.

You may also disable qt/gtk/x11 support in opencv completely. opentrack doesn't use it any way.

sthalik, 2016-01-23:
> Is there a chance to also catch that case in opentracks CMakeLists.txt?

We need to link against highgui. There's no way to use `RTLD_DEEPBIND` except for `dlopen(3)` even if that'd help. I recommend building opencv separately (whether 2.4 or 3.0, I recommend master branch).

There's one more thing you need to fix, see `opencv/cmake/OpenCVFindLibsVideo.cmake`:

    if(HAVE_LIBV4L1 AND HAVE_LIBV4L2)
      set(HAVE_LIBV4L YES)
    else()
      set(HAVE_LIBV4L NO)
    endif()

Change that first line to

    if(HAVE_LIBV4L2)

Otherwise you'll use an obsolete camera interface that crashes on camera stoppage.

closed 2016-05-06 by sthalik

sthalik, 2016-07-07:
@ff2000 hier(7) is followed now in unstable.

sthalik, 2016-12-30:
For the record, we now don't link in the highgui module at all. If you still have it linked, maybe `-Wl,--as-needed` will help.

sthalik, 2016-12-30:
I just tested a fresh Linux build with shared packaged opencv and it doesn't link in highgui. If it gets linked in, delete the build directory and make a new one. There's sometimes stale data in cmake's cache.

xx70x, 2017-01-13:
CMake Error at CMakeLists.txt:1 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!                                              
See also "/tmp/opentrack/CMakeFiles/CMakeOutput.log".                                    
See also "/tmp/opentrack/CMakeFiles/CMakeError.log"

what is missing?

sthalik, 2017-01-14:
```
% which cc
% which c++
```

If this is a debian-based system, install the `build-essentials` package.