opentrack/opentrack #297: inverted image when using the PS3 eye

mr-kori, 2016-01-12:
Hi!
I decided to try Opentrack after freetrack and got the following problem. the camera image is reversed in opentrack. but in other programs, the camera works fine.

Image shows like it should on CL-test:

[image]

But on opentrack it's like this:

[image]
[image]

i try to win7 x64, win8.1 x64, versions opentrack: opentrack-2.3-rc21p7 and opentrack-stable-2.2
versions drivers for PS3 eye: CL-Eye-Driver-5.0.1.0218 and CL-Eye-Driver-5.3.0.0341
in different versions - not work, image inverted.
Any ideas?? I need help.

sthalik, 2016-01-15:
No ideas come to mind immediately.

opentrack opens the camera using a very general interface that works with webcams in general. No ps3 eye-specific code here. Maybe try a different video mode, like 30 FPS and see if this changes anything.

mr-kori, 2016-01-17:
Hello!
i have tried all of the configurations, but the orientation of the camera is not affected.
I decided my problem in a roundabout way. I use the program ManyCam, where i set the desired orientation of the camera. And in Opentrack i do not set PS3, i set a virtual camera ManyCam. It works.
so the question: may have an opportunity in your program to make such an option as an opportunity to change the orientation of the camera?
Thanks for you support!

sthalik, 2016-01-17:
@mr-kori no, barely anyone has use for changing camera orientation. It's good that you're able to use the software despite that camera bug.

closed 2016-01-17 by sthalik

sthalik, 2016-01-19:
@mr-kori see https://sourceforge.net/p/facetracknoir/discussion/1150910/thread/41678ed6/?limit=25#3eda

mr-kori, 2016-01-26:
@sthalik thanks, I looked at this, but in my case did not help. I removed K-Lite Codec Pack, clean up the registry - there is no effect, downloaded the latest version of K-Lite Codec Pack set - there is no effect .. it seems to deal in something else. of the programs that work with cameras in the system only Skype.

sthalik, 2016-12-31:
For the record we have plans to support cameras using a different library. But there's a few priority items in the backlog that need to get resolved first. The new library is the same source code as "CL-Eye Test", the code was released and published by Microsoft.

mr-kori, 2017-01-29:
@sthalik hello! thank for you dont forgot old question. where i can download this laybrary, and what should i do?

sthalik, 2017-01-29:
@mr-kori as a solution, install ManyCam and pass the video through it.

You can't download "the library", I have yet to write the stuff.