opentrack/opentrack #298: Yaw inverted when roll 90 degrees

deBoulangerie, 2016-01-12:
Using smartphone tracking via FreePIE, when head is rolled 90 degrees. The yaw movement is inverted. Using with VR goggles.

sthalik, 2016-01-15:
Use the other center method in options menu. If this doesn't help, select a rotation offset in freepie tracker settings.

MathijsG, 2016-01-15:
Or try different device index in the app, there are about 15 to try.

deBoulangerie, 2016-01-15:
Oke ik will try those options. What's does the device index do?

sthalik, 2016-01-15:
@MathijsG no, device index doesn't do anything. It's only for freepie compatibility.

deBoulangerie, 2016-01-19:
had no succes so far, i tried setting centering method to relative (inertial device). this works perfect for roll and pitch but when i enable yaw, open track translate yaw head movement in to roll? When only yaw is enabled yaw works fine.

sthalik, 2016-01-20:
You can also remap axis in options screen. When the phone is rotated the meaning of axis changes.

We can't fix centering well enough because protocols, including FreePIE, send Euler angles and not robust representation like quaternion, Rodrigues angles, etc.

deBoulangerie, 2016-01-25:
I think the problem i related to how the phone is orientated ( sounds logic). All tracking is working fine is you place the phone on top of your head, but when moving the phone in to the vr glasses position all motions are mixed up.

Boomkop3, 2016-02-08:
I think the problem is found in the opentrack software, not the phone

I tried running three instances of the freepie app on my phone, forwarding each to a seperate udp port
I had three instances of opentrack listening to one of the freepie instances, now pitch, yaw and roll where captured seperately (moving correctly)

Then I forwarded all three opentrack instances to a fourth instance of opentrack, combining the three udp streams

The tracking was captured correctly, Yaw was not translated into roll
(not counting the fact that each axis constantly jumping back and forth from 0)

sthalik, 2016-02-09:
That each of three rotation axis work fine separately, doesn't change anything. The problem will exist because of interchange representation of rotations in the freepie udp protocol and the same clearly suboptimal representation used in actual game protocols, not just "freetrack 2.0" but all of them actually.

We can't feasibly change any of these, the latter definitely not ever. That's why the axis shift feature exists at all, even if not ideal. We don't have control over the other elements needed that would allow for a mathematically correct solution.

You should
- enable centering type for inertial devices
- negate any rotation offsets caused by keeping the device rotated

After the offsets are corrected, the centering type will actually work correctly, unlike the camera device centering type. You can verify offsets looking at the raw data panel. Keep in mind that raw data isn't influenced by triggering pose center so it's reliable. Rule of thumb, negate the change that's caused by rotating the device.

There were already few reports of proper operation once both of these conditions are met. The offset shift functionality got tested and was verified to help prior to release.

I'll ask some questions to existing users.

sthalik, 2016-02-09:
Here's an example of shifting the rotation axis.

Assume you get a neutral rotation if the phone's lying on its back and the screen in portrait mode. You need to have it standing up and in landscape mode.

Then, you need to negate rotation in two degrees of freedom, one for back -> upward, second for portrait -> landscape. Notice which degrees of freedom these are, since both are changed by `+- 90`. Negate these rotations and program should operate correctly in this chosen position.

We already take some care to choose a more "canonical" representation from the ones that the gimbal lock problem causes. Adding more injury, the trackers in opentrack even including Rift and Arduino, return Euler angles that cause the gimbal lock problem. Even if they returned a non-ambiguous representation (e.g. Rodrigues angles) we STILL have to convert back to Euler for the protocols that games use.

If you can't get centering and rotation to work properly after following these instructions I'll look into making changes.

closed 2016-05-06 by sthalik