opentrack/opentrack #299: freetrack on linux

nathanvl, 2016-01-16:
Hello,
thank you for this amazing software. Wasn't easy to get starting, but so far works great.

My only problem is that I can only see 2 protocols in the dropdown: FlightGear & UDP receiver
It would be lovely if I had access to Freetrack, but it is windows-only it seems.
In the readme it says:
Wine freetrack glue protocol (Linux, OSX)
I cannot find this plugin, or something similar.

I have wine, how do I make this work? Freetrack doesn't really seem to work with wine. I can easily run 3rd party "Freetracktest.exe", but that doesn't recognise anything as the protocol is not running.

I'm using my android phone with freepie, but that part works like a charm.

nathanvl, 2016-01-16:
now that I think of it, "Virtual joystick output (Linux, Windows)" should work as well. what protocol do I need for that?

sthalik, 2016-01-16:
You need to enable Wine and set its prefix (/usr or something else) in cmake cache.

For virtual joystick output it's enough to install libevdev.

nathanvl, 2016-01-16:
thanks, will try!

sthalik, 2016-01-16:
Keep in mind that on a 64-bit system you need a lot of libraries for `winegcc -m32` to function. Their names aren't obvious at all and it's a pain for me to do on a fresh system. It's doable of course, in the worst case you can copy 32-bit .so's according to multiarch.

nathanvl, 2016-01-17:
I'll try to get the virtual joystick working first in that case. I'll do it after my exams finish, I am still kind of  a linux noob and installing opentrack was a real chore for me, after the install it took me more than half an hour to figure out how on earth I would run this program... I'm guessing making the virtual joystick work is a bit harder than just installing libevdev-dev. I did that, but the steps between having it installed and having "virtual joystick" as an output option in the opentrack GUI aren't exactly clear. I guess there's a lot of documentation for me to read before I get to that point.

sthalik, 2016-01-17:
You need read and write permissions for the device node that allows creating virtual joysticks.

That's why Linux isn't supported. It's not an OS ready for end-users.

closed 2016-05-06 by sthalik

wleoncio, 2018-03-13:
If I understood correctly, OP changed the topic from Freetrack to virtual joysticks right from the start. However, since the title is still on Freetrack, it didn't feel right to open a new issue just to ask this: what are the current plans for Linux support on the Freetrack protocol?

sthalik, 2018-03-14:
Hey @wleoncio,

freetrack works for Wine games, using the Wine protocol. It's not supported for native games since no native Unix game uses freetrack. I used it a lot when booting Linux directly and it's expected to work at all times. You must set `SDK_WINE_PREFIX` to where you installed Wine, e.g. `/usr` or `/opt/wine`. If you installed from distro packages, it's `/usr`.

It doesn't detect automatically because some Wine packages are missing devel files. It was either Ubuntu or Debian, I forgot. Then the build fails despite technically having Wine installed.

You can open a new issue with questions or suggestions. We're a friendly bunch that doesn't bite.

Sirtz, 2019-01-20:
Hi, 
I'm a linux noob trying to get this running, I installed opentrack using the arch aur, which uses opentrack 2.3.10-1. Now opentrack launches fine, problem is freetrack does not show up in the input list. Where would I set SDK_WINE_PREFIX? Is the a build option? Or in a config file?

YosypC, 2024-03-28:
> Hi, I'm a linux noob trying to get this running, I installed opentrack using the arch aur, which uses opentrack 2.3.10-1. Now opentrack launches fine, problem is freetrack does not show up in the input list. Where would I set SDK_WINE_PREFIX? Is the a build option? Or in a config file?

I have the same issue. I can't find a string to attach the wine prefix.

sthalik, 2024-03-28:
> I can't find a string to attach the wine prefix.

Set `PATH`.