opentrack/opentrack #300: Exact packet data layout of FreePIE UDP data?

magwo, 2016-01-17:
Hi,

I'm trying to get the Sensor Data app on iOS to work with opentrack (FreePIE UDP receiver) on OSX.

It did not appear to work at all out of the box, so I wrote a UDP transformation utility in node, that retransmits the data.

As far as I can see the Sensor Data app sends a comma separated list of values in plain text.
Looking at the source code of the FreePIE UDP receiver, it seems like the packet data layout is something like this:

8 bits padding (all zeroes?)
8 bits with flags (only bit 0 and 1 used?)
6 x 64-bit doubles (little endian?): xyz, yaw pitch roll

I've got it almost working. Values are getting received in opentrack, but they are all weird and make no sense. And when I change one value, for example roll, all the other values in opentrack unexpectedly change. Also I've only ever seen opentrack receive yaw, pitch, roll - never XYZ translation values.

Is the data layout correct? What should the flags look like? I only got it working with unsigned int8 of value 3, meaning that flags 0 and 1 are set to 1. Should there be any kind of padding between the double float components?

Best regards and thanks for a cool project!

sthalik, 2016-01-17:
The packet can have other layouts if raw sensor data is sent. This is the simplest one containing only yaw, pitch, roll, components:

- uint8: padding, any value
- uint8: flags, described below
- float[3]: yaw, pitch, roll in radians

For the flags component, send a value of `(1 << 1)` aka `2` which means that you don't send sensor data. The sensor data is always unused by opentrack, the logic to receive it only exists for compatibility with the freepie Android app.

sthalik, 2016-01-19:
We might want to support Sensor Data directly. Also particular there are various software transmitting only sensor data but no estimated pose. We can borrow pose estimation from FreePIE. I'll probably deal with that following issue reports.

magwo, 2016-01-19:
Ok!

I successfully wrote the UDP translator after your clarifications.

So I can use the phone for head tracking with Sensor Data app.
Unfortunately the Sensor Data app is somewhat unstable and crashes when the phone screen locks it seems.

Anyway, I was unable to figure out how to send YPR values to X-Plane from Opentrack (OSX), so I ended up sending them directly to X-Plane via UDP, first with DATA packets, then with DREF packets.

Works pretty well, but the DREF setting is kinda stuttery. Not sure if it could be due to both phone and computer being on Wifi and not proper ethernet.

magwo, 2016-01-19:
By the way.. I would help with implementing Sensor Data support and/or X-Plane UDP sending, but I'm not really into C/C++ code. :(

sthalik, 2016-01-19:
you can use the opentrack.xpl plug-in for x-plane of you put it in an appropriate directory structure. it works with the Wine protocol in opentrack. there's only a 64-bit version of the x-plane protocol in the osx version of opentrack binaries.

sthalik, 2016-01-19:
is sensor data the most popular iPhone app for the purpose or is there another one that should be implemented instead?

sthalik, 2016-01-19:
i don't need third party code, all i need is the protocol given app talks.

magwo, 2016-01-19:
Sensor Data seems quite popular. I found another one called Sensor or Sensor streamer or something. Seems about as good but not as configurable.

The Sensor Data app allows the user to select which sensor components to send in the stream.
So the protocol can vary. But the default seems to be to send Accelerometer + YPR in a comma separated list, plain text.

The first component is a time value that can either be elapsed time or absolute time depending on configuration. Next up are the 6 comma separated numbers in plain text. Just split the string on comma and map the axis correctly!

sthalik, 2016-01-19:
send me sensor data network communication  and we'll support it natively. also whether i tcp or udp

magwo, 2016-01-19:
It's UDP. Example packet from Sensor Data:

```134.002,0.0234,-0.5232,-0.0213,0.0009,0.9321,0.0206```

sthalik, 2016-01-19:
Accelerometer data is useless for us since it's relative to the device's current orientation.

I'll work on it when able, and report back when I have something to test.

magwo, 2016-01-19:
Cool.

sthalik, 2016-01-19:
I'm curious though, are there any issues with the OSX version of the X-Plane plugin? I couldn't get OpenGL to work on OSX but the code is identical to the Linux version.

magwo, 2016-01-19:
Not as far as I know. But I was unable to find it when I looked for it. When I downloaded the latest release I think I only got the main Opentrack application. Where is the plugin located?

sthalik, 2016-01-19:
Open `opentrack.app` as a directory and you'll see a bunch of files. The one you want is `Content/MacOS/opentrack.xpl`. You still need to put in plugin directory structure and rename the file. I'm unaware of the details here but it's the same process as any other X-Plane plugin.

magwo, 2016-01-19:
Ok thanks.

magwo, 2016-01-19:
By the way, the reason I can't check the exact packet structure right now is because I'm at work and don't have my UDP translator app available. Can get back to you later.

sthalik, 2016-01-19:
Okay. I don't have any iPhones available, so can't do this at all without your description.

magwo, 2016-01-19:
Slightly off topic, but are you aware if X-Plane can accept pilot-head-roll or not? I only found head pitch and yaw in the data refs.

magwo, 2016-01-21:
I got the plugin working fine in X-Plane by the way. With my udp transformer as a bridge between the Sensor Data app and opentrack.

magwo, 2016-01-21:
It would be kind of neat with a reception method in opentrack that was like a "configurable UDP" receiver.. but I realise that it would be a big development effort. Something that I personally as a JavaScript-proficient developer would appreciate would be like a "Custom UDP" receiver, where you would paste JavaScript that would interpret the UDP packet for opentrack.

sthalik, 2016-01-23:
@magwo X-Plane can do roll, but only through OpenGL trickery like linuxtrack's that rotates the view manually. This is more intrusive and error-prone.

We don't have scripting support at all for now. You can do the same though with some easy C++ code.

epaga, 2017-01-23:
@magwo @sthalik Starting to look into this - what is the latest, recommended way to try using opentrack with X-Plane 64-bit on OSX using an iPhone (or Android device, I have an old one somewhere) as the sensor? Is there an official build somewhere? Do I need to build it myself?

magwo, 2017-01-23:
I think this is the last/latest version of OpenTrack with an OSX release: https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3-rc49p6

I would guess it would work fine for testing/development purposes.

epaga, 2017-01-23:
Thanks, I got that app up and running, but when I install the "opentrack.xpl" that is in that .app file into Resources/plugins it gives me the following error in Log.txt

```
dlerror:dlopen(/Users/johngoering/Desktop/X-Plane 11/Resources/plugins/opentrack.xpl, 6): Library not loaded: @executable_path/XPLM
  Referenced from: /Users/johngoering/Desktop/X-Plane 11/Resources/plugins/opentrack.xpl
  Reason: image not found
Failed: /Users/johngoering/Desktop/X-Plane 11/Resources/plugins/opentrack.xpl.  (This file is missing, not a DLL or could not be loaded due to another missing DLL.)

```

sthalik, 2017-01-24:
We need a programmer volunteering to maintain the OSX build. I don't even dual-boot OSX. The build only needs few minor changes but so far no one volunteered.

epaga, 2017-01-24:
So I got an Xcode project created with CMake at least, but when I try and build the Xcode project, it fails with:

```
Undefined symbols for architecture x86_64:
  "isinf", referenced from:
      _nanp in nan.o
  "isnan", referenced from:
      _nanp in nan.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

What do I need to change?

sthalik, 2017-01-24:
Try add `link_libraries(subproject-name m)`.

epaga, 2017-01-24:
Sorry, total CMake noob here -> add that where? To some CMakeLists.txt somewhere? In the Xcode project? Or before creating the Xcode project in the first place?

Serggnn, 2018-02-11:
After searching for apps that work with the Freepie UDP receiver on ios, I left a bit disappointed. There is apparently only a couple of apps that can send sensor data through UDP.
There is only one app that have a free version (you need to tap the screen after 3 min so the app can continue sending data). The app name is **Sensor Data Streamer**, it has almost the same name as the one in the first comment but are completely different apps.
The Freepie UDP receiver can read data sent by the app but it makes the octopus from Opentrack spin in almost random directions when I move the iPhone. 
The app also shows the data packet format: https://imgur.com/a/IEJG5 . I would like to have this working correctly but I lack the programming knowledge.
Since I am in a similar situation presented by the first post and this might not be worth creating an issue for, I decided to leave my comment here to ask if there is a possibility to add support for this app. 
It is unfortunate that there isn't more sensor apps on the ios.

 Sorry for bothering and thank you for this great project!

sthalik, 2018-02-12:
@Serggnn the biggest problem for the project is that there's no active developer building for OSX. We can't expect things to work unless someone actually runs Apple's OS.

There are some things not done correctly for OSX. For instance keybindings don't work with few basic keys. Providing binary builds isn't possible as there's no one to test them. The build isn't even tested to work on OSX.

Unless you're running on Windows (or Linux) for the iPhone supporting that data streamer app is pointless to begin with due to lack of proper OSX support.

Serggnn, 2018-02-12:
I forgot to say that I am running Windows. 
I guess it would be a nice step to support this, given it is currently the only imu streamer available for free to test on iPhone.
Also, is there any way to have more settings/info in the GUI about the data that FreePIE UDP receiver is reading? 
Maybe would be possible for the user to get an app working/debug inside the Freepie receiver window and export the result like a profile... 

Well, maybe this is too much asking. thanks for answering

closed 2018-08-04 by sthalik