opentrack/opentrack #301: Camera movement not smooth (stuttering)

watermanpc, 2016-01-18:
Hi!! thanks for the GREAT opentrack soft!!!

I have a problem with the tracking. 

Im using Opentrack 2.3-rc21p7 with PS3 eye and 3 point IR clip (also a 4 ways polarized filter to COMPLETELLY eliminate visible light while not afecting one bit IR component  and so using very low exposition and zero gain in CL eye filter so this is not the problem), and when using mk4 accela filter I have to set smoothing and sensitivities WAY HIGHER than what I would consider "normal" to get an SMOOTH movement in games (DCS)...I mean when doing that response is uterly lost and movements are too lagged to be usable.

The problem is that even at very high levels of smoothness and sensitivity camera in game stutters and its hard to make an smooth movement folowing objets in game. tracking seems to be making little jumps to the next position and I think this is not normal behaviour.

If I use other filter this is not happening and camera moves very smooth between positions.

I have tried with different settings in fps (60/125) but it doesnt matter, it stutters a lot...same with track hat.

Any idea mates??

Len62, 2016-01-18:
Here are my accela settings:

[image]
These work good for me in DCS. Re-check all of your other settings in OT to try and smooth it out. Not sure what could be causing that issue.

watermanpc, 2016-01-18:
Many thanks for your reply mate!!

I have tried your settings but...I feel they are too slow for me...that way it loses all the precision and speed and I know, I know one must sacrifice speed in favour of smoothness but I think there is something wrong with the filter...the tracking makes like "jumps" and stutters instead of moving camera smoothly, like if the tracking is skipping some steps in the movement . 

Also, I can confirm its not any problem with my dots, they are perfectly clear in all positions and bright enough to dont lose the tracking at any time...

dbaarda, 2016-01-19:
A large deadzone will make it jump/stutter... deadzone means "ignore any
movement smaller than this", which means you get no movement until it is
larger than that, at which point it will jump to the new position.

deadzone is a pretty brutal way to make it perfectly smooth when still, but
it's pretty nasty on movement.

I've not looked at this stuff for ages and I don't know what state it's in
now, but the EWMA2 filter used to be pretty good and might work better for
you.

Len62, 2016-01-19:
While you have OT running hold your head as still as possible and watch the "raw data" area, are the numbers jumping around?
Did you remove the IR filter from the PS3 eye?

sthalik, 2016-01-19:
@watermanpc you might have your model set up wrong.

watermanpc, 2016-01-19:
@dbaarda :

Good point!!...at first I though that could be the reason as I used a 0.2º dead zone, but after some testing even to zero dead zone the result is almost the same...it improves a bit, but response stills being stuttery...EWMA2 is smoother but the filter itself it worse than accela and I think its not worth it.

@Len62:

No, numbers in the raw area are perfectly still as are the game data. About the filter, no, I didnt removed the IR filter in my PS3 eye because its the bad model. But there is not any problem with that as for some reason my PS3 eye is really sensitive to IR light (my led almost blind the camera...in fact I would say it doesnt have any IR filter for some reason. Also, as I said, dots are perfectly visible for the camera and the are 4 to 7 pixels diameter as reported in OT. The background is perfectly black due to the zero gain and max exposure. Again, for some reason my ps3 eye acts as there is not any ir filter on it, light bulbs looks incredible shiny and with my polarized filter there is NOT ANY brightness loss.

Tried also with the protocols...no success.

@sthalik :

I dont know what do you mean exactly...I made the calibration and it seems to be perfectly fine...my head movements are very accurate and symmetric...could you elaborate please??

Anyway, the stutter is not that bad, and maybe Im a bit fussy, I dont know, but looking some videos with original track ir I can clearly see how it works perfectly smooth and people can perfectly follow moving planes with the tracking and camera moves incredible smooth with no stutters nor skipping steps...I dont think there should be any problem achieving that with OT but I cant find the way...

thanks guys!!

Len62, 2016-01-19:
Your aren't by any chance running cleye while running OT are you? Once you have your settings in cleye you can shut that program down.

sthalik, 2016-01-19:
@dbaarda accela deadzone doesn't jump immediately since low deltas have low gain as well. 

deadzone is here a measure of first resort since sensitivity slider stretches the whole grain function. 

@watermanpc the numbers you put in model tab in pt settings aren't accurate enough probably.

kristianw87, 2016-01-19:
@watermanpc I used to get stutter as well when rotating slowly, and reducing the deadzone helped me a lot. (think I have it at 0.2 or 0.1, will check when I get home)

I get best results with lens FOV at 56 degrees (red dot, which is a bit blurry if you have removed the irfilter) 640x480@60hz, and I have noticed that I can use lower values in the accela filter the bigger my point size is. My points are usually around 9-11 pixels. 

This makes movement in DCS is smooth and very stable when looking down at cockpit instruments for me.

watermanpc, 2016-01-19:
@Len62 

no, I dont have it opened when OT is running.

@sthalik 

I dont think so because I made my model based on the "standar" dimension given by OT (80,70,40,30 mm) and I have just checked they are perfectly correct...

@kristianw87

I have deadzone to 0,2 and even at zero movement still being not smooth, its like if it makes micro accelerations and brakings while the camera moves...

You talked about something interesting also...FOV...I have no idea whats best for ps3 eye nor what exactly it does a part from increasing or decreasing the fov in game which i dont understand also. Right now I have it set at 75º and the phisical ps3 eye zoom ring is in the blue dot (wide). How should I set it up??...

I have to try also what you said about dots size, good point here!!...but to make them so big I will have to use the zoom in the camera and probably the 56º in OT and even then Im not sure if I will be able to reach those sizes...also, I will lost a lot of room for movements right??

Many thanks to all guys...

kristianw87, 2016-01-19:
@watermanpc 
There are two physical zoom modes on the ps3 eye camera, the blue one is 75º and the red one is 56º. You will get less room for movement with 56º, but exactly how much room depends on how close you are to the camera.

There are a lot of individual factors in Opentrack, so you will have to experiment a bit to get it right.

[image]

watermanpc, 2016-01-19:
@kristianw87 

Wow, I tested what you told me about dots size and it improved a lot smoothnes!!

I have tested with hat track which is basically OT.

I have set ps3 phisical zoom to the red dot, then in OT set fov to 56º. In PT 75fps and those parameters:

[image]
[image]

Its not perfectly smooth but it improved a lot!!...thanks mate!!!

Also, I have some questions about dots size. Whats the ideal thing here...

-very large dots or small dots??
-what does "automatic threshold" do?? (exactly)
-When the check mark is selected whats working then??, the auto, the threshold slider or the min/max pixels selectors?¿?¿
-Do you think I should change my dynamic pose timeout to 1000ms as you or can I let it at 25ms as I have it now??
-Finally, I wonder whats best when using the threshold slider to achieve...a very little "blue" dot inside the white one or hte most posible blue dot inside the white one???...an example:

[image]

What would be the best, 1,2 or 3??

Many, many thanks!!

sthalik, 2016-01-20:
@watermanpc 

Larger dots are more precise.
Automatic threshold detects based on point size, not exact lightness.
When in auto threshold mode slider controls point size.
Change pose timeout to 500 or 1000 ms.
You can skip min/max if you have no other light sources in the frame.
For the threshold slider, detect the largest light amount possible. Turn exposure further on so that more of the light is visible. As before, larger the light, less filtering required.

It's not a problem if lights are of different size.

There's something broken there. My Accela settings are like below 0.6 degree sensitivity and you have three degrees.

You also want correct FOV in PT settings for translation not affecting rotation.

MathijsG, 2016-01-20:
I never need that much of smoothing, I use way less values than you and it's perfect here. Maybe check the treshold bar? Sometimes it's too sensitive or maybe too less.

What did you use for filtering? When I want to make the leds more bright with the CL test program, more and more bright items in my room are also showing up. I used a single piece of floppy diskette, put it INSIDE the lense, but I'm interested in a better filter, which really blocks ALL light, except IR.

watermanpc, 2016-01-20:
@sthalik 

Thanks mate!!

You said the bigger the better for dots but there are 2 things here:

1-The size of the points themselves (I mean the size of the leds visible to the camera in black and white)
2-The amount/size of the BLUE dot that OT changes with the threshold slider.

How both variables work is what I would like to understan but I asume the best is: Bigger dots for the leds and most of the led dot tinted with blue. 

BTW, I dont know why my dot color in OT is blue while some of you have it green with the same version...

Also, why are higher values for pose tiemout better???I try to understand, sorry for the inconveniences guys!!

About the exposure, its at maximum right now but gain is at zero because after some testing I can asure you that the blacker the background the better the stability...just having a bit of noise is really bad for the tracking so I think thats the best possible filter setting for cl eye.

About the filter, dont worry, thats because Im using track Hat (which is basically OT but with minor changes in UI) so the filter settings are a bit odd in numbers but its the same percentage in the sliders as OT...I mean that despite you see 50ms smooth, its equivalent to 20 or 30ms in OT...dont know why its done that way.

The FOV I set is 56º in software and the red dot in PS3 eye (zoomed)...is that correct??

@MathijsG 

As I said before there is something weird with track hat and values but the 50ms in TH is the same as 20/30ms in OT...it works great for me right now as it is there, really fast and smooth.

For filtering I changed the polarized filter for 2 layers of photographic film, which works amazing (not a bit of lose in brightness for IR and zero visible light) and its way better that floppy disk...you should try it pal!!

MathijsG, 2016-01-20:
You just use a piece of exposed analog film? I will try!

watermanpc, 2016-01-20:
yes, just that (double leyered but may not be necesary)...its just the best filter!!

sthalik, 2016-01-23:
@watermanpc 

Bigger LEDs make contour detection select a more stable point center, hence less need to filter.

For dynamic pose resolution - when a LED disappears it's typically for a larger fraction of a second. With a larger timeout it's taken care of. But if you select an absurdly high timeout you can't cover your LEDs in case they get confused and nothing works. One or two seconds should be fine.

We don't mask the detected portions with blue color anymore, but you can still see point radius changing as text overlayed in red.

You shouldn't have any filter differences in trackhat and opentrack. I'm sure that's the same code.

watermanpc, 2016-01-24:
Thanks mate, but I dont understand well what you said about the larger timeout?¿?¿?

Also the blue portions and point radius??where can I see that??...test overlayed changes also??I just see the radius in pixels...

sthalik, 2016-01-29:
@watermanpc the timeout should be higher than the time a point disappears. It's typical to stop seeing a point for around a second in extreme cases (rotation, failed thresholding). If dynamic pose timeout occurs there, the tracker might start confusing which point is which.

closed 2016-06-16 by sthalik