opentrack/opentrack #303: What resolution/fps is best for PS3 eye??

watermanpc, 2016-01-18:
I cant find any info about this regard...

I have tried 640x480@60/75fps and 320x240@125+ fps and I cant see much of a difference but to me, the lower res works a bit worse as I cant make the dots to stay still (they are constantly jittering) and I dont see any benefit by the considerable fps increment, even at 180fps...

What option do you guys use??

Also, I wonder if when you use vsync in game, does this affect fps tracking in opentrack??? are the frames from the camera synchronized with those in the game??...whats best??60fps or 75fps in opentrack when vsync is enabled in game??

Also, does fps limiters afect opentrack fps???

thanks!!!

Len62, 2016-01-18:
I keep mine at 640x480 @75. Really smooth for me at those settings. I've never run into an FPS issue while using OT with DCS/IL2-BOS/ROF.

watermanpc, 2016-01-18:
Ok, thanks mate!! anyone else?? I would like to know more opinions...

sniperwolfpk5, 2016-01-19:
Well 60 fps are more than enough. I use 640X480 @60 and things are working perfectly. 640X480 is the cream resolution. Increasing and decreasing resolutions from 640X480 is not wise for traking point of view. More resolution require a lot of memory and processing. It is hard to keep high fps plus less resolution are difficult to keep track of dots. So 640X480 are the best settings.

sthalik, 2016-01-19:
Don't use 320x240. Lower resolution need more filtering.

watermanpc, 2016-01-19:
Ok guys!!, thanks for your replies, its exactly what I wanted to know...As I expected, lower res is very jittery no matter how you always get unexpected movements and fps increase is not very noticeable...at 640x480 camera is perfectly stable and I cant see any decrease in speed or something by the reduced fps...

sthalik, 2016-01-19:
you shouldn't feel a difference in tracking from 30 to any high value in point tracker. 

higher resolution is more accurate but cameras don't support that on reasonably high framerates at all

watermanpc, 2016-01-19:
well, at 30fps I DO really feel a difference mate...but beyond 60fps is quite hard to notice...

MathijsG, 2016-01-19:
I use 640 * 480 @ 75 because it is the max, and I don't benefit from the spare CPU time when I would use 60 FPS. I guess it also depends on what people's screens refresh rate are. Many new monitors are already doing 120 or 144Hz, then you rather want something like 75 in Opentrack instead of just 30 FPS.

watermanpc, 2016-01-19:
Ok mate, thanks, I think that as soon as there seem to not be any OT vsync I guess its best to have the faster possible frame rate so I will stay at 75fps if there is not any other opinion saying the opposite...

sthalik, 2016-01-20:
Human head is very imprecise controller though. You benefit way more from videos at 60 Hz. The need to filter is mostly since heads shake on their own. I've adjuster filtering to a hand-held model and it was horribly shaky when i put it on my headphones.

RoarGi, 2016-02-12:
Hi, guys! Opentrack don't work with my PS3 Eye. When I push Start button nothing happens. Any ideas?

MathijsG, 2016-02-12:
Please make a separate issue and please provide some more info: What operating System, did you installed the PS3 Eye CL drivers? Does the webcam work with other webcam programs? Without any proper information we simply can't point in a direction.

Trollgaard-Sr, 2016-05-06:
I can not get higher fps than 30 in any resolution on Windows 10.
Win10x64
Opentrack-2.3-rc21p11
PS3eye CL driver 5.3.0.341

When starting CL-EyeTest as administrator I get all the different fps options
Starting Opentrack without adminrights gives no image.
With administrator rights in Opentrack I get good tracking and everything work except that the fps always stays at 30 fps regardless of resolution.

Has anyone had this problem?

sthalik, 2016-05-07:
> EDIT: Turns out there was an error with the Directshow filter because of a corrupt installationpackage. 

I recommend to everyone, install K-Lite codec pack the biggest version on a fresh system and not touch the codecs otherwise in any way.

closed 2016-06-16 by sthalik