opentrack/opentrack #305: A couple of nagging issues

Pulstar85, 2016-01-22:
First of all thank you for your effort into this great application, secondly I have been trying to see if I can play non-TrackIR/FT games using the UDP client, especially for walking simulators. Is there a semi-ideal way of configuring mouse emulation to enable effects similar to Arma and Sir You're Being Hunted? However hard I try there's a lot of camera compensation to do to make it an enjoyable experience... And to add further: Is it possible to consider mapping the roll function to keys (e.g. E and Q) in case said game supported linear leaning?

sthalik, 2016-01-23:
We're already doing the best we can for mouse support. Maybe linear mapping can help you.

We could have a "snapview" protocol at some point that binds threshold values to keys. But we don't have multiple-protocol support yet. Once multi-tracker support is in-tree I'll see what can be done about that.

Pulstar85, 2016-01-24:
That sounds encouraging, cheers. :)

sthalik, 2017-06-18:
Actually we already have game detection support. Duplicate a single profile and assign it with different protocols, each connected to game detection.

closed 2017-06-18 by sthalik