opentrack/opentrack #308: Stable 2.2 on Linux: munmap_chunk error

bar0nP, 2016-01-29:
Hello
I use opentrack 2.2-stable-20150728 on UbuntuMATE-14.04. Compiles ok, but starting it ends with munmap_chunk error.

$ sudo /usr/local/opentrack.bin
*** Error in `/usr/local/opentrack.bin': munmap_chunk(): invalid pointer: 0x00007fffebe4d120 ***

sthalik, 2016-01-29:
No need to use `sudo`. Run with `valgrind` and post a backtrace.

bar0nP, 2016-01-29:
Without `su` (sorry for so many strings):

$ valgrind /usr/local/opentrack.bin
==8834== Memcheck, a memory error detector
==8834== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==8834== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==8834== Command: /usr/local/opentrack.bin
==8834== 

** (opentrack.bin:8834): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-YIiC1IpHNG: В соединении отказано
==8834== Conditional jump or move depends on uninitialised value(s)
==8834==    at 0x15E084EA: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x82045E6: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x821D087: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x821E211: g_signal_emit_by_name (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x820BB38: g_object_set_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x820C396: g_object_set (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x5CD8EE5: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==8834==    by 0x5C5C07C: QStyleFactory::create(QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==8834==    by 0x5BFB98B: QApplication::style() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==8834==    by 0x5BFBCE4: QApplicationPrivate::initialize() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==8834==    by 0x5BFBD32: QApplicationPrivate::construct() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==8834==    by 0x44741D: main (in /usr/local/opentrack.bin)
==8834== 
==8834== Conditional jump or move depends on uninitialised value(s)
==8834==    at 0x84BB320: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4002.0)
==8834==    by 0x15E0B6D0: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15E0BC5D: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15E0BF4E: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x82043B7: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x8215556: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x821DA28: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x821DCE1: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x15F8FA4F: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15F90D27: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15E395F2: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x8207486: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834== 
==8834== Conditional jump or move depends on uninitialised value(s)
==8834==    at 0x84BB34F: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4002.0)
==8834==    by 0x15E0B6D0: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15E0BC5D: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15E0BF4E: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x82043B7: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x8215556: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x821DA28: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x821DCE1: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834==    by 0x15F8FA4F: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15F90D27: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x15E395F2: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==8834==    by 0x8207486: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==8834== 
==8834== Invalid free() / delete / delete[] / realloc()
==8834==    at 0x4C2C2BC: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==8834==    by 0x21A68BCB: FTNoIR_FilterDll::~FTNoIR_FilterDll() (in /usr/local/libopentrack-filter-accela.so)
==8834==    by 0x434434: dylib::dylib(QString const&, dylib::Type) (in /usr/local/opentrack.bin)
==8834==    by 0x444345: _ZN9__gnu_cxx13new_allocatorI5dylibE9constructIS1_IR7QStringRNS1_4TypeEEEEvPT_DpOT0_ (in /usr/local/opentrack.bin)
==8834==    by 0x44406D: _ZNSt16allocator_traitsISaI5dylibEE12_S_constructIS0_IR7QStringRNS0_4TypeEEEENSt9enable_ifIXsrNS2_18__construct_helperIT_IDpT0_EEE5valueEvE4typeERS1_PSA_DpOSB_ (in /usr/local/opentrack.bin)
==8834==    by 0x443C21: _ZNSt16allocator_traitsISaI5dylibEE9constructIS0_IR7QStringRNS0_4TypeEEEEDTcl12_S_constructfp_fp0_spcl7forwardIT0_Efp1_EEERS1_PT_DpOS8_ (in /usr/local/opentrack.bin)
==8834==    by 0x44346D: std::_Sp_counted_ptr_inplace<dylib, std::allocator<dylib>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<QString&, dylib::Type&>(std::allocator<dylib>, QString&, dylib::Type&) (in /usr/local/opentrack.bin)
[... cut ...]
==8834==    indirectly lost: 22,087 bytes in 835 blocks
==8834==      possibly lost: 328,901 bytes in 738 blocks
==8834==    still reachable: 1,646,342 bytes in 18,696 blocks
==8834==         suppressed: 0 bytes in 0 blocks
==8834== Rerun with --leak-check=full to see details of leaked memory
==8834== 
==8834== For counts of detected and suppressed errors, rerun with: -v
==8834== Use --track-origins=yes to see where uninitialised values come from
==8834== ERROR SUMMARY: 5 errors from 5 contexts (suppressed: 0 from 0)

bar0nP, 2016-01-29:
Damp wrong button

sthalik, 2016-01-29:
There's no `dylib` class in `stable/2.2`. Check what your checked-out source is as per `git status`. It might not be a fresh copy of either of the branches.

I recommend you build `unstable` anyway. It's in a decent shape right now despite the name.

I've built `unstable` (that's where the `dylib` class exists) and it gives no valgrind errors on startup.

When you have a checked-out tree of either of the branches and still have problems, add `-g` to C/C++ compile flags for line information and drop another backtrace. This one doesn't make sense since `GetMetadata` always uses `new` and `delete` is fine on it. It's never stack-allocated.

bar0nP, 2016-01-31:
It's not simple. If use opentrack-2.3-rc21p7 I get that errors:

$ valgrind /usr/local/opentrack.bin
==2243== Memcheck, a memory error detector
==2243== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==2243== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==2243== Command: /usr/local/opentrack.bin
==2243== 

** (opentrack.bin:2243): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-YIiC1IpHNG: В соединении отказано
==2243== Conditional jump or move depends on uninitialised value(s)
==2243==    at 0x15E244EA: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x82205E6: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8239087: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x823A211: g_signal_emit_by_name (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8227B38: g_object_set_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8228396: g_object_set (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x5CF7EE5: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==2243==    by 0x5C7B07C: QStyleFactory::create(QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==2243==    by 0x5C1A98B: QApplication::style() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==2243==    by 0x5C1ACE4: QApplicationPrivate::initialize() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==2243==    by 0x5C1AD32: QApplicationPrivate::construct() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.2.1)
==2243==    by 0x445126: main (in /usr/local/opentrack.bin)
==2243== 
==2243== Conditional jump or move depends on uninitialised value(s)
==2243==    at 0x84D7320: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4002.0)
==2243==    by 0x15E276D0: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15E27C5D: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15E27F4E: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x82203B7: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8231556: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8239A28: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8239CE1: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x15FABA4F: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15FACD27: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15E555F2: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x8223486: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243== 
==2243== Conditional jump or move depends on uninitialised value(s)
==2243==    at 0x84D734F: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4002.0)
==2243==    by 0x15E276D0: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15E27C5D: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15E27F4E: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x82203B7: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8231556: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8239A28: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x8239CE1: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243==    by 0x15FABA4F: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15FACD27: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x15E555F2: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.27)
==2243==    by 0x8223486: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4002.0)
==2243== 
Loading "libopentrack-filter-accela.so" 
Loading "libopentrack-filter-ewma.so" 
==2243== Invalid free() / delete / delete[] / realloc()
==2243==    at 0x4C2C2BC: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==2243==    by 0x21EA212F: FTNoIR_FilterDll::~FTNoIR_FilterDll() (in /usr/local/libopentrack-filter-kalman.so)
==2243==    by 0x433706: dylib::dylib(QString const&, dylib::Type) (in /usr/local/opentrack.bin)
==2243==    by 0x443617: _ZN9__gnu_cxx13new_allocatorI5dylibE9constructIS1_IR7QStringRNS1_4TypeEEEEvPT_DpOT0_ (in /usr/local/opentrack.bin)
==2243==    by 0x44333F: _ZNSt16allocator_traitsISaI5dylibEE12_S_constructIS0_IR7QStringRNS0_4TypeEEEENSt9enable_ifIXsrNS2_18__construct_helperIT_IDpT0_EEE5valueEvE4typeERS1_PSA_DpOSB_ (in /usr/local/opentrack.bin)
==2243==    by 0x442EF3: _ZNSt16allocator_traitsISaI5dylibEE9constructIS0_IR7QStringRNS0_4TypeEEEEDTcl12_S_constructfp_fp0_spcl7forwardIT0_Efp1_EEERS1_PT_DpOS8_ (in /usr/local/opentrack.bin)
==2243==    by 0x44273F: std::_Sp_counted_ptr_inplace<dylib, std::allocator<dylib>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<QString&, dylib::Type&>(std::allocator<dylib>, QString&, dylib::Type& [... cut ...]
==2243==    indirectly lost: 22,087 bytes in 835 blocks
==2243==      possibly lost: 330,254 bytes in 755 blocks
==2243==    still reachable: 1,658,633 bytes in 18,768 blocks
==2243==         suppressed: 0 bytes in 0 blocks
==2243== Rerun with --leak-check=full to see details of leaked memory
==2243== 
==2243== For counts of detected and suppressed errors, rerun with: -v
==2243== Use --track-origins=yes to see where uninitialised values come from
==2243== ERROR SUMMARY: 5 errors from 5 contexts (suppressed: 0 from 0)

bar0nP, 2016-01-31:
I don't understand last paragraph. What utility for I must use option -g?
Sorry, I'm not good in source code works (as you see).

sthalik, 2016-02-01:
> I don't understand last paragraph. What utility for I must use option -g?

`-g` will give line numbers.

    ==2243== by 0x21EA1467: FTNoIR_FilterDll::getShortName(QString) (in /usr/local/libopentrack-filter-kalman.so)

This doesn't make sense. There's no `getShortName` member function in the Kalman filter. I can't provide support unless you build from my source.

bar0nP, 2016-02-01:
Thank, it's better now, but I don't know where are use that option. Installing opentrack by steps:
1. cmake CMakeLists.txt
2. make
3. sudo make install
I need to write "make -g"? But with option is absent in "man make". Or I must to add "-g" in some configuration file?
Sorry for stupid question.

sthalik, 2016-02-09:
First of all, create a build subdirectory.

`mkdir build; cd build; cmake ..`

Add debug info by running `ccmake .` (you may need to install `cmake-curses` or a similarly-named package. You need to add `-g` to both `CMAKE_C_FLAGS` and `CMAKE_CXX_FLAGS`.

Finally, unless you check out one of the branches (`stable/2.2` or `unstable`) and not have some inconsistent tree I won't be able to help you. If your working tree is already a mess, remove the crap `git clean -dfx`, sync the tree `git fetch -all`, checkout the branch, then verify you're working on a clean tree `git status`.

sthalik, 2016-05-06:
Closing due to inactivity. You can reopen with new comments.

closed 2016-05-06 by sthalik