opentrack/opentrack #310: Tobii EyeX eye tracking support

sthalik, 2016-02-09:
There's an SDK at <<http://developer.tobii.com/>>

For eye tracking we need both "accumulative" and "fov extension" methods. I have the general outline in an email conversation.

sthalik, 2016-05-15:
Status update. I have the hardware. My asmedia usb 3.0 controller doesn't have enough bandwidth. A friend has a proper controller and is willing to help. It's going to take more time than I thought previously.

sthalik, 2018-08-03:
Cancelled due to obsolete hardware and an API change.

closed 2018-08-03 by sthalik

thor, 2020-09-28:
Would it be relevant or interesting to create a new issue for the Tobii EyeTracker 4C and 5?

sthalik, 2020-09-28:
See #1037. There's no copyright information (illegal for me to include it), and it hasn't been tested by a third party.