opentrack/opentrack #311: Yaw Axis Inverts

SoulMaril, 2016-02-10:
Sometimes at extreme angles Yaw axis inverts. I have to restart.

sthalik, 2016-02-11:
Do you use dynamic pose resolution feature in pointtracker? If so, what's the timeout?

SoulMaril, 2016-02-11:
Yes, 500ms.

SoulMaril, 2016-02-13:
I tired to fix this by bind keys to "start tracking" "stop tracking" and "toggle tracking". but they didnt work within the game. Only stop and start is working when the opentrack is in focus. Toggle tracking never worked.

The tracking inverts when I move my head more than 90 degrees. when I close one of the leds by my hand. The tracker do not invert.

sthalik, 2016-02-17:
This is related to dynamic point resolution and the software distinguishing LEDs from each other in general.

- you yaw more than 90 degrees, one of LEDs becomes invisible
- dynamic pose resolution timer fires, it gets reset
- LED becomes visible, but closer to 90 degrees than center
- dynamic pose resolution state is invalid, as it got initialized close to 90 degrees yaw, not center

You can use two solutions here

- cover the LED for a second while the model is centered so that dynamic pose resets again
- use the "toggle tracking" keybinding and tap it twice

There's no correct mathematical solution for the distinguishing LEDs problem.

closed 2016-02-17 by sthalik