opentrack/opentrack #312: PointTracker 1.1 crash issue

marcusdoesstuff, 2016-02-10:
Installed 2.3-rc21p8. PointTracker 1.1 causes crash upon camera activation.

marcusdoesstuff, 2016-02-10:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	opentrack.exe
  Application Version:	0.0.0.0
  Application Timestamp:	56ba706b
  Fault Module Name:	Qt5Core.dll
  Fault Module Version:	5.5.1.0
  Fault Module Timestamp:	5631f7fe
  Exception Code:	c0000005
  Exception Offset:	0001a1d0
  OS Version:	6.3.9600.2.0.0.768.101
  Locale ID:	2057
  Additional Information 1:	5861
  Additional Information 2:	5861822e1919d7c014bbb064c64908b2
  Additional Information 3:	a10f
  Additional Information 4:	a10ff7d2bb2516fdc753f9c34fc3b069

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=280262

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-GB\erofflps.txt

marcusdoesstuff, 2016-02-10:
Fixed by making new profile. Here's a snip of the default profile that was making it crash though if it's any use...
[image]

marcusdoesstuff, 2016-02-10:
Further experimentation shows that crashes occur when starting after adjusting Point extraction > Threshold away from new profile default. It is possible to start, adjust threshold and it won't crash but it will once process is stopped and started again.

sthalik, 2016-02-17:
Download this build - https://db.tt/IlPpjY7T

Change to the "gdb" subdirectory and run the batch file.

- Type `run`
- Make it crash
- Type `thread apply all bt`

Send me the output.

marcusdoesstuff, 2016-02-22:
I don't seem to be getting the same error but the program is hanging without an error now...

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb>cd /d E:\Downloads\opentrack
-2.3-rc21p8-DEBUG\install\gdb\

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb>cd ..

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install>gdb\bin\gdb.exe --nh --exec=open
track.exe
E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb\bin\gdb.exe: warning: cannot
 automatically find executable file or library to read symbols.
Use "file" or "dll" command to load executable/libraries directly.
GNU gdb (GDB) 7.10.50.20151207-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) run
Starting program: E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\opentrack.exe

[New Thread 7560.0x1a88]
warning: Can not parse XML library list; XML support was disabled at compile tim
e
[New Thread 7560.0xdd4]
warning: Loading "libopentrack-filter-accela.dll"
warning: Loading "libopentrack-filter-ewma.dll"
warning: Loading "libopentrack-filter-kalman.dll"
warning: Loading "libopentrack-tracker-aruco.dll"
warning: Loading "libopentrack-tracker-freepie-udp.dll"
warning: Loading "libopentrack-tracker-hatire.dll"
warning: Loading "libopentrack-tracker-hydra.dll"
warning: Loading "libopentrack-tracker-joystick.dll"
warning: Loading "libopentrack-tracker-pt.dll"
warning: Loading "libopentrack-tracker-rift-025.dll"
warning: Loading "libopentrack-tracker-rift-042.dll"
warning: Loading "libopentrack-tracker-rift-080.dll"
warning: Loading "libopentrack-tracker-rs.dll"
warning: Loading "libopentrack-tracker-udp.dll"
warning: Loading "libopentrack-proto-fgfs.dll"
warning: Loading "libopentrack-proto-freetrack.dll"
warning: Loading "libopentrack-proto-fsuipc.dll"
warning: Loading "libopentrack-proto-simconnect.dll"
warning: Loading "libopentrack-proto-udp.dll"
warning: Loading "libopentrack-proto-vjoy.dll"
warning: Loading "libopentrack-proto-win32-mouse.dll"
warning: bundle + "opentrack-ui"
warning: joy list refresh
[New Thread 7560.0xa8]
[New Thread 7560.0x1ebc]
[New Thread 7560.0x1d78]
[New Thread 7560.0x1ae8]
[New Thread 7560.0x17b4]
warning: add receiver 49561848
warning: joy list refresh
warning: bundle + "tracker-pt"
warning: failed CoInitializeEx -2147417850
warning: bundle - "tracker-pt"
warning: bundle + "proto-freetrack"
[New Thread 7560.0x1b24]
warning: Application "\??\E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\TrackI
R.exe" found in cache
warning: bundle + "tracker-pt"
warning: bundle + "Accela"
[New Thread 7560.0x1bac]
warning: add receiver 153813480
warning: joy list refresh
warning: Tracker:: Applying settings
[New Thread 7560.0x187c]
[New Thread 7560.0x1f10]
[New Thread 7560.0x1364]
[Thread 7560.0x1364 exited with code 0]
[New Thread 7560.0x1bd0]
[Thread 7560.0x1bd0 exited with code 0]
[New Thread 7560.0xc84]
[New Thread 7560.0x1ddc]
[Thread 7560.0xc84 exited with code 0]
[Thread 7560.0x1ddc exited with code 0]
[New Thread 7560.0x1da4]
[Thread 7560.0x1da4 exited with code 0]
[New Thread 7560.0x1e54]
[Thread 7560.0x1e54 exited with code 0]
[New Thread 7560.0x1d84]
[New Thread 7560.0x560]
warning: Tracker::apply ends
[New Thread 7560.0x1a80]
warning: failed CoInitializeEx -2147417850
warning: TrackerDialog:: Tracker registered
warning: Tracker:: Applying settings
warning: pt: freeing camera
[Thread 7560.0x1d84 exited with code 0]
[Thread 7560.0x1a80 exited with code 0]
[Thread 7560.0x560 exited with code 0]
warning: pt camera: assuming stopped
[New Thread 7560.0x1f9c]
[Thread 7560.0x1f9c exited with code 0]
[New Thread 7560.0x1e4c]
[Thread 7560.0x1e4c exited with code 0]
[New Thread 7560.0x1f38]
[New Thread 7560.0x10dc]
[Thread 7560.0x1f38 exited with code 0]
[Thread 7560.0x10dc exited with code 0]
[New Thread 7560.0x750]
[Thread 7560.0x750 exited with code 0]
[New Thread 7560.0x1de4]
[Thread 7560.0x1de4 exited with code 0]
[New Thread 7560.0x1e20]
[New Thread 7560.0xac0]
warning: Tracker::apply ends
[New Thread 7560.0x1ee0]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 7560.0x1bac]
0x6695a1d0 in ?? ()
(gdb)

marcusdoesstuff, 2016-02-22:
Another time, when I just tried at 75Hz, it seemed to work with the debug tool you sent:

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb>cd /d E:\Downloads\opentrack
-2.3-rc21p8-DEBUG\install\gdb\

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb>cd ..

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install>gdb\bin\gdb.exe --nh --exec=open
track.exe
E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb\bin\gdb.exe: warning: cannot
 automatically find executable file or library to read symbols.
Use "file" or "dll" command to load executable/libraries directly.
GNU gdb (GDB) 7.10.50.20151207-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) run
Starting program: E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\opentrack.exe

[New Thread 4884.0x798]
warning: Can not parse XML library list; XML support was disabled at compile tim
e
[New Thread 4884.0x824]
warning: Loading "libopentrack-filter-accela.dll"
warning: Loading "libopentrack-filter-ewma.dll"
warning: Loading "libopentrack-filter-kalman.dll"
warning: Loading "libopentrack-tracker-aruco.dll"
warning: Loading "libopentrack-tracker-freepie-udp.dll"
warning: Loading "libopentrack-tracker-hatire.dll"
warning: Loading "libopentrack-tracker-hydra.dll"
warning: Loading "libopentrack-tracker-joystick.dll"
warning: Loading "libopentrack-tracker-pt.dll"
warning: Loading "libopentrack-tracker-rift-025.dll"
warning: Loading "libopentrack-tracker-rift-042.dll"
warning: Loading "libopentrack-tracker-rift-080.dll"
warning: Loading "libopentrack-tracker-rs.dll"
warning: Loading "libopentrack-tracker-udp.dll"
warning: Loading "libopentrack-proto-fgfs.dll"
warning: Loading "libopentrack-proto-freetrack.dll"
warning: Loading "libopentrack-proto-fsuipc.dll"
warning: Loading "libopentrack-proto-simconnect.dll"
warning: Loading "libopentrack-proto-udp.dll"
warning: Loading "libopentrack-proto-vjoy.dll"
warning: Loading "libopentrack-proto-win32-mouse.dll"
warning: bundle + "opentrack-ui"
warning: joy list refresh
[New Thread 4884.0x1f88]
[New Thread 4884.0x1f54]
[New Thread 4884.0x1cc4]
[New Thread 4884.0x1974]
[New Thread 4884.0x1dc8]
warning: add receiver 50151672
warning: joy list refresh
warning: bundle + "tracker-pt"
warning: failed CoInitializeEx -2147417850
warning: bundle - "tracker-pt"
warning: bundle + "proto-freetrack"
[New Thread 4884.0x1b14]
warning: Application "\??\E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\TrackI
R.exe" found in cache
[... 97 lines cut ...]
[New Thread 4884.0x14b4]
warning: remove receiver 152957968
[Thread 4884.0x1d68 exited with code 0]
warning: Tracker:: Thread stopping
[Thread 4884.0x408 exited with code 0]
warning: pt: freeing camera
[Thread 4884.0x774 exited with code 0]
[Thread 4884.0x14b4 exited with code 0]
[Thread 4884.0x1bc exited with code 0]
warning: pt camera: assuming stopped
warning: bundle - "tracker-pt"
warning: bundle - "Accela"
[Thread 4884.0x1b14 exited with code 0]
warning: bundle - "proto-freetrack"
warning: remove receiver 50151672
warning: bundle - "opentrack-ui"
[Thread 4884.0x1cc4 exited with code 0]
[Thread 4884.0x15a8 exited with code 0]
[Thread 4884.0x1dc8 exited with code 0]
[Thread 4884.0x1974 exited with code 0]
[Thread 4884.0x1f54 exited with code 0]
[Thread 4884.0x16c0 exited with code 0]
[Thread 4884.0x824 exited with code 0]
[Inferior 1 (process 4884) exited normally]
(gdb)

marcusdoesstuff, 2016-02-22:
Another hang (turns out Hz didn't make any difference as this was 75Hz too):

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb>cd /d E:\Downloads\opentrack
-2.3-rc21p8-DEBUG\install\gdb\

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb>cd ..

E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install>gdb\bin\gdb.exe --nh --exec=open
track.exe
E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\gdb\bin\gdb.exe: warning: cannot
 automatically find executable file or library to read symbols.
Use "file" or "dll" command to load executable/libraries directly.
GNU gdb (GDB) 7.10.50.20151207-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) run
Starting program: E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\opentrack.exe

[New Thread 4504.0xe50]
warning: Can not parse XML library list; XML support was disabled at compile tim
e
[New Thread 4504.0x1c40]
warning: Loading "libopentrack-filter-accela.dll"
warning: Loading "libopentrack-filter-ewma.dll"
warning: Loading "libopentrack-filter-kalman.dll"
warning: Loading "libopentrack-tracker-aruco.dll"
warning: Loading "libopentrack-tracker-freepie-udp.dll"
warning: Loading "libopentrack-tracker-hatire.dll"
warning: Loading "libopentrack-tracker-hydra.dll"
warning: Loading "libopentrack-tracker-joystick.dll"
warning: Loading "libopentrack-tracker-pt.dll"
warning: Loading "libopentrack-tracker-rift-025.dll"
warning: Loading "libopentrack-tracker-rift-042.dll"
warning: Loading "libopentrack-tracker-rift-080.dll"
warning: Loading "libopentrack-tracker-rs.dll"
warning: Loading "libopentrack-tracker-udp.dll"
warning: Loading "libopentrack-proto-fgfs.dll"
warning: Loading "libopentrack-proto-freetrack.dll"
warning: Loading "libopentrack-proto-fsuipc.dll"
warning: Loading "libopentrack-proto-simconnect.dll"
warning: Loading "libopentrack-proto-udp.dll"
warning: Loading "libopentrack-proto-vjoy.dll"
warning: Loading "libopentrack-proto-win32-mouse.dll"
warning: bundle + "opentrack-ui"
warning: joy list refresh
[New Thread 4504.0x1938]
[New Thread 4504.0x1f0c]
[New Thread 4504.0x1fdc]
[New Thread 4504.0x1d2c]
[New Thread 4504.0xdf8]
warning: add receiver 50348280
warning: joy list refresh
warning: bundle + "tracker-pt"
warning: failed CoInitializeEx -2147417850
warning: bundle - "tracker-pt"
warning: bundle + "proto-freetrack"
[New Thread 4504.0xbd0]
warning: Application "\??\E:\Downloads\opentrack-2.3-rc21p8-DEBUG\install\TrackI
R.exe" found in cache
warning: bundle + "tracker-pt"
warning: bundle + "Accela"
[New Thread 4504.0x784]
warning: add receiver 153033464
warning: joy list refresh
warning: Tracker:: Applying settings
[New Thread 4504.0x1e4c]
[New Thread 4504.0x1dac]
[New Thread 4504.0x1db4]
[Thread 4504.0x1db4 exited with code 0]
[New Thread 4504.0x878]
[Thread 4504.0x878 exited with code 0]
[New Thread 4504.0x1b5c]
[New Thread 4504.0x268]
[Thread 4504.0x1b5c exited with code 0]
[Thread 4504.0x268 exited with code 0]
[New Thread 4504.0x15a0]
[Thread 4504.0x15a0 exited with code 0]
[New Thread 4504.0x10cc]
[Thread 4504.0x10cc exited with code 0]
[New Thread 4504.0x1c78]
[New Thread 4504.0x1220]
warning: Tracker::apply ends
[New Thread 4504.0x6a0]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 4504.0x784]
0x6695a1d0 in ?? ()
(gdb)

marcusdoesstuff, 2016-02-22:
Once the debugger hangs, it can only be closed via the task manager.

sthalik, 2016-02-25:
@marcusdoesstuff do it differently:

- run opentrack normally
- wait for it to crash, don't close the crash window
- open a Windows console (cmd.exe)
- type `tasklist`
- grab the PID for opentrack.exe
- run `gdb`
- in `gdb` type `attach pid` where `pid` is the numeric value from tasklist

You should be able to now access `thread apply all bt`

sthalik, 2016-03-07:
See if opentrack-2.3 rc21p9 fixes.

sthalik, 2016-05-10:
Any luck @marcusdoesstuff? I need more info. If you come back WRT this issue reopen please.

closed 2016-05-10 by sthalik