opentrack/opentrack #314: Opentrack doesn't work with PS3 Eye

RoarGi, 2016-02-12:
Hi everybody! I hope someone will help me with my issue.

Opentrack  running on Windows 7(without any service pack) doesn't start PS3 Eye when I push start button. In "Tracker" optoins PS3 Eye selected, but it just don't start.
Tried to fix this by using different combinations of versions.
PS3 Eye drivers was installed:
CL-Eye-Driver-4.0.2.1017
CL-Eye-Driver-5.0.1.0218
CL-Eye-Driver-5.3.0.0341
Opentrack versions:
opentrack-2.3-rc19p3-win32
opentrack-2.3-rc21p6-win32
opentrack-2.3-rc21p8-win32
opentrack-2.3-rc4
and maybe some of unstable versions.

CL-Test, Skype and FreeTrack 2.2 work well.

Thank you in advance for your response.

Len62, 2016-02-12:
Are you running Opentrack as an Administrator ? If not, try that. I don't have any problems with my PS3 eye and Opentrack. I'm on windows 10 but that should not matter.

RoarGi, 2016-02-12:
Len62, thank you for your answer. I tried that. No result.
Looks like it's very rare bug and may be related to my system because i cannot find any problems like mine in the internet.

sthalik, 2016-02-12:
Try replug the camera into a different USB port. The current port is fine, just that changing the port makes windows do some things.

RoarGi, 2016-02-12:
Changed some ports USB 2.0 and USB 3.0. No effect.
Also, when I try to open Camera settings dialog in Tracker options nothing happens. Looks like opentrack cannot access camera driver or something.

MathijsG, 2016-02-14:
Maybe try to upgrade to service pack 1? Any particular reason why you stayed with 7 without service pack 1 anyway?

alterscape, 2016-02-14:
Do you have a driver for the PS3 Eye camera installed? I know it's a basic thing, but in case you aren't aware: the PS3 Eye camera doesn't work out of the box with Windows, you need a driver (Code Labs' is the one most people use -- he's charging $5 or some other nominal amount now).

RoarGi, 2016-02-14:
MathijsG, maybe I should. I'll write here in any case when I get updated.
alterscape, sure. In OP I wrote 3 versions of driver I tried.

Thanks for your attention to my problem. I'll inform you if something change.

sthalik, 2016-02-17:
Are you using any antivirus software? Try disabling it completely entirely.

Archibas, 2016-02-24:
I have Opentrack does not work with PS3 Eye , too.

sthalik, 2016-02-25:
Do you have any antivirus software running?

sthalik, 2016-02-25:
@Archibas  Do you have any antivirus software running? Also what Windows version?

Archibas, 2016-02-25:
Yes , I turned off anti-virus. Windows 7 64 
maximum. It did not help.

Archibas, 2016-02-25:
I tried to connect to FreeTrack camera works .

RoarGi, 2016-02-25:
@Archibas , I have same problem. Tried switching off antivirus software but no effect.
FreeTrack and FaceTrackNoIR works fine.
Interesting thing that when I try to capture video with Sony Vegas it cannot connect PS3 Eye too.

Archibas, 2016-02-25:
Yes, the only problem with opentrask. It's a shame.

Archibas, 2016-02-25:
Да   у меня Sony Vegas it cannot connect PS3 Eye too.

Archibas, 2016-02-26:
With only two people with the same problem ?

Len62, 2016-02-26:
Since Sony Vegas cannot connect either I would assume it has to be some other issue then with Opentrack itself. This is a stumper for sure. Maybe an issue with Windows 7?

Archibas, 2016-02-27:
Maybe. Even likely . But that hinders normal operation.

clovenlife, 2016-04-29:
I am also have this issue. Opentrack appears to connect to camera but raw data is black. I have no access to camera settings sliders. It opens other cameras and reads fine. PS3 camera works with other software such as skype.

sthalik, 2016-04-29:
Change USB port.

clovenlife, 2016-04-29:
changed usb port, uninstalled drivers, reinstalled drivers, updated opentrack, tried previous versions of opentrack. Opentrack had worked flawlessly for months before this new issue.

Archibas, 2016-05-04:
Again it began. Installed WIN 10 Opentrask worked yesterday. Today, I stopped working. No black window capture. All porvtorilos as in windows 7. No public program is not installed.

sthalik, 2016-05-04:
Is it broken with the exact same opentrack version you used, or did the problem start occuring in a new version only?

clovenlife, 2016-05-04:
I should add, I've worked around this issue by using accessing the ps eye camera indirectly through manycam software's virtual webcam

Archibas, 2016-05-05:
I confirm that after installing ManyCam opentrack started working .

sthalik, 2016-05-05:
I can't help you guys. It works for thousands of users. Furthermore, details such as DirectShow and codecs are brittle things just asking to break at any time.

If you can provide a real solution, please do. I'll keep the issue open for the time being.

Archibas, 2016-05-05:
The problem observed in the new and old versions . opentrack-2.3-rc21p11, opentrack-2.3-rc21p10-win32. But I start with opentrack-2.3-rc21p11. It is strange but the problem persists as the windows 7 and the windows 10 .

sthalik, 2016-05-05:
Can you reproduce with opentrack 2.2?

There are some changes to the build environment back-and-forth between rc's but shouldn't be the cause.

Archibas, 2016-05-05:
Yes, but I try for the evening . Maybe you can collect logs of work?

Archibas, 2016-05-05:
opentrack I started working after installing ManyCam . ManyCam run but do not just install it all worked.

Archibas, 2016-05-05:
I try to remove ManyCam. Let's see what happens.

Archibas, 2016-05-05:
Yes, after the removal of ManyCam opentrack stopped working .

sthalik, 2016-05-06:
It can be a bug in opentrack, but the way we open the camera is very common.

This bit of code in the external library is very complex, due to innate complexity of video input handling in Windows in general.

It's most probable that it's a problem with your system configuration, e.g. codecs. I've been unable to reproduce that on a fresh install of Windows 10, 8.1 and so forth. Only K-Lite codec pack Mega was installed.

sthalik, 2016-05-06:
I can only help given technical basis of what we're doing wrong. Code is always welcome.

clovenlife, 2016-05-06:
Thank you for mentioning codecs. I had installed the SAF codec pack to fix an issue with my PVR system. I uninstalled the codec pack and opentrack now functions as intended.

sthalik, 2016-05-07:
@clovenlife this is important information, thanks for sharing that.

My personal recommendation: install K-Lite codec pack biggest version on fresh Windows and don't touch codecs ever again at all. Then they won't break.

RoarGi, 2016-08-05:
Hi everyone! My problem was solved by installing ManyCam! Thank you!

sthalik, 2016-10-06:
That's most likely a codec problem, but it's good that it works with for you with a workaround at least.

closed 2016-10-06 by sthalik