opentrack/opentrack #315: Unhandled exception crash

crucified77, 2016-02-14:
I get the "opentrack.exe has stopped working" window Opentrack crashes right away after pressing "Start".  My configuration has 'Main tracker' as Joystick and fails for any 'Game protocol'.  I've tried different joystick inputs (not just EDTracker) and it still crashes).  Reboots didn't help.

Seen with opentrack-2.2 and opentrack 2-3-rc21p8

opentrack-2.2 release crash:

Unhandled exception at 0x3E001810 in opentrack.exe: 0xC0000005: Access violation reading location 0x3E001810.

Call stack:
>	3e001810()	Unknown
 	[Frames below may be incorrect and/or missing]	
 	Qt5Core.dll!6695de8a()	Unknown

 	[External Code]	

Breaking I did see something else: 
Unhandled exception at 0x51AD3C43 (dinput8.dll) in opentrack.exe: 0xC0000005: Access violation reading location 0xD6DB2365.

The opentrack 2-3 debug info is slightly different: 
The thread 0x2cb4 has exited with code 0 (0x0).
Unhandled exception at 0x3E001810 in opentrack.exe: 0xC0000005: Access violation reading location 0x3E001810.

E:\Users\David\Downloads\opentrack-2.3-rc21p8\install

>	3e001810()	Unknown
 	[Frames below may be incorrect and/or missing]	
 	libopentrack-api.dll!6eb4b174()	Unknown
 	libopentrack-tracker-joystick.dll!68642446()	Unknown
 	[External Code]	
 	libopentrack-api.dll!6eb4a944()	Unknown
 	Qt5Core.dll!66960b3e()	Unknown
 	[External Code]

sthalik, 2016-02-17:
Download this build - https://db.tt/IlPpjY7T

Change to the "gdb" subdirectory and run the batch file.

- Type `run`
- Make it crash
- Type `thread apply all bt`

Send me the output.

crucified77, 2016-02-17:
Thanks for the help, I tried it but I wasn't able to get it working.  First, I was able to download and extract that debug build.  Then I opened a cmd window and ran the 'run-with-debugger.bat' in the gdb subfolder.  Then I went to the debug install folder and double clicked on opentrack.exe, and it didn't open.  I only saw the circle that says something is loading but it doesn't run (and it locked up the window).  Rebooting and retrying didn't help.  Running as administrator didn't help.  Attached is a screenshot of the cmd window and the folder.  Thanks

--------------------------------------------

 

 
 
 
 
 
 
 
 
 it on GitHub.

sthalik, 2016-02-17:
run-with-debugger.bat should automatically start gdb.exe and `run` should run opentrack with the debugger attached.

crucified77, 2016-02-17:
When I ran the script there was no way to enter "run".  The screenshot I sent shows the state but doesn't show pressing keys didn't echo back anything.  Should I try something else?   

--------------------------------------------

 

 
 it on GitHub.

sthalik, 2016-02-17:
I haven't received any screenshots. Attach them through github's web site.

You can also run gdb.exe manually, then `file ../opentrack.exe` and proceed as before.

crucified77, 2016-02-18:
No luck with the gdb from the zip you sent but I installed gdb from MinGW.  It hangs on trying to start opnetrack:

(gdb) file opentrack.exe
Reading symbols from e:\Users\David\Downloads\opentrack-2.3-rc21p8-DEBUG\install\opentrack.exe...done.
(gdb) run
Starting program: e:\Users\David\Downloads\opentrack-2.3-rc21p8-DEBUG\install/opentrack.exe opentrack
<the curser is frozen and no opentrack is opened>

Clicking on opentrack.exe outside of gdb it doesn't open as well.

--------------------------------------------

 

 
 
 it on GitHub.

sthalik, 2016-02-18:
Are you sure the files were extracted properly? Here's an md5 sum for opentrack.exe: `ef2bf34798b5320fc96bc5d11c83a354`.

That you can't open gdb.exe either makes me think the extracted directory contents are corrupted.

The build was made with the same process releases are made, and adding debug info is pretty much known not to cause additional problems. I'm able to run this build I sent as well.

crucified77, 2016-02-19:
I verified that md5 checksum is the same.  Also I extracted it again and tried running opentrack.exe and saw it hang.  The latest released version I downloaded came up.  

I assume the difference between your machine and mine is the installed dependencies like Qt.  Perhaps I need to install all the dependencies and build it on my machine?

---
David Weichelt
http://www.djweichelt.net/

--------------------------------------------

 

 
 
 
 it on GitHub.

crucified77, 2016-02-19:
I've cloned the latest git repo and am installing Qt 5.5/5.4.  After that, I'm hoping I can build this and get it working so I can get more information.  Am I on the right track?

sthalik, 2016-02-19:
All the dependencies are included in the releases, including the debug build.

I downloaded the debug .zip on a fresh Windows installation and it works fine.

Installing the dependencies is a problem for non-programmers - you'd need to actually compile Qt on your own using the same compiler version as opentrack. I'd recommend against spending time trying to deal with compiling, especially (as I assume) lacking the prerequisite programming knowledge.

From your messages I don't know what's the cause of the debug build not starting. It can be file and directory permissions in Windows, antivirus software acting up, many things.

For the permissions part - since you have MinGW installed already, run `msys/1.0/msys.bat` to open a terminal, then

`
# first navigate to opentrack debug build's directory
chown -R $USERNAME .
chmod -R u+rwx .
`

This is just an idea of what might be wrong. On my part I can say that the builds are pretty much identical, except for debug information which doesn't affect executed code at all.

crucified77, 2016-02-19:
Okay, I'll try this later.  Thanks again for the help!  I do hope this can work so I can use my EDTracker across more games.  It would be nice to figure out what's going wrong (need some logs).  

I am a Software Engineer for John Deere by the way.  I could attempt a build if we can't get the debug build to run.  Not sure exactly how to do this but I see a CMake file can can try running with MinGW (or QtCreator).  I'm not sure if I need to install those dependencies or what version of Qt.  Are there some detailed build steps I'm missing?
--------------------------------------------

 

 
 
 
 
 
 
 
 
 
 
 
 it on GitHub.

sthalik, 2016-02-19:
Hey,

There's some information on <<https://github.com/opentrack/opentrack/wiki/building-on-cygwin-targetting-windows>> that should get you started.

You may run into problems with compiling Qt5 due to mingw-w64 and lack of a shell and other basic tools. I recommend installing both MinGW and mingw-w64, then starting a MinGW msys shell and adjusting PATH so that mingw-w64 compilers are up front.

The GNU compilers are here used to make the builds. Sometime earlier I verified that MSVC++ 2015 works, but it's not checked on a regular basis. In case you're more familiar with the Microsoft compiler it can be useful. If there are build or runtime errors with MSVC++ 2015 it's a bug in opentrack that needs fixing of course.

SoulMaril, 2016-02-21:
Same here. Today I got crashes right after it started. Didnt changed anything. A default profile worked for a while then it started to crash too. Then I swiched to my personal profile. It worked for a while than started to crash again.

sthalik, 2016-02-22:
@SoulMaril please provide debug instructions as in https://github.com/opentrack/opentrack/issues/315#issuecomment-184991604

sthalik, 2016-02-25:
@crucified77 @SoulMaril see <<https://github.com/opentrack/opentrack/issues/312#issuecomment-188633130>>

sthalik, 2016-02-25:
@crucified77 @SoulMaril you still need a debug build https://db.tt/IlPpjY7T

crucified77, 2016-02-28:
Thanks for the help sthalik.  I tried chown in mingw32 and the command isn't implemented but the permissions and owner seem fine when I do an 'ls -laX'.  properties in windows show the ownership is setup as expected to David Administrators.  As far as building it, I haven't been able to, as I've not had the time and energy.  Today I tried to see if EDTracker is supported in Aussetto Corsa but it wasn't supported, so I would like to see OpenTracker work.  

BTW: I retried the steps using cmd as Admin with mingw32 and gdb but it hung as before on run of opentrack.exe.  

[opentrackAttemptToDebug.txt](https://github.com/opentrack/opentrack/files/149773/opentrackAttemptToDebug.txt)

sthalik, 2016-03-07:
See if opentrack-2.3 rc21p9 fixes.

crucified77, 2016-03-07:
This doesn't run for me.  I'll I've seen was the spinning windows logo after clicking opentrack.exe (and saw that same thing with the setup exe).  I've rebooted, turned off SmartScreen, ran as Administrator, and no luck.

I wasn't sure if something had happened to my system so I retried the older version of opentrack (opentrack-2.3-rc21p8) and it came up.  It does crash after pressing Start but that seems to be another issue.  

This reminds me of what I see when I tried to run the opentrack-2.3-rc21p8-DEBUG version.  

Thanks...

--------------------------------------------

 

 
 it on GitHub.

sthalik, 2016-03-07:
Hey,

I've ninja updated the .zip file with changed permissions. Files inside are the same. Can you confirm it fixes anything?

My install directory had permissions only for the user himself and no one else, even to read or execute. 

Also note that on Windows lack of permissions for e.g. SYSTEM or Administrator won't let that user do stuff, other than override ownership and replace permissions. It's a mess compared to Unix systems.

crucified77, 2016-03-10:
The latest release version worked opens the app (I used the exe to install it).  It still crashes after I press Start with my EDTrackerPro though.  If you have a new DEBUG version I could try it and get you some logs.  Perhaps the debug version will work after applying the same fix you did here.

sthalik, 2016-03-10:
Here's a build with debugging functionality: <https://www.dropbox.com/s/l0e7vyneq8s41dc/opentrack-DEBUG-2.3-rc21p10.7z>

When it crashes and a Windows popup happens asking to close the app, go to `gdb` directory and run `gdb.bat`. In the same directory as with `gdb.bat` a file named after a number will appear. Send this file here.

Don't click 'close' in the Windows crash dialog, first run gdb.bat otherwise the information is lost.

Thanks!

crucified77, 2016-03-11:
I unzipped it to a folder and double clicked on opentrack.exe and it just hangs.  See the spinning circle but that's it.  

--------------------------------------------

 

 
 
 
 
 it on GitHub.

sthalik, 2016-03-12:
@crucified77 use the installer, it sets correct permissions.

https://www.dropbox.com/s/qad74rn1meyll8p/DEBUG-opentrack-2.3-rc21p10-3-g4733796-dirty-win32-setup.exe

crucified77, 2016-03-15:
I opened the file and it just hung.  I didn't open any setup wizard or install anything.  I only see the rotating circle icon over the document folder.

sthalik, 2016-03-15:
@crucified77 people previously reported these kinds of symptoms with archives. The installer's a full installation, except a big bigger due to all the debug stuff in it.

crucified77, 2016-06-28:
I just installed the latest version (2.3-rc49p7-win32) and I have the same problem.  When ever I select joystick with an input (ED Tracker or my other joystick) OpenTrack crashes after I press Start.  If I try other protocols I don't see the problem.  If I can provide any debug information, that could help diagnose I would.

sthalik, 2016-06-28:
@crucified77 does it also happen when setting joystick shortcuts in options menu?

crucified77, 2016-06-28:
Hi Stanislaw, thanks for reopening and the quick reply.  In the Options I am trying to bind shortcuts but it doesn't respond to the ED Tracker joystick input.  Is that where I'm to try?

crucified77, 2016-06-28:
I was able to get some debug info from MS Visual Studio:

Unhandled exception at 0x3D001810 in opentrack.exe: 0xC0000005: Access violation reading location 0x3D001810.
>	3d001810()	Unknown
 	[Frames below may be incorrect and/or missing]	
 	libopentrack-logic.dll!002cb73c()	Unknown
 	libopentrack-tracker-joystick.dll!667893fc()	Unknown
 	[External Code]	
 	libopentrack-logic.dll!002cbe51()	Unknown
 	libopentrack-logic.dll!002cbe51()	Unknown
 	Qt5Core.dll!6695d211()	Unknown
 	Qt5Core.dll!6695d211()	Unknown
 	[External Code]

sthalik, 2016-06-28:
@crucified77 first see https://www.dropbox.com/s/swybqsdere07npo/opentrack-test-20160628_1.7z if there are any interesting issues in the debug window. Joystick input looked fine but alas, it had some edge cases. If it's still broken then we'll work with debug builds.

sthalik, 2016-06-28:
By "debug window" I mean the console window that pops up in that particular build.

crucified77, 2016-07-01:
I ran that debug version and it crashed.  When Virtual Studio came up to debug I saw it was on this line: 
>	dinput8.dll!_CDIDev_GetDeviceState@12()	Unknown

The output had:
The thread 0x16cc has exited with code 0 (0x0).
Unhandled exception at 0x51083C43 (dinput8.dll) in opentrack.exe: 0xC0000005: Access violation reading location 0x54349F41.

I see assembly code at that address:
51083C42  ?? ?? 
51083C43  idiv        eax,dword ptr [esi+5010A151h]  
51083C49  or          dword ptr [ecx+33h],edx  
51083C4C  lds         ecx,fword ptr [ecx+5653FC45h]  
51083C52  mov         esi,dword ptr [ebp+8]  
51083C55  push        edi  
51083C56  mov         edi,esi  
51083C58  push        ecx  
51083C59  mov         eax,dword ptr [esi]  
51083C5B  sub         edi,dword ptr [eax-4]  
51083C5E  mov         ecx,edi  
51083C60  call        _CDIDev_EnterCrit_@12 (5107B5C8h)

sthalik, 2016-07-01:
@crucified77 since you have a debugger - what are other threads doing at the time? This is a race condition so that sadly won't suffice.

sthalik, 2016-07-02:
@crucified77 there's no need to post assembly.

sthalik, 2016-07-02:
@crucified77 you have a "threads" view in Visual Studio along other debug windows. If not, go to the "View" menu.

crucified77, 2016-07-06:
Threads and stacks for each:
[4368] msvcr.dll!_threadstartex
>	44001810()	Unknown
 	[Frames below may be incorrect and/or missing]	
 	libopentrack-logic.dll!003cbc69()	Unknown
 	libopentrack-tracker-joystick.dll!667893ec()	Unknown
 	libopentrack-tracker-joystick.dll!667893ec()	Unknown
 	[External Code]	
 	libopentrack-logic.dll!003cc141()	Unknown
 	libopentrack-logic.dll!003cc141()	Unknown
 	Qt5Core.dll!6695d211()	Unknown
 	Qt5Core.dll!6695d211()	Unknown
 	[External Code]	

[12400] msvcrt.dll!_threadstartex
 	[External Code]	
>	Qt5Core.dll!66957686()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for Qt5Core.dll]	
 	Qt5Core.dll!6695798a()	Unknown
 	libopentrack-logic.dll!003cbf2a()	Unknown
 	libopentrack-logic.dll!003c601b()	Unknown
 	Qt5Core.dll!6695d211()	Unknown
 	[External Code]	

[12780] dinput8.dll!CEm_LL_ThreadProc
 	[External Code]	

[12576 Main Thread]
 	[External Code]	
>	Qt5Core.dll!66ae60b8()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for Qt5Core.dll]	
 	[External Code]	
 	nvsmartmax.dll!6cc77bb2()	Unknown
 	[External Code]	
 	nvsmartmax.dll!6cc77bb2()	Unknown
 	[External Code]	
 	nvsmartmax.dll!6cc77bb2()	Unknown
 	[External Code]	
 	nvsmartmax.dll!6cc77bb2()	Unknown
 	Qt5Core.dll!66baebe9()	Unknown
 	Qt5Core.dll!66aaabe2()	Unknown
 	[External Code]

crucified77, 2016-07-06:
BTW: I do see Qt in the stack and so I checked to see if I have Qt installed separately.  Not sure if it can interfere but it was installed on 2/19/2016 (Version 2.0.1).  So it's probably out of date Qt 5, I'm guessing.  I tried to update the Qt Components but I have no C drive space, so I need to do more cleanup.

sthalik, 2016-07-06:
@crucified77 there's Qt in the executable's directory so it's tried first. You can verify that with Dependency walker.

sthalik, 2016-07-06:
In depends.exe select "show full paths".

sthalik, 2016-07-06:
@crucified77 the "nvsmartmax" is some "nvidia smart maximize helper". It force-loads itself into opentrack. Try disabling it.

Can you run gdb as the debugger though? It's no problem to make a debug build, but MSVC++ won't give you a stack trace even with debug build.

crucified77, 2016-07-06:
The Dependency Walker showed the Qt used was the one that I is from the opentrack folder.  There were some some issues found with the tool but not sure if they are significant.  Saved stuff:

[opentrack_dep_walker.zip](https://github.com/opentrack/opentrack/files/349162/opentrack_dep_walker.zip)

crucified77, 2016-07-06:
I killed two "NVidia smart Maximize helper"'s and re-ran it crashed the same but the treads changed a bit.  I still see nvsmartmax.dll in the main thread call stack.  There are two ntdll.dll threads and a combase.dll!CRpcThreadCache, with the two msvcrt.dll and dinput8.dll.  

Could I run with gdb?  I can retry.

crucified77, 2016-07-06:
With gdb I get it to hang:

<http://www.gnu.org/software/gdb/bugs/>.
(gdb) file opentrack.exe
Reading symbols from E:\Users\David\Downloads\opentrack-test-20160628_1\install\opentrack.exe...(no debugging symbols found)...done.
(gdb) run
Starting program: E:\Users\David\Downloads\opentrack-test-20160628_1\install/opentrack.exe
<hangs - don't see GUI>

sthalik, 2016-07-06:
@crucified77 wait for it to crash. Don't interact with the dialog saying it crashed.

In cmd.exe, type `tasklist -fi "IMAGENAME eq opentrack.exe"`

Then in gdb, type `attach PID` where PID is from tasklist.

sthalik, 2016-07-06:
@crucified77 debug build - https://www.dropbox.com/s/p2zb2y9amoq42ot/opentrack-debug-20160706.7z

crucified77, 2016-07-07:
I attached to it after it crashed with gdb.  Unfortunately I'm not sure if it's helpful.  
[gdbLog1.zip](https://github.com/opentrack/opentrack/files/351146/gdbLog1.zip)

Then I tried again using gdb with 'file opentrack.exe' and 'run' and I actually got it to run and then crash.  After it crashed I typed 'thread apply all bt'.  Maybe this file has enough to debug?
[gdbLog2.zip](https://github.com/opentrack/opentrack/files/351151/gdbLog2.zip)

sthalik, 2016-07-07:
@crucified77 in `Thread 5 (Thread 11028.0x14c4)` it tries to create a dinput8 handle but freezes rather than being done with it. I now moved dinput8 handle creation to as soon as program starts.

You should see a debug message "made di handle" really fast, even before plugins are loaded. For me it's just below "registering stream operators".

It's unfortunate that we have no line information in gdb. Some toolchain snafu as usual.

A build that hopefully fixes: https://www.dropbox.com/s/zwqb5r1kih19bhf/opentrack-debug-20160707.7z

crucified77, 2016-07-09:
I ran it and it still crashes.  Attached are the logs from running it from gdb:

[gdbLog3.zip](https://github.com/opentrack/opentrack/files/355744/gdbLog3.zip)

crucified77, 2016-07-09:
I retried after logging out and back in.  This time I killed the two "NVidia smart Maximize helpers" and killed the Gigabyte overclocking app OC GURU.  Good news is it didn't crash!  I'm not sure what the key was, but if I figure it out i can report it.

sthalik, 2016-07-10:
@crucified77 thank you for the report. As for the libraries, they inject themselves into running opentrack and mess around with it. opentrack doesn't care, but these libraries do something funky.

I'm closing. Are we good?

sthalik, 2016-07-10:
<pre>
Thread 7 (Thread 11612.0x1090):
#0  0x774ac2ac in ?? ()
#1  0x75ce8a9b in KERNEL32!DeviceIoControl () from C:\WINDOWS\SysWOW64\kernel32.dll
#2  0x5a4b85c6 in ?? () from C:\WINDOWS\SysWOW64\dinput8.dll
#3  0x5a4b50cf in ?? () from C:\WINDOWS\SysWOW64\dinput8.dll
#4  0x100018c6 in GvKmMouseRButtonClick () from C:\Program Files (x86)\GIGABYTE\GIGABYTE OC_GURU II\GvKm.dll
</pre>

Wow this is way bad.

closed 2016-07-10 by sthalik

sthalik, 2016-07-10:
@crucified77  This is gigabyte oc guru that's the rogue library here.