opentrack/opentrack #316: Intel RealSense

SinusoidDelta, 2016-02-14:
i'm interested in purchasing a real sense camera for a tracking but I don't know what the capabilities of it. Are there any videos demonstrating the accuracy in capability? Is functionality comparable to a 3 point model?

sthalik, 2016-02-14:
@ph0b can you elaborate here please?

ph0b, 2016-02-15:
I've made a demo video last week, you can have a look at it: 

[image]

I'd honestly can't tell about the performance compared to a 3 point model as I haven't tried it.

I've made the video with an F200 camera, which is being replaced really soon by the new generation (SR300) which should work even better, especially in low light and at greater distance than the F200.

SinusoidDelta, 2016-04-15:
Impressive but the response doesn't seem to rival the 3 point model. I would love to stop wearing this thing on my head. I hope that is a possibility in the near future. Thanks for the response!

sthalik, 2016-04-23:
Depends. The filter settings may not be optimal. For more noisy trackers it's also possible to achieve a fast response that's hard to see as inferior. I don't know how noisy it is.

sthalik, 2016-04-23:
@ph0b does RealSense have filtering of the 6DOF pose enabled? FaceAPI had and it was horrible. If it has any try disabling it. In worst case we can change Accela.

ph0b, 2016-04-26:
@sthalik No, it's already disabled (set it to its minimum, to be precise), as it was indeed better than with more filtering. The next release should hopefully bring less noisy tracking.