opentrack/opentrack #317: Support for Wiimote

xAccesXnova, 2016-02-18:
Hello,

I've noticed that a lot of users want to use a Wiimote in Opentrack, me included. 
If it's possible i'd really like if support for a Wiimote was built into Opentrack.

Greetings!

fred41, 2016-02-23:
... yes, that would be really great ...

At least linux, already provide a very good kernel driver 'xwiimote' and example library for easy usage.

sthalik, 2016-02-25:
There were barely any requests for Wiimote support actually.

If any Wiimote tracker gets written at all, it needs to work on Windows first and foremost. Let's not kid ourselves, Linux has less than 2% of total users. It's also only supported for developer convenience, not general usage as per FAQ.

If you write the code you can subject it to review via a pull request as usual.

fred41, 2016-02-26:
... yeah, lets hope (and help) that it will be more then just 2% in the near future (steamOS, Vulkan) ...

As an little research project, i am considering to write a virtual device (uinput) that transforms xwiimote's IR point data to, 6 ordinary joystick/gamepad axis. Games like 'WarThunder' are supporting this, as alternative view control input. Maybe later a plugin for opentrack will be grow out of that.

sthalik, 2016-02-26:
Writing an opentrack plugin is trivial actually. The "hacking opentrack" page on the wiki describes the interface to be implemented. Also the header file is heavily commented.

jguenther, 2016-02-27:
I would love this. Freetrack crashes a lot and is very unreliable in general and hasn't been updated since 2008.

fred41, 2016-03-05:
I have started to make a small plugin for WiiMote (based on point tracker). 

While tested opentrack (fresh from git) on debian i found the following strange behavoir:
 
If i stop a running tracker, it stops the tracker loop (console message: 'Tracker:: Thread stopping'), but leaves the userinterface disabled (ctrl-C to terminate).
Is this a known issue?

sthalik, 2016-03-05:
Too little information. Are you inheriting from `QThread`? What's the thread's backtrace in case of the hang?

fred41, 2016-03-05:
looks like it waits in Tracker_PT::~Tracker_PT destructor for something, i got this backtrace:

#0  0x00007ffff580b48d in nanosleep () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007ffff580b324 in __sleep (seconds=0) at ../sysdeps/unix/sysv/linux/sleep.c:138
#2  0x00007fffe2397844 in Tracker_PT::~Tracker_PT (this=0xc25db0, __in_chrg=<optimized out>) at /home/player/opentrack/tracker-pt/ftnoir_tracker_pt.cpp:43
#3  0x00007fffe239790e in Tracker_PT::~Tracker_PT (this=0xc25db0, __in_chrg=<optimized out>) at /home/player/opentrack/tracker-pt/ftnoir_tracker_pt.cpp:45
#4  0x00007ffff79bde94 in std::_Sp_counted_ptr<ITracker*, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0xc26c10)
    at /usr/include/c++/5/bits/shared_ptr_base.h:374
#5  0x000000000041e04e in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xc26c10) at /usr/include/c++/5/bits/shared_ptr_base.h:150
#6  0x000000000041dcc3 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7fffffffd038, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/shared_ptr_base.h:659
#7  0x00000000004347cc in std::__shared_ptr<ITracker, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7fffffffd030, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/shared_ptr_base.h:925
#8  0x000000000043b39f in std::__shared_ptr<ITracker, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<ITracker, (__gnu_cxx::_Lock_policy)2>&&) (
    this=0x6f6bc0, __r=<unknown type in /usr/local/opentrack/opentrack.bin, CU 0x144e57, DIE 0x1d1583>) at /usr/include/c++/5/bits/shared_ptr_base.h:1000
#9  0x0000000000438c1c in std::shared_ptr<ITracker>::operator=(std::shared_ptr<ITracker>&&) (this=0x6f6bc0, 
    __r=<unknown type in /usr/local/opentrack/opentrack.bin, CU 0x144e57, DIE 0x1ce03c>) at /usr/include/c++/5/bits/shared_ptr.h:294
#10 0x00000000004354a5 in SelectedLibraries::operator=(SelectedLibraries&&) (this=0x6f6bc0) at /home/player/opentrack/opentrack/selected-libraries.hpp:21
#11 0x0000000000435587 in Work::~Work (this=0xc44e90, __in_chrg=<optimized out>) at /home/player/opentrack/opentrack/work.hpp:59
#12 0x00000000004464ee in __gnu_cxx::new_allocator<Work>::destroy<Work> (this=0xc44e90, __p=0xc44e90) at /usr/include/c++/5/ext/new_allocator.h:124
#13 0x00000000004463b5 in std::allocator_traits<std::allocator<Work> >::_S_destroy<Work> (__a=..., __p=0xc44e90) at /usr/include/c++/5/bits/alloc_traits.h:285
#14 0x0000000000446225 in std::allocator_traits<std::allocator<Work> >::destroy<Work> (__a=..., __p=0xc44e90) at /usr/include/c++/5/bits/alloc_traits.h:414
#15 0x0000000000445991 in std::_Sp_counted_ptr_inplace<Work, std::allocator<Work>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0xc44e80)
    at /usr/include/c++/5/bits/shared_ptr_base.h:531
#16 0x000000000041e04e in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xc44e80) at /usr/include/c++/5/bits/shared_ptr_base.h:150

sthalik, 2016-03-05:
As per code it sleeps for exactly one second. Is this not the case?

fred41, 2016-03-05:
no, it really hangs, from the moment you press the stop button. Doesn't respond anymore.

sthalik, 2016-03-05:
So you have in frame 2

`portable::sleep(1000);`

That expands to

`usleep(1000U * 1000U);`

as from `usleep(3)` that takes microseconds. This looks correct.

But I don't understand how frame 2 comes from frame 3. Please rerun with valgrind (it's good you already have `-g` hence line information). Disable leak checking, and ignore the few initial errors coming from Qt5, X11 and so forth. There should be something bad happening that corrupts the stack or similar.

fred41, 2016-03-05:
Did you try to reproduce on your  side (without camera)?

There is a comnent in your code:
    // fast start/stop causes breakage
    portable::sleep(1000);

... probaply related ?

fred41, 2016-03-05:
... funny, just now, opentracks windows pops in the forground, obviously not dead, just sleeped a few minutes. scary :)

sthalik, 2016-03-05:
The sleep works for me, time spent is close to one second, see:

`
#include <unistd.h>
int main(void)
{
    usleep(1000U * 1000U);
    return 0;
}
`

Maybe the sleep call is fine but `camera.stop()` isn't? See how frames 2 and 3 from your backtrace don't make sense, 3 exists from the dtor but 2 jumps in a random point in the function. That's why Valgrind's necessary.

Also opencv has two video input implementations based on libv4l1 and libv4l2. The check in their `cmake/` subdirectory is incorrect, looking for both the obsolete and working version. try `grep` for `LIBV4L` and you can find the check. I have to correct it myself each time building a static version of opencv on Linux.

fred41, 2016-03-10:
... finally my wiimote tracker works.

I will upload it to my opentrack fork now (hope i get it sorted :).
If you have the time to review it, i would appropriate suggestions/ideas for improvements.

fred41, 2016-03-11:
WiiMote-tracker plugin project canceled, sorry.

sthalik, 2016-03-12:
Too bad it happened. Closing due to moot issue.

closed 2016-03-12 by sthalik

jguenther, 2016-03-16:
@fred41: I might be interested in maintaining that project, but the link is a 404 now. Is the code hosted anywhere?

fred41, 2016-03-17:
You can extract it from my pull request, here in this original repository.

Important code is in files:

camera.* 
ftnoir_tracker_wiimote.*
wiimote_monitor_widget.h

Please read readme.txt, it is important too for setup.

Code is already working really fast and stable, just the user interface have to be refined and maybe some axes fliped.

If you have questions, just let me know.

jguenther, 2016-03-31:
@fred41 ah right, thanks I'll take a look!