opentrack/opentrack #318: Opentrack PT Losing Tracking for Seemingly No Reason PS3 Eye

Rumils, 2016-02-23:
Hey guys! I am having some trouble with opentrack tracking constantly compare to other programs like facetracknoir (pt 1.0 not perfect but 99%) and freetrack 2.2 (best results). Not really sure what my problem is. The program keeps tracking all my dots however it loses the red cross over my center cross. This red cross then proceeds to flash back and forth between the center dot in and the top left corner. Even if I basically hold my head in the same position. I have tried many different camera settings and have played with the threshold and point size for hours. I also tried to put a piece of tape over the top left corner of the camera which had no effect. I am using the ps3 eye as my camera and have removed the ir filter and put in a piece of floppy disk. Am I missing something? I really like the software as it has a lot more customization compared the the previously mentioned programs. Any advice would be appreciated :) Thanks for your time :)

Pictures attached showcase my problem. Turned the lights off to further isolate any chances of stray light. 

[image]
[image]

Len62, 2016-02-23:
Have you calibrated the camera? With the program running click on the box next to PointTracker.  Click on the Model tab and make sure you have the proper dimensions for your clip. Then in the Model position click the link for the wiki on how to do the calibration. Once completed the red cross will be somewhere to the left or right side of the 3 green crosses.

Rumils, 2016-02-23:
I do have the correct model sizes. However I have tried to calibrate it however I think I may have done it wrong. I will give it another shot. Thank you :)

MathijsG, 2016-02-23:
Please add some screens, for example the Track settings so we can see more of your settings.

sthalik, 2016-02-25:
The raw values look fine from the screenshots.

Proper calibration procedure has its own page on the wiki.

Rumils, 2016-02-25:
Hey guys sorry. Work got busy and I am currently out of town. I will post the screenshots this weekend. Thank you for the suggestions so far :)

Rumils, 2016-02-25:
As for calibration if I remember right. I kept having issues even when following the guide. I would get all three numbers at the maximum lowest setting -65665 or. something. This is probably due to loosing tracking during the calibration. Like I said before not sure why I can't seem to get as steady tracking with pt 1.1 as I can with pt 1.0 
Thanks again. :)

MathijsG, 2016-02-28:
I remember I had kind of the same with a friend where I made a trackhat for. Despite not seeing strange things with the leds going on while moving I did notice that the calibration went wrong with numbers like -65665 indeed. But the leds never where invisible on screen.

sthalik, 2016-02-29:
@MathijsG use the wiki's procedure for calibration.

The process is statistical and given not enough input will indeed lead to deconvergence.

closed 2016-06-21 by sthalik