opentrack/opentrack #319: vJoy 2 (from sourceforge) support broken?

kromit, 2016-02-24:
I've seen #40 so I guess it was working once bit the current opentrack version does no output to current vJoy 2.1.6 devices. Headsoft VJoy 1.2 seems to work (after change via uninstall/reinstall) but vJoy 2.1.6 does not even react.

sthalik, 2016-02-25:
I'd like to replace the proprietary vJoy protocol support with this opensource project with the same functionality <<https://github.com/shauleiz/vJoy>>. Any objections?

kromit, 2016-02-25:
Is this a fork of the SF version or a working copy? I would be happy to see the project moving from SF to GitHub

mirh, 2016-07-06:
They are indeed developing on github (just try to press code [here](https://sourceforge.net/projects/vjoystick/)). 
They still use SF to provide binaries, forums and website then.

sthalik, 2016-09-21:
A module's been implemented and replaces headsoft vjoy.

closed 2016-09-21 by sthalik