opentrack/opentrack #322: [CMake] GNUInstallDirs support

reavertm, 2016-03-08:
It would be nice (for distro packagers) if for its core binaries and libs, opentrack supported standard linux directories via GNUInstallDirs: https://cmake.org/cmake/help/v3.0/module/GNUInstallDirs.html, 

At the moment, buildsystem's default 'install' target is unfortunately not very useful under Linux.

sthalik, 2016-03-08:
Change the install prefix to `/opt/opentrack/libexec/opentrack` or `/usr/local/libexec/opentrack`.

Then, create a shell script in `bin/` with the equivalent of `exec "$(dirname "$0")/../libexec/opentrack/opentrack.bin "$@"`

Also note that Linux isn't officially supported. Support exists only for developer convenience and to increase portability.

sthalik, 2016-03-08:
Furthermore, using Qt library wrappers rather than `dlopen(3)` causes symbol conflicts between plugins and we have plenty of these. For a long time this was the reason opentrack wouldn't work on Unix sytems with don't have implicit equivalent of `RTLD_DEEPBIND` as on Windows.

sthalik, 2016-03-12:
Closing WONTFIX

closed 2016-03-12 by sthalik

reavertm, 2016-03-12:
Using QtStandardPaths would increase portability all around, but hey...

sthalik, 2016-03-13:
You can conditionalize `cmake` install() to use libexec/opentrack and copy a shell script with `@CMAKE_INSTALL_DIR@/libexec/opentrack`. I'll review any patch like that.