opentrack/opentrack #324: can't auto save camera settings

liuxyc, 2016-03-09:
i need to adjust the camera settings every time when start the software

MathijsG, 2016-03-09:
Can you provide some more details?

What Camera, what driver version? Which operating system?

liuxyc, 2016-03-10:
i'm using win10-x64, my camera/driver info:
[image]

sthalik, 2016-03-10:
opentrack only pops up the camera dialog. Saving is the system's business. There's nothing the opentrack project can do about the problem.

sthalik, 2016-03-10:
Edit I have to do the same with a generic camera driver. Camera functions are delegated to opencv which has other problems like not opening the PS3 Eye dialog properly.

I'd like to help but DirectShow graph is a complex thing to construct and I've never had any experience with it.

For now please use other program like Skype to open camera settings, it's known to do it the right way.

liuxyc, 2016-03-10:
thanks sthalik for analyzing this issue, i use freetrack setup the camera and the settings saved.

sthalik, 2016-06-18:
@liuxyc We've now made changes to opencv in order to invoke the dialog properly.

You can retry using newest rc49, maybe it'll work.

sthalik, 2016-07-19:
There's nothing I can do unless someone else contributes working code.

closed 2016-07-19 by sthalik