opentrack/opentrack #327: Arduino leonardo CPU usage problem

drdanilov21, 2016-03-13:
HI all!
I have a problem with Arduino leonardo and GY-85, it works sucsessfully, but OpenTrack using about 30% of CPU. The same PC and UNO+GY-85 using about 1-2% of CPU. Its not just my problem, someone else have the same problem. Is it problem with Leonardo or OpenTrack? How I can fix it? Thanks.

drdanilov21, 2016-03-13:
i know what the FaceTracknoir with this HW dont using CPU. Its Haitre v.2.1.0 in FTnoir and haitre v.2.1.1 in OpenTrack. How I can change the version of Haitre in OpenTrack?

sthalik, 2016-03-15:
My bet is that the data is sent at a very fast rate, and opentrack has to read the incoming bytes 250 times a second.

What you can do to help me debug it - initiate a serial transmissions through `minicom` or `telnet` etc. but dump the data transmission to a file, for at least several seconds. Then I'll replay it in my debug environment.

drdanilov21, 2016-03-15:
[puttyLog.zip](https://github.com/opentrack/opentrack/files/173376/puttyLog.zip)

It's a Putty log file. Is it OK?

sthalik, 2016-03-15:
How many seconds is it?

drdanilov21, 2016-03-15:
its about 10 sec.

drdanilov21, 2016-03-15:
1 Minute Putty Log here.
[puttyLog1MIN.zip](https://github.com/opentrack/opentrack/files/173465/puttyLog1MIN.zip)

Ec0nomist, 2016-04-10:
Same problem. Any news?

sthalik, 2016-04-10:
I'm on hiatus so contributing is the best way of changing things now.

On the other hand, try this simple "fix" - https://www.dropbox.com/s/1c7eeoydwcq765p/opentrack-debug-hatire-v1.7z?dl=0

Does it change anything?

drdanilov21, 2016-04-10:
CPU usage 2%, problem solved, thanks!

sthalik, 2016-04-10:
Please try this version also, and compare CPU usage

https://www.dropbox.com/s/6p1j5g2eomle7r9/opentrack-debug-hatire-v2.7z?dl=0

Ec0nomist, 2016-04-10:
opentrack-debug-hatire-v1 ~ 1-2%
opentrack-debug-hatire-v2 ~ 10-12%

sthalik, 2016-04-10:
v1 limits to 60 serial reads per second. Are you guys fine with that? I'm committing this as a fix for now.

sthalik, 2016-04-10:
@Ec0nomist @drdanilov21 please try this version. The previous "solution" used an explicit sleep statement for 15 ms, blocking other work and the UI. This one uses a timer and so is harmless.

https://www.dropbox.com/s/bbd7fqlw819m7as/opentrack-debug-hatire-v3.7z?dl=0

Ec0nomist, 2016-04-10:
v3 -16-18%

drdanilov21, 2016-04-10:
v3 - 17%

sthalik, 2016-04-10:
Thanks! The "sleep" solution blocks other stuff and I need to update hatire some more. In the meantime use v1.

sh

sthalik, 2016-04-29:
@drdanilov21 @Ec0nomist try this version https://www.dropbox.com/s/x4w7u7wtqn5wg7b/opentrack-2.3-rc21p11-19-g38dd6e5-20160429103403.zip

Tell me if it works, if it improves and other relevant details.

Ec0nomist, 2016-04-30:
0%? :) Sometimes shows 1%. The Falcon and the DCS world works, but more sharply than in opentrack-debug-hatire-v1? Checked "on the go". Better check can only in the middle of next week ....

Ec0nomist, 2016-05-01:
In opentrack-debug-hatire-v1 more smoothly. 
https://www.youtube.com/watch?v=fe6h8inDsxA
https://www.youtube.com/watch?v=JSuZSuHEIAc

sthalik, 2016-05-04:
@Ec0nomist please compare CPU usage and smoothness https://www.dropbox.com/s/ixgddnjr0je02sy/opentrack-hatire-test-proper-v4.7z

sthalik, 2016-05-04:
@Ec0nomist @drdanilov21 the new build has 120 Hz update rate, previous one had 60 Hz.

sthalik, 2016-05-04:
@Ec0nomist @drdanilov21 updated build at 200 Hz: https://www.dropbox.com/s/dzz9cznbo8fq71v/opentrack-hatire-test-proper-v5.7z

drdanilov21, 2016-05-04:
v5 - 15-17% CPU usage

sthalik, 2016-05-04:
200 Hz means sleep interval of 5 ms. I'll now try 90 Hz, sleep interval of 10 ms.

Bear with me please, we need a smooth interval with no heavy CPU usage.

The build: https://www.dropbox.com/s/yf1p765eu9wf9s2/opentrack-hatire-test-proper-v6.7z

sthalik, 2016-05-04:
Keep in mind we may use a smaller sleep interval as long as there's no CPU business like this. I have no device hence the need for iterative builds.

drdanilov21, 2016-05-04:
v6 - 5-7% CPU usage. I think its normal, but maybe lets try 80hz?

sthalik, 2016-05-04:
Let's try the same sleep value but after I optimized the code path.

https://www.dropbox.com/s/zwn9zhljejud9na/opentrack-hatire-test-v7.7z

Thanks @drdanilov21 for the data dump.

drdanilov21, 2016-05-04:
U R welcome!
v7 - 6-9% CPU usage.

sthalik, 2016-05-04:
Here's a further optimized build with 83 Hz sleep timer.

https://www.dropbox.com/s/6mcp9g6ofgw2ozy/opentrack-hatire-test-v8.7z

On Windows we can't sleep in a resolution better than a millisecond.

Note that for the 140 KB test file I can replay it every second with no CPU load.

sthalik, 2016-05-04:
I have a theory that it's only the serial/usb/bt access that causes the high load. I can replay a really large text file with no issues and no load.

I'll later provide instructions on setting up SysInternals Process Explorer. We should validate that.

@drdanilov21 @Ec0nomist I'd like to know details about the way you connect the arduino to the PC.

sthalik, 2016-05-04:
Also are you getting debug info `Can't find HAT frame` at all? If you do, do you get it often?

sthalik, 2016-05-04:
Another question. During high CPU usage do you get tracking data at all? There's something that looks like a bug in resync code that I might've fixed in v8.

Ec0nomist, 2016-05-04:
arduino to pc via usb.
In version 2.3p11 cpu - 30%, tracking - the best.
In opentrack-hatire-test-v8 has a small delays. How many Hz in https://www.dropbox.com/s/1c7eeoydwcq765p/opentrack-debug-hatire-v1.7z?dl=0 ? 
Ordered arduino nano v3, I check how it will operate with the same firmware and the same sensor.

sthalik, 2016-05-04:
v1 does it differently. It polls on demand, unlike the Qt boot loop with polling the serial.

People complain about the same problem with certain serial -> usb cables.

I'll give an updated version without polling on demand. It's not hard.

sthalik, 2016-05-06:
@Ec0nomist @drdanilov21 here's a test build with a console window https://www.dropbox.com/s/4nmuxihes7yl3ot/opentrack-hatire-test-v9.7z

You'll have a bunch of lines like

<pre>stat: avg 11.5346 stddev 0.649343
stat: avg 11.5149 stddev 0.627456
stat: avg 11.5196 stddev 0.610663</pre>

Let the software run for a minute or so and after that paste me 5-10 of these lines. They come every seconds. You can make a screenshot or a text copy. Both numbers are relevant.

Ec0nomist, 2016-05-06:
http://i.piccy.info/i9/a877ce61605ef1724569fd921f60ad6c/1462558143/257897/1031623/1.jpg

sthalik, 2016-05-07:
I think high CPU usage is a hardware fault, in particular the USB to serial converter you use with the device. It's not opentrack that's generating the CPU usage I'm sure despite not having concrete CPU usage data for threads (one that sysinternals procexp would provide for each thread).

How is the smoothness though? It should be better than earlier versions. There was one particular change made. Was v1 better?

This belongs as a checkbox in hatire options (CPU usage bug workaround) and not enabled by default that's for sure.

sthalik, 2016-05-07:
Even in case of the hardware bug I should give you guys a proper workaround, let there be no doubt on that.

sthalik, 2016-05-07:
@Ec0nomist for your earlier question, with a delay of 11 ms we have 90 Hz.

Ec0nomist, 2016-05-07:
opentrack-2.3 rc21p11(26-30%), test V1(12%) and v9(3-6%) very similar in use (in my opinion)
rc21p11: https://www.youtube.com/watch?v=WWj7kDKhvSU
test 1: https://www.youtube.com/watch?v=MylbkkVcC3c
test 9: https://www.youtube.com/watch?v=tzXmPlQiVhM
"..I think high CPU usage is a hardware fault, in particular the USB to serial converter you use with the device..." - showed topic Mega_mozg, mmjoy2 developer. About Promicro and analogues he should know all. I, unfortunately , do not know the theme , can only test. And there are problems with English :)

sthalik, 2016-05-09:
@Ec0nomist you say listed versions are very similar, and that v9 CPU usage is lowest.

How about we now add a boolean to sleep at all as workaround for some hardware, and treat issue as solved?

sthalik, 2016-05-10:
@Ec0nomist @drdanilov21 please test https://www.dropbox.com/s/yful5m3hsbhjxe9/opentrack-hatire-test-v10.7z

You need to go into hatire settings and select the workaround on top of the first tab. If it's okay with you I'll release a new patchlevel.

**BUT** I've converted hatire to opentrack options API, in effect removing few hundreds of lines. Please tell me if anything else broke that's not related. If something's broke it's serial speed selector and the other options below.

I really like removing code while preserving functionality :)

sthalik, 2016-05-10:
@huliqan can you too see if hatire is broken in any way in this build, please?

Ec0nomist, 2016-05-10:
Trying to switch on com5 (default com1), but the settings are not saved , the tracker does not work

sthalik, 2016-05-10:
@Ec0nomist which settings aren't saved in particular? Is it only the port name?

sthalik, 2016-05-10:
I see it. New build - https://www.dropbox.com/s/a3dk3sz69tucx9w/opentrack-hatire-test-v11.7z?dl=0

I need to leave I'm late.

Ec0nomist, 2016-05-10:
com-ports now ok. In hatire setup yaw and pitch change synchronously. RotX in pitch - and yaw RotX after save. RotY in yaw - Y in pitch....

huliqan, 2016-05-10:
opentrack 2.3-rc21p11 (release ver.) ---   hatire works perfect
opentrack-hatire-test-v10  ---  do not working(serial port reset to wrong)My hatire port is 30?but test-prog_v10 constantly reset port to 12(my Bluetooth)
opentrack-hatire-test-v11  ---  hatire works very well. just same as  opentrack 2.3-rc21p11 (release ver.)

Ec0nomist, 2016-05-11:
huligan - и все оси нормально назначаются? Пробовал на нескольких компьютерах, (и без подключенной ардуино) - везде оси х и y синхронизируются....

and all axes normally appointed? Tried on several computers (without Arduino) - all x-axis and w synchronized ....

huliqan, 2016-05-11:
oops:(((
sorry, really axes settings slipping into RotX + RotX or RotY + RotY: (((

sthalik, 2016-05-12:
Thanks,

https://www.dropbox.com/s/e96i8bazgc7p4r5/opentrack-hatire-test-v12.7z

Please report any further issues.

sthalik, 2016-05-12:
Archive size changed since I fixed the compression level.

huliqan, 2016-05-12:
not working too. Slipping axis again
[image]
[image]

sthalik, 2016-05-12:
Due to IDE issue the binaries weren't generated to the right place.

https://www.dropbox.com/s/fl98hv16mu47wgh/opentrack-hatire-test-v13.7z?dl=0

These are fresh. Sorry for the mix-up.

Ec0nomist, 2016-05-12:
chicly

huliqan, 2016-05-12:
perfect!
bui  I do not understand why these test versions, so if the release version (opentrack 2.3-rc21p11) works fine?

sthalik, 2016-05-13:
@huliqan only through code quality improvement opentrack surpassed ftnoir. We have about 20k lines of code with way more functionality. In 2013 we had the same functionality but few thousand lines less.

The code I replaced (and you guys helped me get it to work without error, thank you) was working worse than the original options code. It's 3x shorter across the whole code base, works better in general for all modules (trackers, filters, protocols), has no edge cases like the "x" button not cancelling or bug due to not repeating 3x or more a single option.

If you doubt the code quality argument, consider John Carmack's work in game development. Guy was the only one in the whole industry (maybe simplifying here a lot) who pushed for modern development practices from Quake, QuakeWorld, Quake 2 all the way while his franchises were selling.

The old ftnoir project leaked memory on purpose cause the code's author was too lazy to keep track of .dll's opened, openly. And many other things.

I'll give it some time if you find any other problem.

sthalik, 2016-05-13:
@Ec0nomist come again please?

huliqan, 2016-05-13:
I have no doubt that whatever you do, professionally and correctly!
Thank you for your works!

Ec0nomist, 2016-05-13:
@huliqan - Some trackers are highly loaded cpu when working with the release version.

[7 more replies]

closed 2016-05-14 by sthalik