opentrack/opentrack #328: Opentrack enhancements for custom models

gyrovague, 2016-03-16:
(copied from my comment at the end of https://github.com/opentrack/opentrack/issues/287 )

Some useful features for opentrack might be:

- Have some more help and pictures in the custom model tab
- Display a preview of the custom model based on the user inputs, the user can then compare this to his actual model expectations and more easily fix input errors. This preview can perhaps just show side and front views, similar to the views in the clip and hat tabs
- Support multiple custom models instead of only one

I wouldn't mind implementing these features if somebody can point me to the simplest steps to get an opentrack build going locally on my machine.

sthalik, 2016-04-06:
1) Custom models are more like multiple profiles. What's the use case?
2) We could project the model like we rotate and translate the octopus rather than having that kind of complicated UI.
3) This is Euclidean coordinates, why'd you need a guide to begin with rather than merely a ruler?

gyrovague, 2016-04-07:
1. Exactly yes, multiple profiles. The use case is: I have a few custom made 3pt headtrackers, with different coordinates on each, so right now I need to nuke my config and enter all coords manually again when switching. In addition, this would make other projects like TrackHat and DelanClip and iRtrack ( http://forums.eagle.ru/showthread.php?t=156362 ) less reliant on a custom branch of OpenTrack, if they could for instance add or import a named profile file (ignoring for the moment the other changes in TrackHat branch like the simplified UI) representing their model.
2. Yes, sure, if 3D rotatable object is easier, by all means. I was thinking of simply showing side and front views like the pictures currently shown on the clip and hat tabs, which requires no 3D code at all, e.g. front view just ignores Z axis (and thus uses X and Y axes for the 2D picture) and side view just ignores X axis, could even include a top view which ignores Y axis. This is all just so the user can be extra sure that the coords he entered correspond to what he thinks his model should look like, helps eliminate errors and increases understanding of the OpenTrack coord system.
3. When I first used the custom model feature of OpenTrack, it was not obvious which coordinates map to which direction on the head, even though they are "just" Euclidean coordinates, I had to go by trial and error until my tracking worked correctly (e.g. whether this is from camera point of view, or head point of view, what does negative Z mean (towards camera or away from camera etc.), is positive X towards the right from user or camera etc.). Having (2) above would already help a lot with that, or just a simple picture and help text describing which coordinate is which, and whether the view is from camera point of view or user face point of view etc. It's all obvious once you know it, but before then it is not. When you open the custom tab, you're greeted with "Location of the two remaining model points with respect to the reference point in default pose"... A newcomer sees that and goes "wait, what is the 'reference point'?? What does 'default pose' mean??" and then abandons it all and runs for the hills. Same thing for the lower part of the dialog, "Model position", the "Instructions" link (to the wiki) is only relevant for the calibration button, and none of it describes what "model position" even refers to. Sure, it's obvious to me now, but I remember when I first encountered this dialog I was confused by it all. I feel like a simple diagram and a bit of text can completely clarify all these aspects of the custom model and demystify it and make it as simple as it really is.

sthalik, 2016-04-07:
@gyrovague implementing a projection using our matrix and pose widget code is a good idea. But we have lots of high-priority stuff I haven't touched yet and it's not critical.

gyrovague, 2016-04-07:
Right, like I said at the outset of this ticket, I can maybe work on these features and submit a pull request.

I wasn't even thinking of using matrix and pose widget at all, but a dead simple 2D coordinate plot for each view, ignoring the axis facing the viewer... e.g. for side view, get the entered Z and Y coords of the points, normalize them, and plot them (reference point as 0,0) etc. etc.

sthalik, 2016-04-07:
@gyrovague it's good to use perspective since that's how stuff looks in reality, hence matrix math.

gyrovague, 2016-04-07:
Good point regarding perspective...

sthalik, 2016-09-21:
Send a pull request if you want this feature.

closed 2016-09-21 by sthalik