opentrack/opentrack #329: Aruco tracker crashes the Opentrack app when yaw angle is too high!

duamutefmc, 2016-03-21:
Hello there Stanisław! First of all congratulations on your amazing work on Opentrack, Accela and the Aruco implementation.

I'm trying to emulate TrackIR via Opentrack/Aruco with an Asus ROG G751JY laptop. Its webcam runs at 30fps, and the refresh frequency around 30hz when no light/exposure correction is used.

I've printed out a 100×100 mm Aruco marketn (id #800) and basically pasted it over my Samsung Gear VR visor cover.

When testing it (even before connecting the VR goggles!) it works like a charm *but* when I look too far towards the left or the right (yaw angle > 60°) the Opentrack app crashes.

Is there any way you can help me solve the issue?

Thank you in advance for your help! :)

sthalik, 2016-04-06:
@duamutefmc please send a video from the camera (not video of opentrack tracking, exactly what camera sees). Also compress the video, even .wmv is fine.

sthalik, 2016-04-06:
@duamutefmc you can record the yawing process then replay it with manycam to see if you have the right video.

sthalik, 2016-05-06:
I'm closing this issue after waiting for a month. @duamutefmc you can always respond with the data I need.

Emton, 2016-05-16:
Hey I've been getting a crash possibly similar to this, I can reproduce it every time practically all I have to do is do a real fast yaw movement as far as I can go and it crashes Opentrack. I'm on windows 8.1, I use PS3 eye cam, aruco tracker with EWMA filter. Getting the crash in IL2 BoS, it doesn't seem to crash doing the same movement within the opentrack program, also I'll try out arma 3 later to see if it crashes there.

sthalik, 2016-05-16:
@Emton please record the video from the camera that causes the crash, I'll replay it and trigger the crash in a debugger.

Emton, 2016-05-16:
How should I do this? I downloaded ManyCam free version but if I try to record in ManyCam I can't use the cam in opentrack.

sthalik, 2016-05-16:
Hey,

Enable the camera as a video source in manycam then you can both use and record in realtime. You can disable the manycam logo in effect options even in the free version.

Please compress the video in any way (Windows Movie Maker is the easiest), it shouldn't be bigger than 50 MB. People used to send GB-ish videos for no reason, just saying.

sthalik, 2016-05-16:
I forgot, then select "manycam video source" as the source in opentrack and it'll pass through the video while recording in real time. You can replay a video selecting it as a source in a similar fashion.

Emton, 2016-05-16:
Here try this and see if it works
http://www.filedropper.com/otbugcrashat42seconds
And here is the last image in opentrack for that crash
[image]
Here are some extra screen shots of crashes, most of them happen on the yaw access just as the tracker loses the image, also a lot of the time when the image gets lost there it takes the yaw value and puts it as the pitch, not sure if that is significant.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sthalik, 2016-05-16:
I have it crashed. This is the most important thing in fixing this. Thank you.

closed 2016-05-16 by sthalik

sthalik, 2016-05-16:
@Emton thank you. I couldn't fix it otherwise.

Here's a build with the fix before the next official installer: https://www.dropbox.com/s/8tny9aosmt19k0x/opentrack-2.3-rc49p0-aruco-fix.7z

sthalik, 2016-05-16:
Also paging @duamutefmc since he might want the build.

duamutefmc, 2016-05-16:
Hi all; well done. That's a good job. I just learnt about your kitten,
Stan, so my condolences. Having lost a few cats myself I know the felling.
=^_^=

Emton, 2016-05-16:
Cool I tried it out for a min and it hasn't crashed. I was just wondering do you know if there will be more image controls for the aruco tracker, maybe stuff like contrast, threshold or anything else that lets us tweak the picture?

sthalik, 2016-05-16:
@Emton If you're using the Eye we could only fake it by a hidden window and programmatic keypresses. The driver authors ignore the settings on purpose to peddle their paid driver.

We could fake it by a hidden window and fake keypresses but there are so many priority issues I won't start it for half a year or more.

If someone contributes code it's a different matter of course.

commit 9d2dd118 (2016-05-16): tracker/aruco: appease bounds check in libaruco

libaruco doesn't clamp the min/max sizes but throws an exception.

Reported-by: @duamutefmc
Data-by: @Emton
Issue: #329

Closes #329