opentrack/opentrack #331: Missing Library

davefoster276, 2016-04-04:
I am getting the following error message when I try to start opentrack.
> ' Library Load Error. One of the libraries failed to load'.
> I have tried re-installing Opentrack but I am still getting the same error?

sthalik, 2016-04-04:
Run Dependency Walker (it has its own website) and see what's missing. It's in the left-top window in Dependency Walker that libraries can be found missing.

lastivka, 2016-04-05:
I have the same issue. After running "Dependency Walker", it comes up with a number of error messages:

Error: At least one required implicit or forwarded dependency was not found.
Error: At least one module has an unresolved import due to a missing export function in an implicitly dependent module.
Error: Modules with different CPU types were found.
Warning: At least one delay-load dependency module was not found.
Warning: At least one module has an unresolved import due to a missing export function in a delay-load dependent module.

Many dozens of DLLs were reported as missing, including:
api-ms-win-core-apiquery-li-1-0.dll
api-ms-win-core-atoms-l1-1-0.dll
api-ms-win-core-delayload-l1-1-1.dll
api-ms-win-core-heap-l1-2-0.dll
api-ms-win-core-heap-l2-1-0.dll
api-ms-win-core-libraryloader-l1-2-0.dll
api-ms-win-core-libraryloader-l1-2-1.dll
api-ms-win-core-localization-legacy-l1-2-0.dll
api-ms-win-core-localization-obsolete-l1-2-0.dll
api-ms-win-core-rtlsupport-l1-2-0.dll
api-ms-win-core-shlwapi-obsolete-l1-2-0.dll
api-ms-win-core-sidebyside-l1-1-0.dll
api-ms-win-core-winrt-error-l1-1-1.dll
api-ms-win-security-base-l1-2-0.dll

I have Windows 10 as my OS.

sthalik, 2016-04-05:
The software works for me in win10.

In case you've used the portable .zip file, use the installer since there were some permission problems in the past.

sthalik, 2016-04-23:
@lastivka @davefoster276 any updates?

davefoster276, 2016-04-25:
Hi Stan

It just started to work again without me doing anything!? Weird. 

Thanks

Dave.

closed 2016-04-27 by sthalik