opentrack/opentrack #332: Trackhat, Opentrack and TS2016

brianwhtn43, 2016-04-05:
Hi there,
I wonder if you could help or supply a workaround to an issue I have with opentrack, trackhat and TS2016.
Currently, I have version 1.2p2 of opentrack, which is a modified version for trackhat by James Allen, however I have an issue with it when I play TS 2016.
It is ok when I want to use trackhat while playing TS2016, but if I elect not to use my tracking hat, TS 2016 does not recognise that opentrack is not running, which means I lose the ability to pan around the cab using a mouse or gamepad. I have found a temporary sollution, but it is not ideal.
I move the file NPClient.dll from the root directory of opentrack to a backup folder and when I want to use the hat for tracking I just move the file back to the main directory.
I have also tried using the latest version of opentrack from the github repositoy V2.3-rc21p11, but it has the same issue.
Is it possibel you can look into this and find a permanent solution?
I have also posted to the steam forums and the trackhat forums, hopefully someone will find a solution to this.
If you need any more information let me know.
Thanks for listening.
Brian

gyrovague, 2016-04-06:
> but if I elect not to use my tracking hat, TS 2016 does not recognise that opentrack is not running

Do you actually quit the Opentrack app, or do you only stop the tracking inside the Opentrack app but leave the app running? I find with DCS World that if I leave Opentrack running but with tracking stopped, I also cannot use mouse look, but if Opentrack app has been quit completely (i.e. the app is not running at all) prior to launching DCS, then mouse look works fine.

brianwhtn43, 2016-04-06:
Hi, Yes i shut down the program and check through windows task menu that it has completely shut down.
This is only happening in Train Simulator 2016, no other games have this issue.
Thanks.

gyrovague, 2016-04-06:
FYI I'm not a dev on this project..

but your symptoms sound pretty weird.. Do you get the same after a clean windows reboot, without Opentrack having ever been run yet after the reboot? Just wondering whether maybe TS2016 itself is at fault here, perhaps not releasing references to NPClient.dll and causing it to never unload, or something like that.

brianwhtn43, 2016-04-06:
I am not alone with these symptoms, there are others on steam that have the issue, but they are blaming facetrackNOIR, and as opentrack uses a facetrackNOIR plugin, i suspect that is the cause. Thier solution is uninstall the software and either buy trackir (i don't think soooo, too expensive), and re-install when you want to use tracking!!! I will stick with trackhat and opentrack.
You are right that it could be something to do with TS2016 itself, but there is no way to tell TS2016 to ignore tracking or turn it off, as there is in x-plane or ezdoc camera for FSX, I also have Eurotruck and that is fine.
So for me the issue is between TS2016 and the opentack software, which is why i posted the workaround above, but was hoping that one of the DEV's would be able to comment or find a solution. Dovetail games does not seem to want to help as it is not so wide spread to cause enough of an issue, but maybe a toggle inside opentrack to disable tracking for TS2016.
For the moment i use my solution and all seems fine, it's just inconvienent as if i forget to re-enable the file i have to stop game, re-enable file, possibly restart windows etc.....

sthalik, 2016-04-06:
@brianwhtn43 
> but they are blaming facetrackNOIR, and as opentrack uses a facetrackNOIR plugin, i suspect that is the cause

We're sharing the base code used but neither is the origin here.

> but if I elect not to use my tracking hat, TS 2016 does not recognise that opentrack is not running, which means I lose the ability to pan around the cab using a mouse or gamepad.

This is a long-standing non-trivial issue.

We can't modify NPClient(64).dll as-is because Arma detects it as trusted in its anti-cheat software. This is a long-standing issue. From the npclient code as I read it doesn't appear to be a way to communicate lack of a sender. This is unfortunate.

Surely, we can remove the registry keys that point to the .dll's for freetrack and npclient dll's as long as we're the only opentrack instance running. But this is a heavy-handed solution.

brianwhtn43, 2016-04-06:
> Surely, we can remove the registry keys that point to the .dll's for freetrack and npclient dll's as long as we're the only opentrack instance running. But this is a heavy-handed solution.

sthalik, 2016-04-06:
@brianwhtn43 
I'm implementing this at the very moment.

brianwhtn43, 2016-04-06:
> Surely, we can remove the registry keys that point to the .dll's for freetrack and npclient dll's as long as we're the only opentrack instance running. But this is a heavy-handed solution.

I understand. I also think that is a drastic solution. And if that stops other games from working with opentrack, then maybe not a good solution.

sthalik, 2016-04-06:
@brianwhtn43 it only adds the requirement to start opentrack first. I'll add a checkbox since this issue is heard from time to time.

brianwhtn43, 2016-04-06:
oh ok, that is cool and very quick.
If it works ok will you be able to implement this for the custom build that trackhat uses as well?

sthalik, 2016-04-06:
@brianwhtn43 it'll be in trackhat as a typical merge from opentrack along with all the other new functionality. Can you check if it works for you in opentrack? Note that it's in protocol settings and defaults to off.

Is it better to clear the dll path on opentrack exit or on tracking stop? I can't decide.

brianwhtn43, 2016-04-06:
@sthalik yes i can test it for you, when will it be ready? and where to download?

> Is it better to clear the dll path on opentrack exit or on tracking stop? I can't decide.

mmm thats not an easy one, but i think on exit, that way if you merely want to stop tracking temporarily and then restart tracking you don't need to restart program.

sthalik, 2016-04-06:
https://www.dropbox.com/s/wgcreixh77gehp1/opentrack-2.3-rc21p11-2-g926b2b1-20160406162510.zip?dl=0

sthalik, 2016-04-06:
I'll add clearing only for program exit in the next build.

brianwhtn43, 2016-04-07:
@sthalik 

> I'll add clearing only for program exit in the next build.
I think that is the best option.

Performed a few tests last night. Two issues, 1st, if I run any other game before i run TS2016, it stops TS2016 from panning the cockpit if opentrack is not running, but if i run TS2016 first,without running opentrack, cockpit panning is ok.(that is with the new protocol ticked), but lose tracking abiltiy if i start opentrack after TS2016. Also if you decide to stop using tracking, shutting down opentrack does not allow cockpit panning unless you restart game.
However if i untick the protocol in any of the profiles except TS2016 and run a game, the dll stays loaded and you lose TS2016 cockpit pan.

2nd, If option is ticked in all profiles, you must start opentrack before the game (and have opentrack start as soon as game is detected, as last night it broke ezdoc from working in FSX because it could not detect npclient.dll. This one could cause users to be quite angry, because at the moment you can start opentrack before or after FSX, but new option requires opentracked to be started first.
This may have issues for other games too.
My conclusion is that this option may cause more issues for users even though it may solve the TS2016 issue.
Maybe, a beta test or an unstable version is available for other users to test, as i think you will get users complaining that opentrack has broken there game etc.......
Your thoughts?

sthalik, 2016-04-07:
> but if i run TS2016 first,without running opentrack, cockpit panning is ok

Do you have in regedit HKEY_CURRENT_USER/Software/Freetrack the "Path" key set to something more than an empty string in this case?

> My conclusion is that this option may cause more issues for users even though it may solve the TS2016 issue.

Yeah, the usability remains limited since npclient.dll can't be changed due to anti-cheat programs. I'm ambivalent as whether to keep the option at all.

brianwhtn43, 2016-04-07:
The path key is blank.

sthalik, 2016-04-23:
I can't do anything further and I'd rather not contact BattlEye due to the politics. It's actually serious.

closed 2016-04-23 by sthalik