opentrack/opentrack #333: Simple step by step instructions for getting opentrack compiled

gyrovague, 2016-04-06:
In order for new people to make contributions to opentrack, it would be very helpful to have simple step by step instructions for getting opentrack compiled, including all dependencies that need to be installed etc.

I'm interesting in making contributions for https://github.com/opentrack/opentrack/issues/328 but don't want to spend hours just figuring out how to get opentrack compiling.

The wiki page with cygwin instructions is a start, but it's not detailed enough in my opinion.

Ideally I'd like something that says:

1. download and install this, this, this and this (optionally with links, but of course those can go stale quickly)
2. open this by doing that
3. run that OR
4. click there

_et voila!_

SoulMaril, 2016-04-06:
+1 Here. I tried to compile/debug the source code but failed. Both with QT
and VS.

sthalik, 2016-04-06:
@gyrovague generate Visual Studio .sln file with cmake. You need to be at least vaguely familiar with cmake. It isn't hard to generate per se.

gyrovague, 2016-04-07:
Well, how about then at least then just the detailed steps to do that, e.g.

1. Install cmake and Visual Studio
2. Open cmd prompt
3. Navigate to directory xxx
4. Type 'cmake foo bar blah'
5. Open folder xyz
6. Double click the .sln file

gyrovague, 2016-04-07:
So I tried above so long, and already encountered first issue:
```
CMake Error at cmake/opentrack-qt.cmake:1 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (include)

-- Configuring incomplete, errors occurred!
See also "E:/games/github/opentrack/CMakeFiles/CMakeOutput.log".
```

Which I guess means I need to install Qt5.. busy installing that right now.

gyrovague, 2016-04-07:
OK, simply installing it wasn't enough, getting the same error... evidently I need to do something more to make cmake happy somehow. I suppose I somehow need to tell it where to find qt5

gyrovague, 2016-04-07:
So, I set QT5_DIR to (maybe) the right thing: E:/Qt/5.6/msvc2015_64/lib/cmake/Qt5

and it got past that error, now getting lots of other errors about OpenCV and a bunch of SDK_ prefixed things... Guessing I need to clone the opentrack-depends repo now, and point cmake to that... Again, would've been nice to have step-by-step instructions for all this instead of having to figure it out.

gyrovague, 2016-04-07:
After getting that repo, and pointing cmake at it, I finally got a opentrack.sln file, which I double clicked and it launched VS. I hit the "build" option, and got 48 errors and 23 warnings. I give up for now, this is exactly what I didn't want to be doing.

sthalik, 2016-04-07:
@gyrovague 

Opencv is optional. It should build cleanly with MSVC++ 2015. I'll try to reproduce.

FlyingCircus-, 2016-04-11:
I must say I had a hard time trying to compile it some time ago. I lack the time now to make contributions anyway (or playing sims), but I'm still following the project. Keep up the good work @sthalik

FlyingCircus-, 2016-04-11:
By the way, I see you implemented the changes I sugested to the interface, but Isn't it still a little messy regarding the settings hierarchy (what depends on the selected profile and what not)?
p.s. Trying to be constructive, eh. I'm aware it isn't the most important feature

sthalik, 2016-04-23:
@FlyingCircus- Everything except game detection (switches profiles on its own) depends on the profile.

sthalik, 2016-09-21:
Nothing more to discuss I think. Also we support MSVC++ 2015+ for sure.

closed 2016-09-21 by sthalik