opentrack/opentrack #334: opentrack.bin crashes on ubuntu

rgh2, 2016-04-07:
When I compile the latest version of opentrack and try to run opentrack.bin I get the following error.

Loading "libopentrack-filter-accela.so"
Loading "libopentrack-filter-ewma.so"
*** Error in `./opentrack.bin': double free or corruption (out): 0x00007ffe33b13aa0 ***
Aborted (core dumped)

I need to use a version that allows using a joystick as an input device for linux as I'm trying to use EDtracker.

Roy

sthalik, 2016-04-07:
Run valgrind.

rgh2, 2016-04-07:
Here is the output of valgrind ./opentrack.bin :

roy@roy-Inspiron-5535:/usr/local$ valgrind ./opentrack.bin 
==8831== Memcheck, a memory error detector
==8831== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==8831== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==8831== Command: ./opentrack.bin
==8831== 
==8831== Conditional jump or move depends on uninitialised value(s)
==8831==    at 0x11AD448D: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x7DC3243: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DDDA45: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DDE5F4: g_signal_emit_by_name (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DCA967: g_object_set_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DCB03B: g_object_set (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x5CDD6E5: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.4.2)
==8831==    by 0x5C61FAC: QStyleFactory::create(QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.4.2)
==8831==    by 0x5BFF704: QApplication::style() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.4.2)
==8831==    by 0x5BFFAEC: QApplicationPrivate::initialize() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.4.2)
==8831==    by 0x5BFFB3E: QApplicationPrivate::construct() (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.4.2)
==8831==    by 0x40F547: main (in /usr/local/opentrack.bin)
==8831== 
==8831== Conditional jump or move depends on uninitialised value(s)
==8831==    at 0x807F1D8: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4600.1)
==8831==    by 0x11AD7A94: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11AD8098: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11AD83CE: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x7DC3014: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DD4B9B: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DDDDFB: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DDE12E: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x11C66F53: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11C68267: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11B06CD2: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x7DC6116: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831== 
==8831== Conditional jump or move depends on uninitialised value(s)
==8831==    at 0x807F205: g_utf8_offset_to_pointer (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4600.1)
==8831==    by 0x11AD7A94: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11AD8098: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11AD83CE: ??? (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x7DC3014: g_closure_invoke (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DD4B9B: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DDDDFB: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x7DDE12E: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831==    by 0x11C66F53: gtk_widget_realize (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11C68267: gtk_widget_set_parent (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x11B06CD2: gtk_fixed_put (in /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0.2400.28)
==8831==    by 0x7DC6116: g_cclosure_marshal_VOID__OBJECTv (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4600.1)
==8831== 
Loading "libopentrack-filter-accela.so"
Loading "libopentrack-filter-ewma.so"
==8831== Invalid free() / delete / delete[] / realloc()
==8831==    at 0x4C2D28B: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==8831==    by 0x1AA874DD: FTNoIR_FilterDll::~FTNoIR_FilterDll() (in /usr/local/libopentrack-filter-kalman.so)
==8831==    by 0x41C030: dylib::dylib(QString const&, dylib::Type) (in /usr/local/opentrack.bin)
==8831==    by 0x41E132: dylib::enum_libraries() (in /usr/local/opentrack.bin)
==8831==    by 0x418124: MainWindow::MainWindow() (in /usr/local/opentrack.bin)
==8831==    by 0x40F575: main (in /usr/local/opentrack.bin)
==8831==  Address 0xffefff550 is on thread 1's stack
==8831==  in frame #2, created by dylib::dylib(QString const&, dylib::Type) (???:)
==8831== 
==8831== 
==8831== Process terminating with default action of signal 11 (SIGSEGV)
==8831==  Bad permissions for mapped region at address 0x1AC8C7E0
==8831==    at 0x1AA86EE1: bool QBasicAtomicOps<4>::deref<int>(int&) (in /usr/local/libopentrack-filter-kalman.so)
==8831==    by 0x1AA86B4B: QBasicAtomicInteger<int>::deref() (in /usr/local/libopentrack-filter-kalman.so)
==8831==    by 0x1AA84EB2: QtPrivate::RefCount::deref() (in /usr/local/libopentrack-filter-kalman.so)
[... cut ...]
==8831==                         length64           : 8,368 bytes in 109 blocks
==8831==                         newarray           : 2,192 bytes in 57 blocks
==8831==         suppressed: 0 bytes in 0 blocks
==8831== Rerun with --leak-check=full to see details of leaked memory
==8831== 
==8831== For counts of detected and suppressed errors, rerun with: -v
==8831== Use --track-origins=yes to see where uninitialised values come from
==8831== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
Killed

sthalik, 2016-04-07:
Is your opencv compiled with qt4? Try `ldd opentrack.bin`.

rgh2, 2016-04-07:
Looks like qt5

roy@roy-Inspiron-5535:/usr/local$ ldd opentrack.bin
	linux-vdso.so.1 =>  (0x00007ffe82be1000)
	libopentrack-pose-widget.so => /usr/local/libopentrack-pose-widget.so (0x00007f1733157000)
	libopentrack-api.so => /usr/local/libopentrack-api.so (0x00007f1732f47000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1732d07000)
	libprocps.so.3 => /lib/x86_64-linux-gnu/libprocps.so.3 (0x00007f1732adf000)
	libopentrack-spline-widget.so => /usr/local/libopentrack-spline-widget.so (0x00007f17328cf000)
	libopentrack-compat.so => /usr/local/libopentrack-compat.so (0x00007f17326bf000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f1732037000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f1731af7000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f1731637000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f17312af000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1731097000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1730cc7000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f17309bf000)
	libopentrack-qxt-mini.so => /usr/local/libopentrack-qxt-mini.so (0x00007f17307b7000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f17305af000)
	/lib64/ld-linux-x86-64.so.2 (0x000055dfdfb98000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f173038f000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f1730137000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f172fe27000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f172fae7000)
	libpng12.so.0 => /usr/lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f172f8bf000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f172f65f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f172f43f000)
	libGL.so.1 => /usr/lib/fglrx/libGL.so.1 (0x00007f172f23f000)
	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007f172edd7000)
	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007f172ea3f000)
	libpcre16.so.3 => /usr/lib/x86_64-linux-gnu/libpcre16.so.3 (0x00007f172e7d7000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f172e5cf000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f172e35f000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f172e137000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f172de8f000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f172dc6f000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f172da57000)
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007f172bf9f000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f172bd97000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f172bb8f000)

sthalik, 2016-04-07:
Which branch is it?

> ==8831== by 0x1AA866EA: FTNoIR_FilterDll::getShortName(QString) (in /usr/local/libopentrack-filter-kalman.so)

this belongs to 2.2

rgh2, 2016-04-07:
Here I'm trying to run with the latest branch (marked unstable). But I saw same problem with the version tagged with 2.3-rc21. They both are presumably 2.3. I have compiled 2.2 in the past and installed, but I installed the latest version over it. I do need a version that has joystick support for linux (so I can use with EDtracker).

sthalik, 2016-04-07:
Your source and/or build directory is messed up. Remove the install directory, delete everything from build directory except for `cmakecache.txt`. Make sure the source directory is a clean copy of `unstable`.

sthalik, 2016-04-29:
I added a cmake script causing an error if build and source directory are the same. It should be enough.

closed 2016-04-29 by sthalik