opentrack/opentrack #335: On linux only FreePie and UDPSender Trackers

rgh2, 2016-04-08:
I've built and installed the latest (unstable) version of opentrack on my ubuntu system. The problem is that when I call opentrack.bin the only trackers that can be selected are the FreePie and UDP sender trackers. I want to use a joystick tracker so that I can use opentrack with the EDtracker.

sthalik, 2016-04-08:
There are dependencies required. For the joystick tracker, install the development package of `libevdev` and `pkg-config`.

rgh2, 2016-04-08:
It seems that Cmake is not building all of the trackers. I have installed libevdev and have pkg-config installed but it seems that all of the trackers except FreePie and UDP are not being built. One thing that I have seen is that you have to manually set the CMAKE_BUILD_TYPE property (I just set to release). How do you have cmake build all the trackers ?

sthalik, 2016-04-08:
Hold on apparently we have no joystick *tracker*. We have joystick *protocol* for output. Let's mark this as a blocker.

rgh2, 2016-04-08:
I see a project in the source tree tracker-joystick.

rgh2, 2016-04-08:
Really what I want is to use joystick as input like seen with EDTracker here -> http://edtracker.org.uk/index.php/using/opentrack.

sthalik, 2016-04-08:
@rgh2 it's missing for both Linux and OSX currently. The `tracker/joystick` module is only for Windows.

sthalik, 2016-07-19:
See #234

closed 2016-07-19 by sthalik