opentrack/opentrack #337: Is it possible to use Aruco markers to track headset left and right sides?

freds72, 2016-04-16:
One thing I don't get with Aruco markers: how to find the "right" markers???
So far, only the 787 image is producing correct results.
Testing with image '1' or some other random values doesn't produce the right rotations.

I'd like to have 3 markers: one on the front of my VR headset and 2 for left and right sides (with a 'forced' rotation of +-90 on the yaw angle.

Cheers,
Freds

sthalik, 2016-04-17:
Having multiple markers is so rarely asked for I won't implement it.

What do you mean here by "*right* rotations"?

freds72, 2016-04-19:
Got it - all Aruco markers are evaluated the same by Opentrack (and must be tilted by 90deg).
I'll switch to point tracker + IR leds, seems to offer a faster/more reliable tracking.

Thanks

sthalik, 2016-04-19:
I recommend that too. PT has less noise as well.

closed 2016-04-19 by sthalik