opentrack/opentrack #338: Opentrack not working

Guildens, 2016-04-16:
Hi all

I´m new to all this so please have patience with me.

My gear is gagagu streamer vr + opentracker + cardboard with iphone 6
 
Was able to set up gagagu to stream quite fluidly and opentracker´s octopus move when i move phone and the values of the game data and raw tracker change as I move the phone as well. However this doesnt translate into camera movement in-game  (im trying HL 1).

Have set up opentracker per the youtube tutorial (UDP Sender & UDP receiver) and also several other configurations but still have the same results, the octopus moves but the in game camera doesnt.

Any help please?

Thanks, Andy

sthalik, 2016-04-17:
Half-Life 1 doesn't support any of our head-tracking protocols.

Guildens, 2016-04-17:
Hi for the quick response!

You have a list of games that are compatible with opentrack?

sthalik, 2016-04-29:
@Guildens most are here https://raw.githubusercontent.com/opentrack/opentrack/unstable/settings/facetracknoir%20supported%20games.csv

There are few more, i.e. Microsoft Flight Simulator, generic joystick emulation, and mouse emulation.

Note that mouse emulation takes over regular aiming.

closed 2016-04-29 by sthalik