opentrack/opentrack #339: Oculus runtime 1.3

ahobel1995, 2016-04-16:
Are you planning to update this to Oculus runtime 1.3 anytime soon? I love this program l, but its more or less useless to the oculus consumer base at this point!

Kianoni, 2016-04-17:
I really would like to see this implemented, so here's my vote for runtime 1.3

sthalik, 2016-04-17:
I'm going to update, but not soon. I'm busy with other work.

kkotowick, 2016-04-27:
Any update on 1.3 support? I have a feeling you're about to get a flood of requests for it, what with the CV1 shipping to all pre-orders now.

sthalik, 2016-04-27:
You just create noise by bumping an issue marked `blocker`. They're the highest priority new features.

bizzclaw, 2016-08-10:
Would Opentrack need to be updated with every new version of the oculus runtime?The assumption is that as long as it would work with earlier versions of the runtime that the CV-1 uses, it should still work even after the runtime is updated, right?

sthalik, 2016-08-11:
Older SDK versions support older devices. opentrack can support several SDKs at once, just like right now.

Piffx1, 2016-08-12:
Hi :)
Is there an approximate period for version 2.3.1 ? 
Greetings piff

Karlomat, 2016-09-04:
Hey nice ! Just what i need ! Cant wait until its finnished . Keep up the good work .

deliciousniko, 2016-09-05:
I hope it´ll be finished soon!

Narfvader, 2016-09-06:
Can't wait for this to be released - keep it up!

ColBurton, 2016-09-11:
Keep up the great work, cant wait to get my hands on !

Soren1793, 2016-09-12:
Really great job ! When will it be updated ?

sthalik, 2016-09-16:
I thought it was 1.4.0 now. I'd rather not support 1.3.0 separately unless there's a reason.

sthalik, 2016-09-30:
Please see issue #464 and my last comment. I need it fixed before rc13.

sthalik, 2016-11-01:
Does the Rift build work?

https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

sthalik, 2016-11-01:
Position and rotation must both work.

Piffx1, 2016-11-02:
wuhu :)  yes it works! great! my weekend is save XD

sthalik, 2016-11-02:
Confirm both position and rotation work, please.

sthalik, 2016-11-02:
I also need to know whether SDK 1.9.0 needs to get implemented.

Piffx1, 2016-11-02:
Yes. It has tested briefly everything runs fine. No problems until now. Could tonight test more must now go to work

sthalik, 2016-11-02:
There's no need to test any more. The code doesn't perform differently with time. Thank you for confirmation.

closed 2016-11-03 by sthalik

kkotowick, 2016-11-03:
So to confirm, the Rift CV1 should be fully supported in 2.3.0-rc15 now, even with the newest Rift update?

sthalik, 2016-11-03:
Yes. You can grab the build from dropbox. The dropbox URL is always in `wiki -> experimental builds` also.

sthalik, 2016-11-03:
And to be precise -- the changes are already in the branch that becomes release candidates.