opentrack/opentrack #341: opentrack-2.3 rc21p11 OS X

iDevPro, 2016-04-23:
Dyld Error Message:
  Library not loaded: @rpath/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Applications/opentrack.app/Contents/MacOS/opentrack.bin
  Reason: image not found

sthalik, 2016-04-27:
This is a bug. Fixing.

sthalik, 2016-04-27:
@iDevPro I've posted a new release. Please note, it works even if I `chmod 000` the Qt prefix in my home directory. If you still have problems, reopen and provide more info.

closed 2016-04-27 by sthalik

iDevPro, 2016-04-29:
Process:               opentrack.bin [1637]
Path:                  /Applications/opentrack.app/Contents/MacOS/opentrack.bin
Identifier:            com.github.opentrack
Version:               opentrack-2.3-rc21p11-14-gfb7e86d
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           opentrack.bin [1637]
User ID:               501

Date/Time:             2016-04-29 21:12:20.367 +0300
OS Version:            Mac OS X 10.11.4 (15E65)
Report Version:        11
Anonymous UUID:        F85BD5B7-4491-BB32-B224-3CC3F04AD619

Sleep/Wake UUID:       049C1B76-F9C0-498E-A96A-F5EC9CECCCC0

Time Awake Since Boot: 5800 seconds
Time Since Wake:       410 seconds

System Integrity Protection: enabled

Crashed Thread:        14  Tracker_PT

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
[1637] stack overflow

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8eb7bf72 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8eb7b3b3 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff984c21c4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff984c168c __CFRunLoopRun + 1356
4   com.apple.CoreFoundation      	0x00007fff984c0ed8 CFRunLoopRunSpecific + 296
5   com.apple.HIToolbox           	0x00007fff96fb0935 RunCurrentEventLoopInMode + 235
6   com.apple.HIToolbox           	0x00007fff96fb076f ReceiveNextEventCommon + 432
7   com.apple.HIToolbox           	0x00007fff96fb05af _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fff92de9efa _DPSNextEvent + 1067
9   com.apple.AppKit              	0x00007fff92de932a -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
10  com.apple.AppKit              	0x00007fff92ddde84 -[NSApplication run] + 682
11  libqcocoa.dylib               	0x000000010773437f 0x107712000 + 140159
12  QtCore                        	0x00000001045907a1 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 401
13  QtCore                        	0x0000000104594a65 QCoreApplication::exec() + 341
14  opentrack.bin                 	0x000000010345f1eb 0x103455000 + 41451
15  libdyld.dylib                 	0x00007fff9e84b5ad start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8eb82efa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff89b9d165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff89b9cdcd _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff909a2578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff909a0341 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff909a2578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff909a0341 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff909a2578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff909a0341 start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
[... 437 lines cut ...]
__DATA                            21.8M      272 
__IMAGE                            528K        2 
__LINKEDIT                        97.3M       42 
__TEXT                           178.9M      269 
__UNICODE                          552K        2 
mapped file                        1.1G       26 
shared memory                     16.3M        9 
===========                     =======  ======= 
TOTAL                              1.6G      797 

Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2,3 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.4f4 17685, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: RAPOO 2.4G Wireless Device
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1